earlephilhower/ESP8266Audio #372: Opus codec not working!

GabeHC, 2021-02-22:
I am trying to play Opus on ESP32-A1S, Got error as follow.
What should I do to fix it?

eading SD music files
Start Playing songs on SD card!
Playing '/gs-16b-2c-44100hz.opus' from SD card...
+0 0x3ffb1ec8
***ERROR*** A stack overflow in task loopTask has been detected.
abort() was called at PC 0x40085e64 on core 1

ELF file SHA256: 0000000000000000

Backtrace: 0x40085bd0:0x3ffae4c0 0x40085e4d:0x3ffae4e0 0x40085e64:0x3ffae500 0x40087a07:0x3ffae520 0x400895fc:0x3ffae540 0x400895b2:0x00000000

Rebooting...

GabeHC, 2021-03-02:
Earl, please help. At least let me know what I did wrong!

JonRobo, 2021-06-04:
I have the same problem

GabeHC, 2021-07-09:
move to uses PSRAM. not helping.
increase stack to 16384. still get "stack overflow error",
increase buffer size to 4090, same thing.

Can any one help on this please?

chenbaijian, 2021-12-31:
i have same problem

Zipdox2, 2023-02-17:
Can confirm I have the same problem.