earlephilhower/ESP8266Audio #373: Paused staggering playing MIDI to 8266, MOD is ok

Spartacus1066, 2021-02-22:
**Wemos D1 mini** 8266 at 160Mhz... cannot play more than the first 3 or 4 notes of the MIDI example (Play from Spiffs) without slowing and pop/interruptions. I have tried a **5102** DAC and the **MAX98357** DAC. I am assuming it's too slow to convert the Sound Font (.sf2) samples while playing or...? 

I wish it worked, but for my purposes, even _producing the "synth" sounds of the sample MOD file while sourcing from a MIDI source would be great._
Thanks in advance!

earlephilhower, 2021-02-22:
Use LittleFS.  It's not the DAC,it's because the SPIFFS FS is too slow.

woutput, 2021-11-17:
I can confirm that this modified MIDI examples works on an ESP32. It's actually "PlayMIDIfromPROGMEM"

```
#include <Arduino.h>
#include <WiFi.h>

#include <AudioOutputNull.h>
#include <AudioOutputI2S.h>
#include <AudioGeneratorMIDI.h>
#include "AudioFileSourcePROGMEM.h"

#include "sf2.h"
#include "mid.h"

AudioFileSourcePROGMEM *sf2;
AudioFileSourcePROGMEM *mid;
AudioOutputI2S *dac;
AudioGeneratorMIDI *midi;

void setup()
{
  WiFi.mode(WIFI_OFF); 

  Serial.begin(115200);
  Serial.println("Starting up...\n");

  audioLogger = &Serial;
  sf2 = new AudioFileSourcePROGMEM(sf2_h, sizeof(sf2_h));
  mid = new AudioFileSourcePROGMEM(mid_h, sizeof(mid_h));
    dac = new AudioOutputI2S();
  midi = new AudioGeneratorMIDI();
  midi->SetSoundfont(sf2);
  midi->SetSampleRate(22050);
  Serial.printf("BEGIN...\n");
  midi->begin(mid, dac);
}

void loop()
{
  if (midi->isRunning()) {
    if (!midi->loop()) {
      uint32_t e = millis();
      midi->stop();
    }
  } else {
    Serial.printf("MIDI done\n");
    delay(1000);
  }
}
```

where "sf2.h" and "mid.h" contain arrays sf2_h and mid_h as converted from the binary files using for example HxD