earlephilhower/ESP8266Audio #376: Unable to play MP3 of type mp4a

ksaye, 2021-03-04:
Using the example WebRadio, I am able to play:  http://kvbstreams.dyndns.org:8000/wkvi-am which is mpga, but I can not play mp4a.

Below you see the Codec, as described by VLC.

works:

[image]

fails:

[image]

KodeMunkie, 2021-03-10:
Possibly pointing out the obvious here... but isn't that an MP4 audio stream in a file with an MP3 file suffix?

jayzakk, 2021-03-20:
Try AudioGeneratorAAC, if you want to playback AAC.
AudioGeneratorMP3 only is for MP3 (Layer 1/2)

ksaye, 2021-03-22:
> Possibly pointing out the obvious here... but isn't that an MP4 audio stream in a file with an MP3 file suffix?

I think you are right.  :)