earlephilhower/ESP8266Audio #385: Massice Memory Use

FrankHovis, 2021-04-11:
Why does even a simple demo using just 

#include "AudioFileSourceSPIFFS.h"
#include "AudioFileSourceID3.h"
#include "AudioGeneratorMP3.h"
#include "AudioOutputI2S.h"

result in seemingly every file in the library being included when compiling? Is it really necessary to have the entire lot when all I want is MP3 playback?

Compiling .pio\build\ttgo-t1\libc19\ESP8266Audio\AudioFileSourceBuffer.cpp.o
Compiling .pio\build\ttgo-t1\libc19\ESP8266Audio\AudioFileSourceFS.cpp.o
Compiling .pio\build\ttgo-t1\libc19\ESP8266Audio\AudioFileSourceHTTPStream.cpp.o
Compiling .pio\build\ttgo-t1\libc19\ESP8266Audio\AudioFileSourceICYStream.cpp.o
Compiling .pio\build\ttgo-t1\libc19\ESP8266Audio\AudioFileSourceID3.cpp.o
Compiling .pio\build\ttgo-t1\libc19\ESP8266Audio\AudioFileSourcePROGMEM.cpp.o
Compiling .pio\build\ttgo-t1\libc19\ESP8266Audio\AudioFileSourceSD.cpp.o
Archiving .pio\build\ttgo-t1\lib6c2\libWiFiClientSecure.a
Compiling .pio\build\ttgo-t1\libc19\ESP8266Audio\AudioFileSourceSPIRAMBuffer.cpp.o
Compiling .pio\build\ttgo-t1\libc19\ESP8266Audio\AudioFileSourceSTDIO.cpp.o
Archiving .pio\build\ttgo-t1\lib8e0\libSD.a
Compiling .pio\build\ttgo-t1\libc19\ESP8266Audio\AudioFileStream.cpp.o
Compiling .pio\build\ttgo-t1\libc19\ESP8266Audio\AudioGeneratorAAC.cpp.o
Compiling .pio\build\ttgo-t1\libc19\ESP8266Audio\AudioGeneratorFLAC.cpp.o
Compiling .pio\build\ttgo-t1\libc19\ESP8266Audio\AudioGeneratorMIDI.cpp.o
Compiling .pio\build\ttgo-t1\libc19\ESP8266Audio\AudioGeneratorMOD.cpp.o
Archiving .pio\build\ttgo-t1\libcf1\libHTTPClient.a
Compiling .pio\build\ttgo-t1\libc19\ESP8266Audio\AudioGeneratorMP3.cpp.o
Compiling .pio\build\ttgo-t1\libc19\ESP8266Audio\AudioGeneratorMP3a.cpp.o
Compiling .pio\build\ttgo-t1\libc19\ESP8266Audio\AudioGeneratorOpus.cpp.o
Compiling .pio\build\ttgo-t1\libc19\ESP8266Audio\AudioGeneratorRTTTL.cpp.o
Compiling .pio\build\ttgo-t1\libc19\ESP8266Audio\AudioGeneratorTalkie.cpp.o
Compiling .pio\build\ttgo-t1\libc19\ESP8266Audio\AudioGeneratorWAV.cpp.o
Archiving .pio\build\ttgo-t1\lib442\libTFT_eSPI.a
Compiling .pio\build\ttgo-t1\libc19\ESP8266Audio\AudioLogger.cpp.o
Compiling .pio\build\ttgo-t1\libc19\ESP8266Audio\AudioOutputBuffer.cpp.o
Compiling .pio\build\ttgo-t1\libc19\ESP8266Audio\AudioOutputFilterBiquad.cpp.o
Compiling .pio\build\ttgo-t1\libc19\ESP8266Audio\AudioOutputFilterDecimate.cpp.o
Compiling .pio\build\ttgo-t1\libc19\ESP8266Audio\AudioOutputI2S.cpp.o
Compiling .pio\build\ttgo-t1\libc19\ESP8266Audio\AudioOutputI2SNoDAC.cpp.o
Compiling .pio\build\ttgo-t1\libc19\ESP8266Audio\AudioOutputMixer.cpp.o
Compiling .pio\build\ttgo-t1\libc19\ESP8266Audio\AudioOutputSPDIF.cpp.o
Compiling .pio\build\ttgo-t1\libc19\ESP8266Audio\AudioOutputSPIFFSWAV.cpp.o
Compiling .pio\build\ttgo-t1\libc19\ESP8266Audio\AudioOutputSTDIO.cpp.o
Compiling .pio\build\ttgo-t1\libc19\ESP8266Audio\AudioOutputSerialWAV.cpp.o
Compiling .pio\build\ttgo-t1\libc19\ESP8266Audio\libflac\crc.c.o
Compiling .pio\build\ttgo-t1\libc19\ESP8266Audio\libflac\fixed.c.o
Compiling .pio\build\ttgo-t1\libc19\ESP8266Audio\libflac\float.c.o
Compiling .pio\build\ttgo-t1\libc19\ESP8266Audio\libflac\format.c.o
Compiling .pio\build\ttgo-t1\libc19\ESP8266Audio\libflac\lpc.c.o
Compiling .pio\build\ttgo-t1\libc19\ESP8266Audio\libflac\md5.c.o
Compiling .pio\build\ttgo-t1\libc19\ESP8266Audio\libflac\memory.c.o
Compiling .pio\build\ttgo-t1\libc19\ESP8266Audio\libflac\stream_decoder.c.o
Compiling .pio\build\ttgo-t1\libc19\ESP8266Audio\libflac\window.c.o
etc.

earlephilhower, 2021-04-11:
Those objects won't be in your final executable when it links since they are not called by your code.

The root cause is PIO(and Arduino) not understanding the specific app dependencies and just building everything in the ESP8266 Audio directory.  This is 100% safe and correct, just consumes some time on the build process (but it should be cached after the first build anyway).

Feel free to submit a PR changing this, but for now there's nothing to do and this isn't an issue in the final executable.

closed 2021-04-11 by earlephilhower

FrankHovis, 2021-04-11:
Hmm, ok. It seems odd though that a sketch to play a mp3 complies to

> RAM:   [=         ]  11.9% (used 38868 bytes from 327680 bytes)
> Flash: [======    ]  55.6% (used 728838 bytes from 1310720 bytes)

yet changing the mp3 to WAV (etc.) compiles to nearly the same size. 

> RAM:   [=         ]  11.9% (used 38860 bytes from 327680 bytes)
> Flash: [======    ]  55.6% (used 728594 bytes from 1310720 bytes)
> Building .pio\build\ttgo-t1\firmware.bin

earlephilhower, 2021-04-11:
Check the .MAP file to see what's being includes.  That will show what functions got linked where in flash.

If the FLAC functions are showing up in an MP3 sketch, then PIO is not properly building w/ function-sections and gc-sections (as is done in the Arduino IDE).