earlephilhower/ESP8266Audio #389: Pinout for Pico + MAX98357A ?

obdevel, 2021-04-19:
I have a Pico and an Adafruit MAX98357A module. Using Earle's Arduino core and the current ESP8266Audio code.

Pipeline is (based on the mixer example):

LittleFS file -> ID3 -> MP3 -> stub -> mixer -> out

I have connected pin 26 to BLCK and pin 28 to DIN, but it's not clear which pin I should use for LRC. It basically works without LRC connected but the sound is very loud and distorted, and SetGain has no effect (either mixer or stub).

Secondly, mp3->loop() blocks for the entirety of the playback. I would eventually like to mix/interleave multiple MP3 streams. I have tried with and without a preallocated buffer (in the AudioGeneratorMP3 constructor).

I appreciate this is a bleeding edge combination, but I'd like to get it working.

Thanks.

earlephilhower, 2021-04-19:
The Pico I2S has pin 27 as LRCLK.  It's always pin(BCLK+1) because that's the way the PIO special hardware is programmed to run it.  Good luck!

closed 2021-04-19 by earlephilhower

obdevel, 2021-04-20:
Thanks Earle.

It is working now with good sound quality, so I have a couple of other questions, if I may ?

I'd like to use the mixer class to mix/interleave multiple MP3 sources. However, at the moment the code blocks in mp3->loop() for the entirety of the MP3 file.

My pipeline is: littlefs -> mp3 -> mixer -> i2s

How does the mixer manage multiple stubs ? Does it round-robin between the stubs and interleave them on the output, or does it combine/mix the active stubs ?

Do I need to adjust the size of the sample buffer(s) ? And if so, which for object(s) ? The mixer, stub, generator, sink ?

My plan is to combine the processor and DAC/amp on a single PCB, using the Sparkfun and Adafruit designs as 'inspriation'.

Thank you, and greetings from Cambridge, UK, the home of Arm and Raspberry Pi :)

Duncan
m: +44 07449 603 906
e: ***@***.***

earlephilhower, 2021-04-21:
Glad to hear it's running!

There's a mixer example you can reference for more usage info.  It's a very basic setup that blocks and adds all outputs together, not something like you'd see on a PC where it would not block and be driven on a timebase.  I'd recommend not using it unless there's no other way, simply because it is something of a pain to get right.

obdevel, 2021-04-22:
Thanks again Earle.

A relevant discussion has started on GitHub so let's see where that goes. With a minor change to the generator loop method, the MixerSample works well for WAV sources, but currently hangs when changed to MP3+LittleFS and second stream is started. That's no big deal for me as I'll eventually create an RP2040 board with a 16MB flash chip, using the Sparkfun designs as inspiration.

I'm creating something similar to the model train sound module referenced by your project, so I only need a few short loops and single-shot samples. WAV will be fine.

Will you continue to maintain your Pico core once the Arduino project releases their 'official' core ? It's disappointing that multi-core support is so limited initially, compared to FreeRTOS on the ESP32, but that won't improve even if Arduino use Mbed OS as a foundation, as they have with other their other Arm-based products. I guess someone may port FreeRTOS eventually.

Kind regards,

D

earlephilhower, 2021-04-22:
There's an Arduino.cc core out for the RP2040 already.  It uses the MBED mini-os and looks to me to be a plain port.  Doesn't allow things like multiple serial ports/SPI, and I didn't see it using the PIO for anything when I looked (but I did not dig super-deep).

It's low effort to keep this up and running, and I do want to add in some multiprocessing.  I think for most Arduino users, though, multicore == bad idea, simply because of the number of ways you can get things very wrong.

You can use the PICO SDK today to staret the 2nd core and run stuff there and the core will mostly work (but no flash writes allowed...i.e. FS or EEPROM).

FreeRTOS, from what I read, does support multithreading, but not actualy muilticore.  You can run 2 completely isolated instances, one on each core, but you can't start a thread and move it around between cores.  Maybe MBED has support for this?  Again, multicore is pretty rare in the small embedded microcontroller world so maybe no...