earlephilhower/ESP8266Audio #395: when play google TTS 2 or 3 times, it crashes

lioujj, 2021-05-19:
Hi
When I use AudioFileSourceICYStream to  play google TTS voice 2 or 3 times, it will crash!
I try to increase the buffer size, but it still not works.

TTS code here:
i2sFile = new AudioFileSourceHTTPStream("http://translate.google.com/translate_tts?ie=UTF-8&client=tw-ob&tl=en&q="+YourTextHere);

Thanks!

JJ-Concept, 2021-07-02:
Same issue here. ESP32 completely freezes after playing google tts. It had worked for half a year perfectly fine but the same problem occurs now on two different devices here that use esp8266audio to play google tts. 

Error (several of these): 
STATUS(mp3) '257' = 'Decoding error 'lost synchronization' at byte offset 33265'

Then it shows "Running for xxx ms" two times (the second one appearing after more than a second) and then the ESP32 completely freezes. No further output on serial, no reaction anymore...

Would be cool to get this error fixed as i prefer google tts instead of esp8266sam becuase it talks more clearly and supports more than only english language ;-)

Thanks

schmurtzm, 2021-12-20:
Same problem here : I play one google tts mp3 then it crash immediately at the end of the sound.
I tried : 

- with esp8266 (wemos D1, no dac) 
- with ESP32 (with "dac" and "no dac")
- with "out->SetRate(22050);" 

Always with the same result : it play correctly but no more response from the ESP and no stack trace...
May be a debugger with ESP32 could be useful for better investigation ?

schmurtzm, 2021-12-20:
It is really due to the end of the file because if you stop during the playing of the file there is no crash of the ESP.

schmurtzm, 2021-12-29:
Hi, easy way to reproduce the bug : Take "StreamMP3FromHTTP" example, and replace : 
`const char *URL="http://kvbstreams.dyndns.org:8000/wkvi-am";`
with
`const char *URL="http://translate.google.com/translate_tts?ie=UTF-8&q=bonjour&tl=fr&client=tw-ob&ttsspeed=1";`

[This is a record ](https://dl.dropbox.com/s/obdzuxx1jza8tdc/GoogleTTS_neverEding.mp3)of the sound that I obtain.

The codec used by Google TTS seems classic : 
[image]

@FedericoBusero You have resolved a similar issue with RTTTL recently (thanks for that !) , do you have an idea for this one too ?
Having good and multilingual TTS  on ESP is a killer feature  😉

schmurtzm, 2021-12-29:
Managing Google TTS seems more complex than what I was thinking...
I found this great I2S library which support Google TTS:  [ESP32-audioI2S](https://github.com/schreibfaul1/ESP32-audioI2S)

As you can see [here](https://github.com/schreibfaul1/ESP32-audioI2S/blob/master/src/Audio.cpp#L748) the author has made a complex function to use Google TTS. This could be a good source of inspiration to make a new AudioFileSourceGoogleTTS.h 😅 

Interesting facts : I also tested to play Google TTS on this library without using this specific function, just with the [URL of Google Translate](http://translate.google.com/translate_tts?ie=UTF-8&client=tw-ob&tl=en&q=bonjour) and the result was similar to ESP8266Audio : at the end of the play there is a buffer problem. It doesn't hang the ESP on this library but it means that Google TTS send the mp3 file in a particular way which require more work than a classic mp3 stream.