earlephilhower/ESP8266Audio #407: MIDI LittleFS

Nyegnav, 2021-07-14:
Apologies if I'm being a dumb dumb, but I can't seem to get the ESP8266Audio PlayMidifromLittleFS example to work;
Running on D1 Mini 4mb.
Monitor returns the following error;

Panic lfs.c:5121 lfs_file_read: Assertion 'lfs_mlist_isopen(lfs->mlist, (struct lfs_mlist*)file)' failed.

joelwetzel, 2021-07-23:
My project had been working fine for a long time, but just recently I started getting this exact error.  Instead of MIDI, I'm using AudioFileSourceLittleFS and AudioGeneratorWAV.  Weirdly, it works fine if I play one wav file at a time, but the crash happens the moment I try to load a second wav file.  (I'm putting them through an AudioOutputMixer.)  This was all working until just recently.

joelwetzel, 2021-07-23:
Well, I think I found the root of the problem, though I only have a workaround, not a true fix.  From the release notes, it looks like the Espressif8266 framework version 3.0, which was released about a month ago, included some changes to how LittleFS works.  I haven't found detailed documentation on the changes, but they appear to break LFS usage in this ESP8266Audio library.  I was able to get my board running again by downgrading to an older version of the Espressif8266 framework.  I'm using PlatformIO, so in my platform.ini file, I changed the platform line to be like:

platform = espressif8266@^2.6.3

After doing that, my project works again.

@earlephilhower

earlephilhower, 2021-07-23:
Thanks for the report.  This looks to be a core issue and PR is in review for the 3.0.2 bugfix release.

Nyegnav, 2021-07-24:
Much thanks. New to all this, and have been slamming my head against a wall trying to see what I was doing wrong.

closed 2021-09-01 by earlephilhower