earlephilhower/ESP8266Audio #412: ESP32 and DAC1334

alexgithub56, 2021-07-19:
I like to connect ESP32 and DAC1334. My hardware:
- WEMOS TTGO ESP32 WiFi Bluetooth 0.96 inch OLED development CP2102 board
- Audio Stereo Decoder CJMCU-1334 I2S DAC module
DAC  - ESP32
DIN   - 22
WSEL - 25
BCLK - 26
Anyone can guide me?
Thank for advance.

alexgithub56, 2021-07-23:
update my hardware:
ESP32
CJMCU-1334
wiring:
DAC - ESP32
DIN - 22
WSEL - 25
BCLK - 26
code:
#include <Arduino.h>
#include <WiFi.h>
#include <SPIFFS.h>
#include <AudioFileSourceSPIFFS.h>
#include <AudioGeneratorMP3.h>
#include <AudioOutputI2S.h>

AudioGeneratorMP3 *mp3;
AudioFileSourceSPIFFS *file;
AudioOutputI2S *out;

void setup() {
  WiFi.mode(WIFI_OFF);
  Serial.begin(115200);
  delay(1000);
  SPIFFS.begin();
  file = new AudioFileSourceSPIFFS("/pno-cs.mp3");
  out = new AudioOutputI2S();
  mp3 = new AudioGeneratorMP3();
  mp3->begin(file, out);
}

void loop() {
  if (mp3->isRunning()) {
    if (!mp3->loop()) mp3->stop(); 
  } else {
    Serial.printf("MP3 done\n");
    delay(1000);
  }
}
result:
no sound

alexgithub56, 2021-08-08:
๋Just change to new library.

closed 2021-08-08 by alexgithub56

noseeflower, 2024-10-31:
> update my hardware: ESP32 CJMCU-1334 wiring: DAC - ESP32 DIN - 22 WSEL - 25 BCLK - 26 code: #include <Arduino.h> #include <WiFi.h> #include <SPIFFS.h> #include <AudioFileSourceSPIFFS.h> #include [...]

I have the same code, could you please tell me how you solved the problem?