earlephilhower/ESP8266Audio #416: ESP32 DAC

medvosa, 2021-07-28:
How do I play sound using ESP32 internal DAC? How to configure AudioOutput? Which AudioOutputClass to use?
Thanks in advance

s-hadinger, 2021-08-02:
It's already written in the documentation 

To use the internal DAC on ESP32, instantiate this class as AudioOutputI2S(0,1), see example PlayMODFromPROGMEMToDAC and code in AudioOutputI2S.cpp for details.

ahmednaser89, 2021-08-17:
/home/hhh/snap/arduino/current/Arduino/libraries/ESP8266Audio-master/src/AudioFileSourceICYStream.cpp:23: warning: "_GNU_SOURCE" redefined
 #define _GNU_SOURCE
 
<command-line>: note: this is the location of the previous definition
/home/hhh/snap/arduino/current/Arduino/libraries/ESP8266Audio-master/src/AudioOutputI2SNoDAC.cpp: In member function 'virtual bool AudioOutputI2SNoDAC::ConsumeSample(int16_t*)':
/home/hhh/snap/arduino/current/Arduino/libraries/ESP8266Audio-master/src/AudioOutputI2SNoDAC.cpp:98:8: error: 'i2s_write_bytes' was not declared in this scope
   if (!i2s_write_bytes((i2s_port_t)portNo, (const char *)dsBuff, sizeof(uint32_t) * (oversample/32), 0))
        ^~~~~~~~~~~~~~~
/home/hhh/snap/arduino/current/Arduino/libraries/ESP8266Audio-master/src/AudioOutputI2S.cpp: In member function 'bool AudioOutputI2S::begin(bool)':
/home/hhh/snap/arduino/current/Arduino/libraries/ESP8266Audio-master/src/AudioOutputI2S.cpp:176:56: warning: 'I2S_COMM_FORMAT_I2S' is deprecated [-Wdeprecated-declarations]
       i2s_comm_format_t comm_fmt = (i2s_comm_format_t)(I2S_COMM_FORMAT_I2S | I2S_COMM_FORMAT_I2S_MSB);
                                                        ^~~~~~~~~~~~~~~~~~~
In file included from /home/hhh/snap/arduino/61/.arduino15/packages/esp32/hardware/esp32/2.0.0-rc1/tools/sdk/esp32/include/hal/esp32/include/hal/i2s_ll.h:30,
                 from /home/hhh/snap/arduino/61/.arduino15/packages/esp32/hardware/esp32/2.0.0-rc1/tools/sdk/esp32/include/hal/include/hal/i2s_hal.h:28,
                 from /home/hhh/snap/arduino/61/.arduino15/packages/esp32/hardware/esp32/2.0.0-rc1/tools/sdk/esp32/include/driver/include/driver/i2s.h:16,
                 from /home/hhh/snap/arduino/current/Arduino/libraries/ESP8266Audio-master/src/AudioOutputI2S.cpp:23:
/home/hhh/snap/arduino/61/.arduino15/packages/esp32/hardware/esp32/2.0.0-rc1/tools/sdk/esp32/include/hal/include/hal/i2s_types.h:70:5: note: declared here
     I2S_COMM_FORMAT_I2S       __attribute__((deprecated)) = 0x01, /*!< I2S communication format I2S, correspond to `I2S_COMM_FORMAT_STAND_I2S`*/
     ^~~~~~~~~~~~~~~~~~~
/home/hhh/snap/arduino/current/Arduino/libraries/ESP8266Audio-master/src/AudioOutputI2S.cpp:176:78: warning: 'I2S_COMM_FORMAT_I2S_MSB' is deprecated [-Wdeprecated-declarations]
       i2s_comm_format_t comm_fmt = (i2s_comm_format_t)(I2S_COMM_FORMAT_I2S | I2S_COMM_FORMAT_I2S_MSB);
                                                                              ^~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/hhh/snap/arduino/61/.arduino15/packages/esp32/hardware/esp32/2.0.0-rc1/tools/sdk/esp32/include/hal/esp32/include/hal/i2s_ll.h:30,
                 from /home/hhh/snap/arduino/61/.arduino15/packages/esp32/hardware/esp32/2.0.0-rc1/tools/sdk/esp32/include/hal/include/hal/i2s_hal.h:28,
                 from /home/hhh/snap/arduino/61/.arduino15/packages/esp32/hardware/esp32/2.0.0-rc1/tools/sdk/esp32/include/driver/include/driver/i2s.h:16,
                 from /home/hhh/snap/arduino/current/Arduino/libraries/ESP8266Audio-master/src/AudioOutputI2S.cpp:23:
/home/hhh/snap/arduino/61/.arduino15/packages/esp32/hardware/esp32/2.0.0-rc1/tools/sdk/esp32/include/hal/include/hal/i2s_types.h:71:5: note: declared here
     I2S_COMM_FORMAT_I2S_MSB   __attribute__((deprecated)) = 0x01, /*!< I2S format MSB, (I2S_COMM_FORMAT_I2S |I2S_COMM_FORMAT_I2S_MSB) correspond to `I2S_COMM_FORMAT_STAND_I2S`*/
     ^~~~~~~~~~~~~~~~~~~~~~~
/home/hhh/snap/arduino/current/Arduino/libraries/ESP8266Audio-master/src/AudioOutputI2S.cpp:179:39: warning: 'I2S_COMM_FORMAT_I2S_MSB' is deprecated [-Wdeprecated-declarations]
         comm_fmt = (i2s_comm_format_t)I2S_COMM_FORMAT_I2S_MSB;
                                       ^~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/hhh/snap/arduino/61/.arduino15/packages/esp32/hardware/esp32/2.0.0-rc1/tools/sdk/esp32/include/hal/esp32/include/hal/i2s_ll.h:30,
                 from /home/hhh/snap/arduino/61/.arduino15/packages/esp32/hardware/esp32/2.0.0-rc1/tools/sdk/esp32/include/hal/include/hal/i2s_hal.h:28,
                 from /home/hhh/snap/arduino/61/.arduino15/packages/esp32/hardware/esp32/2.0.0-rc1/tools/sdk/esp32/include/driver/include/driver/i2s.h:16,
                 from /home/hhh/snap/arduino/current/Arduino/libraries/ESP8266Audio-master/src/AudioOutputI2S.cpp:23:
/home/hhh/snap/arduino/61/.arduino15/packages/esp32/hardware/esp32/2.0.0-rc1/tools/sdk/esp32/include/hal/include/hal/i2s_types.h:71:5: note: declared here
     I2S_COMM_FORMAT_I2S_MSB   __attribute__((deprecated)) = 0x01, /*!< I2S format MSB, (I2S_COMM_FORMAT_I2S |I2S_COMM_FORMAT_I2S_MSB) correspond to `I2S_COMM_FORMAT_STAND_I2S`*/
     ^~~~~~~~~~~~~~~~~~~~~~~
/home/hhh/snap/arduino/current/Arduino/libraries/ESP8266Audio-master/src/AudioOutputI2S.cpp: In member function 'virtual bool AudioOutputI2S::ConsumeSample(int16_t*)':
/home/hhh/snap/arduino/current/Arduino/libraries/ESP8266Audio-master/src/AudioOutputI2S.cpp:272:12: error: 'i2s_write_bytes' was not declared in this scope
     return i2s_write_bytes((i2s_port_t)portNo, (const char *)&s32, sizeof(uint32_t), 0);
            ^~~~~~~~~~~~~~~
/home/hhh/snap/arduino/current/Arduino/libraries/ESP8266Audio-master/src/AudioOutputI2S.cpp:272:12: note: suggested alternative: 'i2s_write_expand'
     return i2s_write_bytes((i2s_port_t)portNo, (const char *)&s32, sizeof(uint32_t), 0);
            ^~~~~~~~~~~~~~~
            i2s_write_expand
[... cut ...]
     I2S_COMM_FORMAT_I2S_MSB   __attribute__((deprecated)) = 0x01, /*!< I2S format MSB, (I2S_COMM_FORMAT_I2S |I2S_COMM_FORMAT_I2S_MSB) correspond to `I2S_COMM_FORMAT_STAND_I2S`*/
     ^~~~~~~~~~~~~~~~~~~~~~~
Multiple libraries were found for "WiFi.h"
 Used: /home/hhh/snap/arduino/61/.arduino15/packages/esp32/hardware/esp32/2.0.0-rc1/libraries/WiFi
 Not used: /snap/arduino/61/libraries/WiFi
Multiple libraries were found for "SD.h"
 Used: /home/hhh/snap/arduino/61/.arduino15/packages/esp32/hardware/esp32/2.0.0-rc1/libraries/SD
 Not used: /snap/arduino/61/libraries/SD
exit status 1
Error compiling for board ESP32 Dev Module.

ahmednaser89, 2021-08-17:
#include <Arduino.h>
#include "AudioFileSourcePROGMEM.h"
#include "AudioGeneratorMOD.h"
#include "AudioOutputI2S.h"
#ifdef ESP32
    #include <WiFi.h>
#else
    #include <ESP8266WiFi.h>
#endif

// enigma.mod sample from the mod archive: https://modarchive.org/index.php?request=view_by_moduleid&query=42146
#include "enigma.h"

AudioGeneratorMOD *mod;
AudioFileSourcePROGMEM *file;
AudioOutputI2S *out;

void setup()
{
  WiFi.mode(WIFI_OFF); //WiFi.forceSleepBegin();
  Serial.begin(115200);
  delay(1000);

  audioLogger = &Serial;
  file = new AudioFileSourcePROGMEM( enigma_mod, sizeof(enigma_mod) );
  out = new AudioOutputI2S(0, 1); //Uncomment this line, comment the next one to use the internal DAC channel 1 (pin25) on ESP32
 // out = new AudioOutputI2S();
  mod = new AudioGeneratorMOD();
  mod->SetBufferSize(3*1024);
  mod->SetSampleRate(44100);
  mod->SetStereoSeparation(32);
  mod->begin(file, out);
}

void loop()
{
  if (mod->isRunning()) {
    if (!mod->loop()) mod->stop();
  } else {
    Serial.printf("MOD done\n");
    delay(1000);
  }
}

trullock, 2021-08-24:
Whats missing from the docs is what pins to use, which are:

```
  bclkPin = 26;
  wclkPin = 25;
  doutPin = 22;
```

sudarshann, 2021-09-09:
@trullock Could you kindly explain where and how to define or initilise these pins

Thanks

trullock, 2021-09-10:
@sudarshann These are already defined (hard coded) inside this library, see here https://github.com/earlephilhower/ESP8266Audio/blob/9dd4f24056821a2591f825c743d152bab862984a/src/AudioOutputI2S.cpp#L46

sudarshann, 2021-09-10:
Thanks 
I managed to set something like this for M5stack Atom Echo

out = new AudioOutputI2S();
out->SetPinout(19, 33, 22);

s-hadinger, 2022-01-07:
M5stack Atom Echo has an external I2S chip and does not use ESP32 internal DAC

CataCluj, 2023-07-19:
Can the I2S pins be changed to other pins without changing the library?
Usually I would do something like:
   Class_instance.bclkPin =0;
But in the code all I see this:
   AudioOutputI2S  *out = NULL;     // so I don't think there is an instance here
   out = new AudioOutputI2S();       // I think after this the pins in the library are already set