earlephilhower/ESP8266Audio #417: I2S.h no such file or directory

SCratORS, 2021-08-01:
When using ESPHome (PlatformIO), the project is not compiled. The I2S file was not found.h, If you use version 1.9.1, then it's fine.
It uses the i2s.h library.

zhujisheng, 2021-08-05:
Same here.
```
/data/8266audio/.piolibdeps/8266audio/ESP8266Audio/src/AudioOutputI2S.cpp:25:19: fatal error: I2S.h: No such file or directory

*************************************************************
* Looking for I2S.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:I2S.h"
* Web  > https://platformio.org/lib/search?query=header:I2S.h
*
*************************************************************

   #include <I2S.h>
                   ^
compilation terminated.
*** [/data/8266audio/.pioenvs/8266audio/lib6fe/ESP8266Audio/AudioOutputI2S.cpp.o] Error 1
/data/8266audio/.piolibdeps/8266audio/ESP8266Audio/src/AudioOutputI2SNoDAC.cpp:25:19: fatal error: I2S.h: No such file or directory

*************************************************************
* Looking for I2S.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:I2S.h"
* Web  > https://platformio.org/lib/search?query=header:I2S.h
*
*************************************************************

   #include <I2S.h>
                   ^
compilation terminated.
*** [/data/8266audio/.pioenvs/8266audio/lib6fe/ESP8266Audio/AudioOutputI2SNoDAC.cpp.o] Error 1
```

earlephilhower, 2021-08-18:
Please update to the 3.0.x release of the ESP8266 core.  There were large changes required and the `I2S` class was introduced (as opposed to the `i2s_xxxx` ESP8266-specific APIs.

closed 2021-08-18 by earlephilhower

juancgalvez, 2022-05-06:
I have ESP8266 core version 3.0.4 installed and getting the same error:

`fatal error: i2s.h: No such file or directory`