earlephilhower/ESP8266Audio #418: Duration / Elapsed

gilphilbert, 2021-08-03:
First of all, awesome library!

I do have one question, how can I get the duration of the current track? Ideally, the elapsed as well (I'm guessing this would be more accurate than using `millis()` to track). I'd like to be able to display the current playback position. I've had a good hunt but I can't find any documentation/examples of this.

Thanks!

earlephilhower, 2021-08-03:
Duration isn't part of the libMAD interface, sorry.  Your best bet would be to guesstimate given the bitrate and the file position, but it's not going to be 100% accurate.  For current playback position, you'll need to write your own little `AudioOutputTimer` which will eat every I2S byte and can use the # of bytes and the output frequency to get the exact offset from start.

Alternatively, like you suggested, millis is a possibility.  As you'd just need 1-second accuracy I think that's not a bad way to go, actually.

Good luck!

gilphilbert, 2021-08-04:
Ah, shame.

Thanks for the hints, I'll see what I can do!

closed 2021-08-04 by gilphilbert