earlephilhower/ESP8266Audio #419: Using both I2S ports on ESP32

AyrtonEstrella, 2021-08-08:
Hey everyone, I'm trying to implement 2 external I2S DACs to play multiple files.
I tested each I2S port individually, and they work if using them one at a time. But when I try to use both together, the playback gets ruined.
Is it possible to play sound to the 2 I2S ports simultaneously? This could be the same file or even 2 separate files.
Thanks!
Here is the code:

```
File dir;
AudioFileSourceSD *source0 = NULL;
AudioFileSourceSD *source1 = NULL;
AudioOutputI2S *i2s0 = NULL;
AudioOutputI2S *i2s1 = NULL;
AudioGeneratorWAV *decoder0 = NULL;
AudioGeneratorWAV *decoder1 = NULL;

void setup() {
  Serial.begin(115200);
  delay(1000);
  Serial.println();

  audioLogger = &Serial;  
  source0 = new AudioFileSourceSD();
  source1 = new AudioFileSourceSD();
  i2s0 = new AudioOutputI2S(I2S0_NUM, i2s0->EXTERNAL_I2S, 8, i2s0->APLL_DISABLE);
  i2s0->SetPinout(I2S0_BCLK, I2S0_WSEL, I2S0_DOUT);
  i2s0->SetGain(0.5);

  i2s1 = new AudioOutputI2S(I2S1_NUM, i2s1->EXTERNAL_I2S, 8, i2s1->APLL_DISABLE);
  i2s1->SetPinout(I2S1_BCLK, I2S1_WSEL, I2S1_DOUT);
  i2s1->SetGain(0.5);

  decoder0 = new AudioGeneratorWAV();
  decoder1 = new AudioGeneratorWAV();

  SD.begin();
  dir = SD.open("/"); 
}

void loop() {
  // I2S0 Playback
  if ((decoder0) && (decoder0->isRunning())) {
    if (!decoder0->loop())
      decoder0->stop();
  }
  else {
    File file = dir.openNextFile();
    if (file) {
      if (String(file.name()).endsWith("wavfile0.wav")) {
        source0->close();
        if (source0->open(file.name())) {
          Serial.printf_P(PSTR("Playing '%s' from SD card...\n"), file.name());
          decoder0->begin(source0, i2s0);
        }
        else {
          Serial.printf_P(PSTR("Error opening '%s'\n"), file.name());
        }
      }
    }
    else {
      Serial.println(F("Playback form SD card done"));
      delay(1000);
    } 
  }

  // I2S1 Playback
  if ((decoder1) && (decoder1->isRunning())) {
    if (!decoder1->loop())
      decoder1->stop();
  }
  else {
    File file1 = dir.openNextFile();
    if (file1) {
      if (String(file1.name()).endsWith("wavfile0.wav")) {
        source1->close();
        if (source1->open(file1.name())) {
          Serial.printf_P(PSTR("Playing '%s' from SD card...\n"), file1.name());
          decoder1->begin(source1, i2s1);
        }
        else {
          Serial.printf_P(PSTR("Error opening '%s'\n"), file1.name());
        }
      }
    }
    else {
      Serial.println(F("Playback form SD card done"));
      delay(1000);
    } 
  }
}
```

medvosa, 2021-08-13:
I finally got it working.
sound on pins 25 and 26

AyrtonEstrella, 2021-08-13:
Hi @medvosa , were you able to play audio on both I2S ports simultaneously?

AyrtonEstrella, 2021-08-13:
Hey everyone, just a quick update: I managed to play the same file in both I2S ports. I'm not sure if this is the most efficient method, but it works.

```
#include <Arduino.h>
#include "Pinout.h"
#include "SDCard.h"
#include "I2S_Functions.h"

#include "AudioFileSourceSD.h"
#include "AudioGeneratorWAV.h"
#include "AudioOutputI2S.h"

#define SPI_SPEED SD_SCK_MHZ(40)
#define WAV0_FILENAME "/wavfile0_22k.wav"
#define WAV1_FILENAME "/wavfile1_22k.wav"

#define OUTPUT_0      0
#define OUTPUT_1      1

File dir;
File file[2];
AudioFileSourceSD *source[2];
AudioOutputI2S *i2s[2];
AudioGeneratorWAV *decoder[2];

void initializeOutputs() {
  i2s[0] = new AudioOutputI2S(I2S0_NUM, i2s[0]->EXTERNAL_I2S, 8, i2s[0]->APLL_DISABLE);
  i2s[0]->SetPinout(I2S0_BCLK, I2S0_WSEL, I2S0_DOUT);
  i2s[0]->SetGain(0.5);

  i2s[1] = new AudioOutputI2S(I2S1_NUM, i2s[1]->EXTERNAL_I2S, 8, i2s[1]->APLL_DISABLE);
  i2s[1]->SetPinout(I2S1_BCLK, I2S1_WSEL, I2S1_DOUT);
  i2s[1]->SetGain(0.5);
}

void play(uint8_t output, const char *filename) {
  if (output == OUTPUT_0) {
    Serial.println("Playing " + String(filename) + " on output " + String(output));
    source[0]->open(filename);
    decoder[0]->begin(source[0], i2s[0]);
  }
  else if (output == OUTPUT_1) {
    Serial.println("Playing " + String(filename) + " on output " + String(output));
    source[1]->open(filename);
    decoder[1]->begin(source[1], i2s[1]);
  }
}

void setup() {
  Serial.begin(115200);
  delay(1000);
  Serial.println();

  audioLogger = &Serial;  
  source[0] = new AudioFileSourceSD();
  source[1] = new AudioFileSourceSD();
  initializeOutputs();

  decoder[0] = new AudioGeneratorWAV();
  decoder[1] = new AudioGeneratorWAV();

  SD.begin();
  dir = SD.open("/"); 

  play(OUTPUT_0, WAV0_FILENAME);
  play(OUTPUT_1, WAV1_FILENAME);
}

void loop() {
  float gain = WAVPLAYER_READ_VOLUME_FACTOR();
  // I2S0 Playback
  i2s[0]->SetGain(gain);
  if ((decoder[0]) && (decoder[0]->isRunning())) {
    if (!decoder[0]->loop())
      decoder[0]->stop();
  }

  // I2S1 Playback
  i2s[1]->SetGain(gain);
  if ((decoder[1]) && (decoder[1]->isRunning())) {
    if (!decoder[1]->loop())
      decoder[1]->stop();
  }
}
```

closed 2021-08-13 by AyrtonEstrella