earlephilhower/ESP8266Audio #420: Moving from esp8266 to esp32

medvosa, 2021-08-08:
Hi!
I have the code that is pretty wprking on Esp8266.
```#include <Arduino.h>
#include "AudioFileSourceSD.h"
#include "AudioOutputSPDIF.h"
#include "AudioGeneratorMP3.h"
#include "AudioOutputI2SNoDAC.h"
#include "AudioOutputI2S.h"
#include <SD.h>

// For this sketch, you need connected SD card with '.MP3' music files in the root
// directory. Some samples with various sampling rates are available from i.e. 
// Espressif Audio Development Framework at:
// https://docs.espressif.com/projects/esp-adf/en/latest/design-guide/audio-samples.html
//
// On ESP8266 you might need to reencode MP3 files with max '-2' compression level 
// (i.e. 1152 maximum block size) or you will run out of memory. MP3 files will be 
// slightly bigger but you don't loose audio quality with reencoding (lossles codec).

// You may need a fast SD card. Set this as high as it will work (40MHz max).
#define SPI_SPEED SD_SCK_MHZ(40)

// On ESP32 you can adjust the SPDIF_OUT_PIN (GPIO number). 
// On ESP8266 it is fixed to GPIO3/RX0 and this setting has no effect
#define SPDIF_OUT_PIN 27

File dir;
AudioFileSourceSD *source = NULL;
AudioOutputSPDIF *output = NULL;
AudioGeneratorMP3 *decoder = NULL;

AudioOutputI2SNoDAC *out;

void setup() {
  Serial.begin(115200);
  Serial.println();
  delay(1000);

  audioLogger = &Serial;  
  source = new AudioFileSourceSD();
//  output = new AudioOutputSPDIF(SPDIF_OUT_PIN);
  out = new AudioOutputI2SNoDAC();
//  out = new AudioOutputI2SNoDAC();
  decoder = new AudioGeneratorMP3();

  // NOTE: SD.begin(...) should be called AFTER AudioOutputSPDIF() 
  //       to takover the the SPI pins if they share some with I2S
  //       (i.e. D8 on Wemos D1 mini is both I2S BCK and SPI SS)
  #if defined(ESP8266)
    SD.begin(SS, SPI_SPEED);
  #else
    SD.begin();
  #endif
  dir = SD.open("/"); 
}

void loop() {
  if ((decoder) && (decoder->isRunning())) {
    if (!decoder->loop()) decoder->stop();
  } else {
    File file = dir.openNextFile();
    if (file) {  
      Serial.println("Playing "+String(file.name()));    
      if (String(file.name()).endsWith(".mp3")) {
        source->close();
        if (source->open(file.name())) { 
          Serial.printf_P(PSTR("Playing '%s' from SD card...\n"), file.name());
//          decoder->begin(source, output);
          decoder->begin(source, out);
        } else {
          Serial.printf_P(PSTR("Error opening '%s'\n"), file.name());
        }
      } 
    } else {
      Serial.println(F("Playback form SD card done\n"));
      delay(1000);
    }       
  }
}```

How Can I port it to ESP32.
I tried running it wothout any changes. And I only see logs in com port monitor
![image](https://user-images.githubusercontent.com/38169798/128631549-2da844f5-ce01-4d3e-b46b-ec421537e5ab.png)
So, I see that SD card is working correctly. On my esp8266 I can hear the sound between RX and G. But on ESP32 I cannot hear it from RX2 or RX0 (I don't ree RX1)

I also tried using AudioOutputI2S but also couldn't hear sound from any pin. Thanks in advance.

schmurtzm, 2021-12-16:
Hi, 

To use internal DAC of the ESP32 you should use "out = new AudioOutputI2S(0, 1);" as recommended in readme.

What kind of ESP32 board do you use ?

I figure that there is different behavior depending the ESP32 board used : 
- I have tested on "wemos D1 mini 32" : I have very bad sound with internal DAC. It works only if you set "CPU frequency" to 80MHZ in arduino IDE.
- I have tested "doit esp32 devkit v4" with better result on the sound quality (similar to esp8266 nodac I think) from the internal DAC. I let the default settings in arduino IDE flash options for this board.

For both when you use the internal DAC you can put your speaker between ping 25 and GND.

I wasn't able to use "no DAC" option. It works on my esp8266 but not on my ESP32, I have only crackling sounds. For what I understand that's a shame because the sound quality could be better comparing to the internal DAC.

FedericoBusero, 2021-12-18:
NoDac works on Esp32, it is by default on pin 22

tkim61, 2022-01-12:
Also working with esp32.   Everything is wired up to aproprietary PCB that I'm stuck with, so I can only output using pin 25 and the DAC.  I mainly want to confirm that "flash frequency 80MHz" is ok for the speed, since I don't see an obvious place to set 
```
Tools->lwIP Variant->v1.4 Open Source, or V2 Higher Bandwidth
Tools->CPU Frequency->160MHz
```
as is suggested in the readme.  It may be the case that there's some loose wiring/soldering on my board or that my speaker is just blown out or not compatible.  I've run "PlayMODFromPROGMEMtoDAC" with no audible output, though the monitor indicates something is being sent and played.