earlephilhower/ESP8266Audio #426: ESP32 RADIO Issue with BBC stations

lidense, 2021-08-31:
hi,
i'm making a small internet radio with my M5Stack Core 2 using the information from this site:
https://community.m5stack.com/topic/2972/core2-as-web-radio-without-extra-hardware/8
It works well except for BBC radio stations where there are a lot of artefacts: interruptions, robotic noises etc.
Initially I thought that it might be a buffer issue and I tried to increase or decrease the buffer with no success.
I heard that there might be a problem with BBC stations and the ICY protoco l(as described here:   "https://www.youtube.com/watch?v=lYyvNDQwm0I"), they mention that this issue can be solved by modifying  "GET HTTP1.0" instead of "GET HTTP1.1" but I don't see where this line could be in the ESP8266Audio files.
Thanks for your help.
Regards
Dave

lidense, 2021-10-08:
> Hi
> Try these for BBC World Service
> http://stream.live.vc.bbcmedia.co.uk/bbc_world_service
> http://vprbbc.streamguys.net/vprbbc24.mp3
> http://16813.live.streamtheworld.com:80/WRVOHD3_SC
>
> I can clearly listen BBC with an ESP32 and his Internal DAC

Thanks for your email:
2nd and 3rd streams work clearly while the first is not good (robotic noises).
I wonder if it might have something to do with the header.

when I do curl -vvv http://vprbbc.streamguys.net/vprbbc24.mp3 or http://16813.live.streamtheworld.com:80/WRVOHD3_SC
I get HTTP/1.0 200 OK while with curl -vvv http://stream.live.vc.bbcmedia.co.uk/bbc_world_service I get HTTP/1.1 200 OK
and the field "icy-genre" is missing in the header.

Could that be the issue? How to solve it as I have other streams that I like that give HTTP/1.1 200 OK.

Thanks and regards

David

----- Mail original -----

d-a-v, 2021-10-08:
HTTP1.1 rings a bell. It might be related with #363.
You can try with `audioSourse->useHTTP10()`.

d-a-v, 2021-10-08:
> I wonder what was the issue with http1.1  ?!

I struggled for some time before discovering that some streaming server (probably very old or stable so not updated) are interleaving non standard data in their HTTP/1.1 stream. When specifically telling them that only HTTP/1.0 is allowed, then the stream becomes compliant and clean.
I wonder whether forcing HTTP/1.0 should be enabled by default in Esp8266Audio.

lidense, 2021-10-08:
Understood, thanks.

----- Mail original -----

lidense, 2022-05-05:
Hi, I'm trying to implement a small webradio with my M5Stack.
One of the streams that I am interested in is an Italian music radio which uses a m3u8 stream, ie it sends 
.ts (mpeg2-TS ??) streams of 10 seconds each. I think I can easily send these to a buffer but then how to read these .ts files? Can ESP8266Audio read them, if not I would have to convert them to .mp3 or AAC, how do I do that? 
Thanks and regards
Dave

lidense, 2022-10-11:
When I open the .ts file with mplayer on my computer this is the info:

Playing segment21425073_1_a-p.ts.
libavformat version 58.29.100 (external)
TS file format detected.
NO VIDEO! AUDIO AAC(pid=256) NO SUBS (yet)!  PROGRAM N. 1
Load subtitles in ./
==========================================================================
Opening audio decoder: [ffmpeg] FFmpeg/libavcodec audio decoders
libavcodec version 58.54.100 (external)
AUDIO: 32000 Hz, 2 ch, floatle, 0.0 kbit/0.00% (ratio: 0->256000)
Selected audio codec: [ffaac] afm: ffmpeg (FFmpeg AAC (MPEG-2/MPEG-4 Audio))

So it seems to be AAC? Would that be compatible with ESP8266Audio?

Thanks

Dave

----- Mail original -----