earlephilhower/ESP8266Audio #427: This library is no longer available on ESP32 Arduino 2.0.0, please fix it as soon as possible.

qnxsgwy, 2021-09-07:
ESP32 Arduino 2.0.0 has been officially released,
Currently the files AudioOutputI2S.cpp and AudioOutputI2SNoDAC.cpp use the function i2s_write_bytes, which has been deprecated, and has been **removed** in the ESP32 Arduino 2.0.0 release!!!

The following error is reported when compiling.
-----------------------------------------------------------------------------------
C:\Users\hnbd\Documents\Arduino\libraries\ESP8266Audio\src\AudioOutputI2SNoDAC.cpp:98:8: error: 'i2s_write_bytes' was not declared in this scope
   if (!i2s_write_bytes((i2s_port_t)portNo, (const char *)dsBuff, sizeof(uint32_t) * (oversample/32), 0))
-----------------------------------------------------------------------------------
'i2s_write_bytes' need to use 'i2s_write' instead. 

And there are still many deprecated functions that need to be replaced. like:
--------------------------------------------------------------------------------------------------------
C:\Users\hnbd\Documents\Arduino\libraries\ESP8266Audio\src\AudioOutputI2S.cpp:179:39: warning: 'I2S_COMM_FORMAT_I2S_MSB' is deprecated [-Wdeprecated-declarations]
         comm_fmt = (i2s_comm_format_t)I2S_COMM_FORMAT_I2S_MSB;
                                       ^~~~~~~~~~~~~~~~~~~~~~~
---------------------------------------------------------------------------------------------------------

please fix it as soon as possible. thanks!

earlephilhower, 2021-09-07:
@qnxsgwy you seem to have the new ESP32 core installed and found a relatively simple to fix issue.  Would you like to try your hand at submitting a PR to fix it?  It should be an almost trivial rename of the function calls and enum vals.

qnxsgwy, 2021-09-08:
@earlephilhower Ok, I've submitted a PR to fix it.
#428

closed 2021-09-08 by earlephilhower