earlephilhower/ESP8266Audio #441: audiologger errors - what's it saying?

hansimglueck, 2021-11-07:
Thanks for this library!
I would like to use it to repeatedly play back an mp3-file from an ESP32 with SPIFFS via  an I2S-DAC.
Fortunately it was easy to get everything running - but:

Even though everything is working as expected (sound is played repeatedly) there are different errors thrown from the audiologger.
Now I am a little bit worried to run into problems later since I don't understand this errors now...

At the end of each playback there is this `MP3:ERROR_BUFLEN 0` - maybe it's the last `mp3->loop()` call at the end of the file?
And from the second playback on, every `mp3->begin(in, out);` throws `E (324297) I2S: register I2S object to platform failed ERROR: Unable to install I2S drives`
 
Can you help me with that?

This is my test-code:
```
#include <Arduino.h>
#include <SPIFFS.h>
#include <SD.h>
#include "AudioGeneratorMP3.h"
#include "AudioOutputI2S.h"
#include "AudioFileSourcePROGMEM.h"
#include "AudioFileSourceSPIFFS.h"

AudioFileSourceSPIFFS *in;
AudioGeneratorMP3 *mp3;
AudioOutputI2S *out;

int tick = 0;

void playAudio()
{
  in->open("/eveninthis.mp3");
  mp3 = new AudioGeneratorMP3();
  mp3->begin(in, out);
}

void setup() {
  Serial.begin(115200);
  SPIFFS.begin();
  audioLogger = &Serial;
  out = new AudioOutputI2S();
  out->SetPinout(33, 26, 25);
  out->SetGain(0.04);
  mp3 = new AudioGeneratorMP3();
  in = new AudioFileSourceSPIFFS("/eveninthis.mp3");  
}

void loop() {
  if (mp3->isRunning())
  {
    if (!mp3->loop())
    {
      Serial.println("stop!");
      mp3->stop();
    }
  }
  if (millis() > tick) {
    Serial.println("play");
    playAudio();
    tick = millis()+12000;
  }
}
```

and this is the serial output:

```
play
+0 0x3ffb2730
MP3:ERROR_BUFLEN 0
stop!
play
+0 0x3ffb2730
E (12166) I2S: register I2S object to platform failed
ERROR: Unable to install I2S drives

MP3:ERROR_BUFLEN 0
stop!
play
+0 0x3ffb2730
E (24172) I2S: register I2S object to platform failed
ERROR: Unable to install I2S drives

MP3:ERROR_BUFLEN 0
stop!
```

JasonWong08, 2022-03-01:
I would like to use it to repeatedly playback an mp3-file from an ESP32( ESP32 Arduino 2.0.2) with SPIFFS via Internal I2S-DAC.
ESP8266Audio(version: 1.9.5)
I modified the tsf.h file to make the compilation successful（https://github.com/arendst/Tasmota/pull/13483/files）
but at the first playback, the sound is played noise,  from the second playback on, every mp3->begin(in, out); throws:
MP3:ERROR_BUFLEN 0
+0 0x3ffb2750
E (20244) I2S: register I2S object to platform failed
ERROR: Unable to install I2S drives
the serial port output :
`
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1324
ho 0 tail 12 room 4
load:0x40078000,len:13508
load:0x40080400,len:3604
entry 0x400805f0
Listing directory: /
  FILE: pno-cs.mp3	SIZE: 321664
Sample MP3 playback begins...
Debug: i2s_comm_format_t: 2 
+0 0x3ffb2750
ID3 callback for: Title = 'Piano Sample'
ID3 callback for: Performer = 'Artist Goes Here'
ID3 callback for: Album = 'Album Goes Here'
ID3 callback for: Year = '2010'
ID3 callback for: eof = 'id3'
MP3:ERROR_BUFLEN 0
Debug: i2s_comm_format_t: 2 
+0 0x3ffb2750
E (20244) I2S: register I2S object to platform failed
ERROR: Unable to install I2S drives

ID3 callback for: Title = 'Piano Sample'
ID3 callback for: Performer = 'Artist Goes Here'
ID3 callback for: Album = 'Album Goes Here'
ID3 callback for: Year = '2010'
ID3 callback for: eof = 'id3'
MP3:ERROR_BUFLEN 0
Debug: i2s_comm_format_t: 2 
+0 0x3ffb2750
E (37493) I2S: register I2S object to platform failed
ERROR: Unable to install I2S drives

ID3 callback for: Title = 'Piano Sample'
ID3 callback for: Performer = 'Artist Goes Here'
ID3 callback for: Album = 'Album Goes Here'
ID3 callback for: Year = '2010'
ID3 callback for: eof = 'id3'
...
`
This is my test code:
`
#include <Arduino.h>
#ifdef ESP32
  #include <WiFi.h>
  #include "SPIFFS.h"
#else
  #include <ESP8266WiFi.h>
#endif
#include "AudioFileSourceSPIFFS.h"
#include "AudioFileSourceID3.h"
#include "AudioGeneratorMP3.h"
#include "AudioOutputI2S.h"

#define FORMAT_SPIFFS_IF_FAILED true

AudioGeneratorMP3 *mp3;
AudioFileSourceSPIFFS *file;
AudioOutputI2S *out;
AudioFileSourceID3 *id3;

void listDir(fs::FS &fs, const char * dirname, uint8_t levels){
    Serial.printf("Listing directory: %s\r\n", dirname);

    File root = fs.open(dirname);
    if(!root){
        Serial.println("- failed to open directory");
        return;
    }
    if(!root.isDirectory()){
        Serial.println(" - not a directory");
        return;
    }

    File file = root.openNextFile();
    while(file){
        if(file.isDirectory()){
            Serial.print("  DIR : ");
            Serial.println(file.name());
            if(levels){
                listDir(fs, file.name(), levels -1);
            }
        } else {
            Serial.print("  FILE: ");
            Serial.print(file.name());
            Serial.print("\tSIZE: ");
            Serial.println(file.size());
        }
        file = root.openNextFile();
    }
}

// Called when a metadata event occurs (i.e. an ID3 tag, an ICY block, etc.
void MDCallback(void *cbData, const char *type, bool isUnicode, const char *string)
{
  (void)cbData;
  Serial.printf("ID3 callback for: %s = '", type);

  if (isUnicode) {
    string += 2;
  }
  
  while (*string) {
    char a = *(string++);
    if (isUnicode) {
      string++;
    }
    Serial.printf("%c", a);
  }
  Serial.printf("'\n");
  Serial.flush();
}

void setup()
{
  WiFi.mode(WIFI_OFF); 
  Serial.begin(115200);
  audioLogger = &Serial;
  delay(1000);
  if(!SPIFFS.begin(FORMAT_SPIFFS_IF_FAILED)){
    Serial.println("SPIFFS Mount Failed");
    return;
  }
  listDir(SPIFFS, "/", 0);
  Serial.printf("Sample MP3 playback begins...\n");
}

void loop()
{
  file = new AudioFileSourceSPIFFS("/pno-cs.mp3"); // get file from SPIFFS
  id3 = new AudioFileSourceID3(file);
  id3->RegisterMetadataCB(MDCallback, (void*)"ID3TAG");
  out = new AudioOutputI2S(0, 1, 8, 0);       // USE Internal DAC
  out->SetGain(0.05);
  mp3 = new AudioGeneratorMP3();

  bool mono = true;
  out->SetOutputModeMono(mono);  // Force mono output no matter the input
  
  mp3->begin(id3, out);
  while(mp3->isRunning()) {
    if (!mp3->loop()){
      mp3->stop();
      break;
    }
  }
  delete file;
  delete id3;
  delete out;
  delete mp3;
}`

Can you help me with that?

JasonWong08, 2022-03-03:
> I would like to use it to repeatedly playback an mp3-file from an ESP32( ESP32 Arduino 2.0.2) with SPIFFS via Internal I2S-DAC. ESP8266Audio(version: 1.9.5) I modified the tsf.h file to make the [...]

By the way this test code is worked fine with  ESP32( ESP32 Arduino 1.0.6) with SPIFFS via Internal I2S-DAC.
ESP8266Audio(version: 1.9.3), for the hardware: GPIO25 connect to a Mono channel amplifier PAM8302 and speaker.

PrinsP75, 2023-03-02:
@hansimglueck hi, did you ever fix this? I'm getting the same messages.

It seems to be stable though. I've been switching between two audiostreams every 10 seconds for more than half an hour now and all is working perfectly, except for the messages in the serial monitor.

jaajuu, 2024-05-27:
Hi, 
just commenting here, that I'm getting the same error, although everything seems to work normally.
I think something has changed in this library or in the ESP core for arduino that's causing this.
I have set of old devices that have been running fine for couple of years (and still are) without these errors. Recently I setup the developing environment again (with ~latest libraries), took my old source code and started developing and i'm getting these errors when I just compiled the old source. But as said, seems to work fine except these messages on the serial monitor.