earlephilhower/ESP8266Audio #442: ESP32-C3 support

sh-user, 2021-11-12:
ESP32-C3
RISC-V processor (RV32IMC)
I am getting compile error, is there a solution to the problem?

`C:\Users\Alex1\Documents\Arduino\libraries\ESP8266Audio\src\AudioOutputULP.cpp:24:10: fatal error: esp32/ulp.h: No such file or directory
 #include <esp32/ulp.h>
          ^~~~~~~~~~~~~
compilation terminated.
`

NoNamedCat, 2021-11-19:
The ESP32-C3 don't have the ULP (Ultra low processor) that is "another core" in the ESP32... Bitluni made the ULP sound driver that is .h header file problen in the compiler log that you posted. I'm not going to tell you how you can solve that but i think that is a glimple for what you can do. :)
Don't use the "AudioOutputULP" with the ESP32-C3 because it does not have an ULP.
(Sorry about my english)

FedericoBusero, 2021-11-28:
Making it compile for ESP32-C3 needs more effort 
- AudioOutputULP.cpp : 
 replace ifdef ESP32 by ifdef CONFIG_ESP32_ULP_COPROC_ENABLED

- uint32_t and size_t is not the same on RISC-V:
 AudioFileSource SPIRAMBuffer.cpp
    uint32_t toReadFromBuffer = std::min(len, (uint32_t)(writePtr - readPtr));

AudioOutputSPDIF.cpp:269:82: error: invalid conversion from 'uint32_t*' {aka 'long unsigned int*'} to 'size_t*' {aka 'unsigned int*'} [-fpermissive]
			   esp_err_t ret = i2s_write((i2s_port_t)portNo, (const char*)&buf, 8 * channels, &bytes_written, 0);

- Also ESP32-C3 lacks an internal DAC, so some code needs to be put in "#ifdef SOC_I2S_SUPPORTS_DAC"

FedericoBusero, 2021-11-28:
Please re-open this issue

sh-user, 2021-11-28:
@FedericoBusero Thank you for your interest in this issue, I will try to run

sh-user, 2021-12-05:
`Arduino 1.8.13` 
Example does not compile `StreamMP3FromHTTP.ino`

```
C:\Users\Alex1\Documents\Arduino\libraries\ESP8266Audio\src\AudioFileSourceICYStream.cpp:23: warning: "_GNU_SOURCE" redefined
 #define _GNU_SOURCE
 
<command-line>: note: this is the location of the previous definition
C:\Users\Alex1\Documents\Arduino\libraries\ESP8266Audio\src\AudioOutputI2S.cpp: In member function 'bool AudioOutputI2S::begin(bool)':
C:\Users\Alex1\Documents\Arduino\libraries\ESP8266Audio\src\AudioOutputI2S.cpp:186:56: warning: 'I2S_COMM_FORMAT_I2S' is deprecated [-Wdeprecated-declarations]
       i2s_comm_format_t comm_fmt = (i2s_comm_format_t)(I2S_COMM_FORMAT_I2S | I2S_COMM_FORMAT_I2S_MSB);
                                                        ^~~~~~~~~~~~~~~~~~~
In file included from C:\Users\Alex1\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.1/tools/sdk/esp32c3/include/driver/include/driver/i2s.h:16,
                 from C:\Users\Alex1\Documents\Arduino\libraries\ESP8266Audio\src\AudioOutputI2S.cpp:23:
C:\Users\Alex1\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.1/tools/sdk/esp32c3/include/hal/include/hal/i2s_types.h:95:5: note: declared here
     I2S_COMM_FORMAT_I2S       __attribute__((deprecated)) = 0x01, /*!< I2S communication format I2S, correspond to `I2S_COMM_FORMAT_STAND_I2S`*/
     ^~~~~~~~~~~~~~~~~~~
C:\Users\Alex1\Documents\Arduino\libraries\ESP8266Audio\src\AudioOutputI2S.cpp:186:78: warning: 'I2S_COMM_FORMAT_I2S_MSB' is deprecated [-Wdeprecated-declarations]
       i2s_comm_format_t comm_fmt = (i2s_comm_format_t)(I2S_COMM_FORMAT_I2S | I2S_COMM_FORMAT_I2S_MSB);
                                                                              ^~~~~~~~~~~~~~~~~~~~~~~
In file included from C:\Users\Alex1\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.1/tools/sdk/esp32c3/include/driver/include/driver/i2s.h:16,
                 from C:\Users\Alex1\Documents\Arduino\libraries\ESP8266Audio\src\AudioOutputI2S.cpp:23:
C:\Users\Alex1\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.1/tools/sdk/esp32c3/include/hal/include/hal/i2s_types.h:96:5: note: declared here
     I2S_COMM_FORMAT_I2S_MSB   __attribute__((deprecated)) = 0x01, /*!< I2S format MSB, (I2S_COMM_FORMAT_I2S |I2S_COMM_FORMAT_I2S_MSB) correspond to `I2S_COMM_FORMAT_STAND_I2S`*/
     ^~~~~~~~~~~~~~~~~~~~~~~
C:\Users\Alex1\Documents\Arduino\libraries\ESP8266Audio\src\AudioOutputI2S.cpp:189:39: warning: 'I2S_COMM_FORMAT_I2S_MSB' is deprecated [-Wdeprecated-declarations]
         comm_fmt = (i2s_comm_format_t)I2S_COMM_FORMAT_I2S_MSB;
                                       ^~~~~~~~~~~~~~~~~~~~~~~
In file included from C:\Users\Alex1\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.1/tools/sdk/esp32c3/include/driver/include/driver/i2s.h:16,
                 from C:\Users\Alex1\Documents\Arduino\libraries\ESP8266Audio\src\AudioOutputI2S.cpp:23:
C:\Users\Alex1\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.1/tools/sdk/esp32c3/include/hal/include/hal/i2s_types.h:96:5: note: declared here
     I2S_COMM_FORMAT_I2S_MSB   __attribute__((deprecated)) = 0x01, /*!< I2S format MSB, (I2S_COMM_FORMAT_I2S |I2S_COMM_FORMAT_I2S_MSB) correspond to `I2S_COMM_FORMAT_STAND_I2S`*/
     ^~~~~~~~~~~~~~~~~~~~~~~
C:\Users\Alex1\Documents\Arduino\libraries\ESP8266Audio\src\AudioOutputSPDIF.cpp: In constructor 'AudioOutputSPDIF::AudioOutputSPDIF(int, int, int)':
C:\Users\Alex1\Documents\Arduino\libraries\ESP8266Audio\src\AudioOutputSPDIF.cpp:97:49: warning: 'I2S_COMM_FORMAT_I2S' is deprecated [-Wdeprecated-declarations]
     .communication_format = (i2s_comm_format_t)(I2S_COMM_FORMAT_I2S | I2S_COMM_FORMAT_I2S_MSB),
                                                 ^~~~~~~~~~~~~~~~~~~
In file included from C:\Users\Alex1\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.1/tools/sdk/esp32c3/include/driver/include/driver/i2s.h:16,
                 from C:\Users\Alex1\Documents\Arduino\libraries\ESP8266Audio\src\AudioOutputSPDIF.cpp:44:
C:\Users\Alex1\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.1/tools/sdk/esp32c3/include/hal/include/hal/i2s_types.h:95:5: note: declared here
     I2S_COMM_FORMAT_I2S       __attribute__((deprecated)) = 0x01, /*!< I2S communication format I2S, correspond to `I2S_COMM_FORMAT_STAND_I2S`*/
     ^~~~~~~~~~~~~~~~~~~
C:\Users\Alex1\Documents\Arduino\libraries\ESP8266Audio\src\AudioOutputSPDIF.cpp:97:71: warning: 'I2S_COMM_FORMAT_I2S_MSB' is deprecated [-Wdeprecated-declarations]
     .communication_format = (i2s_comm_format_t)(I2S_COMM_FORMAT_I2S | I2S_COMM_FORMAT_I2S_MSB),
                                                                       ^~~~~~~~~~~~~~~~~~~~~~~
In file included from C:\Users\Alex1\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.1/tools/sdk/esp32c3/include/driver/include/driver/i2s.h:16,
                 from C:\Users\Alex1\Documents\Arduino\libraries\ESP8266Audio\src\AudioOutputSPDIF.cpp:44:
C:\Users\Alex1\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.1/tools/sdk/esp32c3/include/hal/include/hal/i2s_types.h:96:5: note: declared here
     I2S_COMM_FORMAT_I2S_MSB   __attribute__((deprecated)) = 0x01, /*!< I2S format MSB, (I2S_COMM_FORMAT_I2S |I2S_COMM_FORMAT_I2S_MSB) correspond to `I2S_COMM_FORMAT_STAND_I2S`*/
     ^~~~~~~~~~~~~~~~~~~~~~~
C:\Users\Alex1\Documents\Arduino\libraries\ESP8266Audio\src\libflac\stream_decoder.c: In function 'read_residual_partitioned_rice_':
[... cut ...]
C:\Users\Alex1\Documents\Arduino\libraries\ESP8266Audio\src\libflac\private/bitreader.h:76:77: note: expected 'FLAC__int32 *' {aka 'long int *'} but argument is of type 'int *'
 FLAC__bool FLAC__bitreader_read_raw_int32(FLAC__BitReader *br, FLAC__int32 *val, uint32_t bits);
                                                                ~~~~~~~~~~~~~^~~
fork/exec C:\Users\Alex1\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\3.1.0/esptool.exe: This version of %1 is not compatible with the version of Windows you're running. Check your computer's system information and then contact the software publisher.
Error compiling for board ESP32C3 Dev Module.
```

FedericoBusero, 2021-12-06:
I see a lot of warnings that are perfectly normal: these are things that can be improved in the code. The last lines are the wrong ones:
```
fork/exec C:\Users\Alex1\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\3.1.0/esptool.exe: This version of %1 is not compatible with the version of Windows you're running. Check your computer's system information and then contact the software publisher.
Error compiling for board ESP32C3 Dev Module.

```
I think you first need to check if you can upload anything (Hello World, Blink...) on the ESP32C3. Check that you use the latest ESP32 Arduino board package or maybe you need to update to a newer Arduino IDE? 

P.S.: The warnings are related to
- new API for I2S since IDF 4.2. But they don't impact StreamMP3FromHTTP.ino when run on ESP32C3
It is rather straightforward to change to code the get rid of this warning, but I first want to check compatibility with ESP32, different versions of IDF, ...
- FLAC, which is of course no used in StreamMP3FromHTTP.ino

sh-user, 2021-12-07:
The problem is `esptool`, does not work on 32 bit Windows, only on 64 bit

sh-user, 2021-12-08:
Example `PlayWAVFromPROGMEM` compiles normally, but gives an error to the console
```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x484
load:0x403ce000,len:0x90c
load:0x403d0000,len:0x2398
SHA-256 comparison failed:
Calculated: e0c357829f601add6a50d6ea19f89687ec5acf531610506a38b33d09b4a71c7e
Expected: 77332826ec7bc491df854452f675071e3a094e80858dd7d91dd7e27940204216
Attempting to boot anyway...
entry 0x403ce000
WAV start
+0 0x3fc92d04
E (1205) I2S: i2s_set_pin(294): bck_io_num invalid
WAV done
WAV done
WAV done
```

sh-user, 2021-12-08:
found info for ESP32-C3 
```
#elif CONFIG_IDF_TARGET_ESP32C3
// TODO: change contacts
#define MASTER_BCK_IO 4
#define MASTER_WS_IO 5
#define SLAVE_BCK_IO 14
#define SLAVE_WS_IO 15
#define DATA_IN_IO 19
#define DATA_OUT_IO 18
#define I2S0_DATA_OUT_IDX I2SO_SD_OUT_IDX
#define I2S0_DATA_IN_IDX I2SI_SD_IN_IDX
```
Попробую изменить `AudioOutputI2S.cpp`
```
bclkPin = 26;
wclkPin = 25;
doutPin = 22;
```

sh-user, 2021-12-12:
i tried to change 
```
bclkPin = 4;
wclkPin = 5;
doutPin = 18;
``` 
```
bclkPin = 14;
wclkPin = 15;
doutPin = 18;
```
but the sound could not be received, maybe I'm doing something wrong

FedericoBusero, 2021-12-13:
The test environment in which I developped the ESP32C3 code:
- ESP Module: ESP32-C3-01M / ESP32-C3-14 
The pins you are using are indeed the ones that are specified in the documentation (but in theory all GPIO pins should work as I understand it). Those pins 4/5/18 do work on ESP32-C3-01M but on my ESP32-C3-14, pin 4 doesn't seem work for I2S use. Pins 6/5/18 work on all my chips, but in the example code I used 6/7/10.

- Arduino ESP32 Core version : 2.0.1
This is really important: earlier versions even don't compile the code for ESP32-C3 when using I2S. What is strange is that when trying to make applications for ESP32, you can't use version 2.0.1 when using I2S, because this version is buggy: it worked until 2.0.0rc2. But for ESP32C3 you really need to use 2.0.1

- I2S module: MAX98357A

- comm_fmt=I2S_COMM_FORMAT_STAND_I2S in AudioOutputI2S.cpp
Remark that I created a pull request because the first version compiled, but some things were not perfect:

https://github.com/earlephilhower/ESP8266Audio/pull/461

	- #ifdef I2S_MODE_PDM/I2S_MODE_DAC_BUILT_IN doesn't work because these are enum's which cannot be used by the preprocessor. I replaced them with #if CONFIG_IDF_TARGET_ESP32
	- comm_fmt should be I2S_COMM_FORMAT_STAND_I2S according to the documentation, so I changed following lines in AudioOutputI2S.cpp

```
#if ESP_IDF_VERSION >= ESP_IDF_VERSION_VAL(4, 2, 0)
        comm_fmt = (i2s_comm_format_t) (I2S_COMM_FORMAT_STAND_I2S);
#else
        comm_fmt = (i2s_comm_format_t) (I2S_COMM_FORMAT_I2S | I2S_COMM_FORMAT_I2S_MSB);
#endif
```

The source code of the test program looks like this:

```
#include "AudioFileSourcePROGMEM.h"
#include "AudioGeneratorRTTTL.h"
#include "AudioOutputI2S.h"

#define I2S_SPEAKER_SERIAL_CLOCK GPIO_NUM_6 // BCLK
#define I2S_SPEAKER_LEFT_RIGHT_CLOCK GPIO_NUM_7 // WSEL,LRC 
#define I2S_SPEAKER_SERIAL_DATA GPIO_NUM_10 // DIN

const char rudolph[] PROGMEM = 
"Rudolph the Red Nosed Raindeer:d=8,o=5,b=250:g,4a,g,4e,4c6,4a,2g.,g,a,g,a,4g,4c6,2b.,4p,f,4g,f,4d,4b,4a,2g.,g,a,g,a,4g,4a,2e.,4p,g,4a,a,4e,4c6,4a,2g.,g,a,g,a,4g,4c6,2b.,4p,f,4g,f,4d,4b,4a,2g.,g,a,g,a,4g,4d6,2c.6,4p,4a,4a,4c6,4a,4g,4e,2g,4d,4e,4g,4a,4b,4b,2b,4c6,4c6,4b,4a,4g,4f,2d,g,4a,g,4e,4c6,4a,2g.,g,a,g,a,4g,4c6,2b.,4p,f,4g,f,4d,4b,4a,2g.,4g,4a,4g,4a,2g,2d6,1c.6.";

AudioGeneratorRTTTL *rtttl;
AudioFileSourcePROGMEM *file;
AudioOutputI2S *out;

void setup()
{
  Serial.begin(115200);
  delay(1000);

  Serial.printf("RTTTL start\n");

  audioLogger = &Serial;
  file = new AudioFileSourcePROGMEM( rudolph, strlen_P(rudolph) );
  out = new AudioOutputI2S();
  out -> SetPinout(I2S_SPEAKER_SERIAL_CLOCK,I2S_SPEAKER_LEFT_RIGHT_CLOCK,I2S_SPEAKER_SERIAL_DATA);
  rtttl = new AudioGeneratorRTTTL();
  rtttl->begin(file, out);
}

void loop()
{
  if (rtttl->isRunning()) {
    if (!rtttl->loop()) rtttl->stop();
  } else {
    Serial.printf("RTTTL done\n");
    delay(1000);
  }
}
```

sh-user, 2021-12-13:
@FedericoBusero thanks i checked this example on ESP32-C3-12F (C3FN4), options `4/5/18`, `6/5/18`, `6/7/10` work! 
My mistake is that I connected the speaker to RX like ESP8266

sh-user, 2021-12-18:
Checked `AudioGeneratorMP3 `using `out->SetPinout()` works fine. 
There is a problem with starting `AudioGeneratorAAC`, I get a cyclic reset.

Source code:
```
#include <Arduino.h>
#include <WiFi.h>
#include "AudioFileSourceICYStream.h"
#include "AudioFileSourceBuffer.h"
#include "AudioGeneratorAAC.h"
#include "AudioOutputI2SNoDAC.h"

#define I2S_SPEAKER_SERIAL_CLOCK GPIO_NUM_6       // BCLK
#define I2S_SPEAKER_LEFT_RIGHT_CLOCK GPIO_NUM_7   // WSEL,LRC 
#define I2S_SPEAKER_SERIAL_DATA GPIO_NUM_10       // DIN

const int preallocateBufferSize = 16*1024;
const int preallocateCodecSize = 85332;      //AAC+SBR codec max mem needed
void *preallocateBuffer = NULL;
void *preallocateCodec = NULL;

const char *ssid = "********";
const char *password = "********";
const char *URL="http://ice23.securenetsystems.net/KARP";

AudioGenerator *aac = NULL;
AudioFileSourceICYStream *file = NULL;
AudioFileSourceBuffer *buff = NULL;
AudioOutputI2SNoDAC *out = NULL;

void setup() {
  Serial.begin(115200);
  Serial.println("Connecting to WiFi");
  WiFi.disconnect();
  WiFi.softAPdisconnect(true);
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED) {
    Serial.println("...Connecting to WiFi");
    delay(1000);
  }
  Serial.println("Connected");

  file = new AudioFileSourceICYStream(URL);
  buff = new AudioFileSourceBuffer(file, preallocateBuffer, preallocateBufferSize);
  out = new AudioOutputI2SNoDAC();
  out->SetPinout(I2S_SPEAKER_SERIAL_CLOCK,I2S_SPEAKER_LEFT_RIGHT_CLOCK,I2S_SPEAKER_SERIAL_DATA);
  aac = new AudioGeneratorAAC(preallocateCodec, preallocateCodecSize);
  aac->begin(buff, out);
}

void loop() {
  static int lastms = 0;
  if (aac->isRunning()) {
    if (millis()-lastms > 1000) {
      lastms = millis();
      Serial.printf("Running for %d ms...\n", lastms);
      Serial.flush();
     }
    if (!aac->loop()) aac->stop();
  } else {
    Serial.printf("AAC done\n");
    delay(1000);
  }
}
```
```
Connecting to WiFi
...Connecting to WiFi
[... 31 lines cut ...]
3fc9b8a0: 0x3fc95de4 0x3fc95e4c 0x00000000 0x00000000 0x00000001 0x00000000 0x00000000 0x00000000
3fc9b8c0: 0x4207d65e 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc9b8e0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc9b900: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc9b920: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc9b940: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc9b960: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc9b980: 0x00000000 0x00000000 0x00000000 0x3fc90000 0x00000054 0x00000000 0x3fc9b994 0x00000000
3fc9b9a0: 0x00000000 0x00000000 0x3fc9b9ac 0xffffffff 0x3fc9b9ac 0x3fc9b9ac 0x00000000 0x3fc9b9c0
3fc9b9c0: 0xffffffff 0x3fc9b9c0 0x3fc9b9c0 0x00000001 0x00000001 0x00000000 0x8000ffff 0x00000000
3fc9b9e0: 0x92ed1005 0x3fc9b98c 0x00000100 0x00000000 0x00000000 0x3fc9625c 0x00000000 0x00000000
3fc9ba00: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc9ba20: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc9ba40: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc9ba60: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc9ba80: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc9baa0: 0x00000000 0x0000000a 0x3fc9bf9c 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc9bac0: 0x00000000 0x00000000 0x00000000 0x00000000 0x3fc9baf0 0x00000000 0x3fc9be3c 0x3fc9bde4
3fc9bae0: 0x3fc9bf44 0x3fc9beec 0x3fc9be94 0x000000ec 0x00000200 0x00000200 0x00000002 0x40386660
3fc9bb00: 0x4038615e 0x40385a5a 0x40386604 0x403865e8 0x3fc9bbe0 0x3fc9bbe0 0x3fc9bbe0 0x3fc9bbe0

ELF file SHA256: 0000000000000000

Rebooting...
```

FedericoBusero, 2021-12-18:
In the example code, following lines are in the setup function
```
  // First, preallocate all the memory needed for the buffering and codecs, never to be freed
  preallocateBuffer = malloc(preallocateBufferSize);
  preallocateCodec = malloc(preallocateCodecSize);
  if (!preallocateBuffer || !preallocateCodec) {
    Serial.begin(115200);
    Serial.printf_P(PSTR("FATAL ERROR:  Unable to preallocate %d bytes for app\n"), preallocateBufferSize+preallocateCodecSize);
    while (1) delay(1000); // Infinite halt
  }
```

sh-user, 2021-12-19:
Use `malloc `worked, but a hiss is heard instead of a sound. I have tested different flow rates.

FedericoBusero, 2021-12-19:
I could reproduce the problem, it is situated in the AAC-decoder library which is not yet RISC-V compatible. You can make it work by replacing
```
#ifdef __XTENSA__		
		unsigned int lo32;
		signed int   hi32;
#else
```

by

```
#if defined(__XTENSA__) || defined (__riscv)
		unsigned int lo32;
		signed int   hi32;
#else
```

in libhelix-aac/assembly.h

But maybe, a new section for RISC-V should be created, inspired by 
https://github.com/ultraembedded/libhelix-mp3/commit/0a0e0673f82bc6804e5a3ddb15fb6efdcde747cd

sh-user, 2021-12-19:
works, but you can hear a slowdown in the playback speed by about two times, probably somewhere there is an offset

earlephilhower, 2021-12-19:
@FedericoBusero would you like to do a PR for that fix?  I don't have a RISC-V based chip to work with myself...

FedericoBusero, 2021-12-19:
I'll do a PR. I just compared some implementations of the same functions to see which are the fastest, but the ones in libhelix-mp3 are slower. 
The problem about the playback speed, I guess, is not a bug but just too slow performance of the chip: when the audio has no ticks, the sound is at the correct speed. But if there are a lot of ticks because of slow speed, playback becomes too slow. Maybe the wifi-implementation is not fully optimised yet.

FedericoBusero, 2021-12-19:
https://github.com/earlephilhower/ESP8266Audio/pull/466

sh-user, 2021-12-19:
I found information that it is possible to use a DMA buffer, but this is probably off topic.
https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-reference/peripherals/i2s.html

FedericoBusero, 2021-12-21:
The stuttering sound is solved in my case by changing

`.dma_buf_len = 64,`

into

`.dma_buf_len = 128,`
in AudioOutputI2S.cpp

sh-user, 2021-12-22:
`.dma_buf_len = 128` works well, no more stuttering

earlephilhower, 2021-12-22:
@FedericoBusero or @sh-user , want to do a PR?  Making it 128 for all ESP32 variants is safe...

sh-user, 2021-12-25:
checked `.dma_buf_len = 128` on ESP8266, works well including `320kbps`

commit 4971e5ce (2021-12-26): Increase I2S buffer size (#472)

Increase I2S buffer size from 64 samples/ block to 128 samples/block. 
Thus fixes sound artifacts described in issue https://github.com/earlephilhower/ESP8266Audio/issues/442

Fixes #442
closed 2021-12-26 by earlephilhower