earlephilhower/ESP8266Audio #447: Web radio with button works but ... it crashes at 5 channel :-(

massimo2021, 2021-11-22:
This code works but if I connect a button to change radio,  at the "if (CH == 5)"  line (the sixth radio station) it crashes.

Someone can help me?

```
#include <Arduino.h>
#include "AudioFileSourceICYStream.h" 
#include "AudioGeneratorMP3.h" 
#include "AudioOutputI2SNoDAC.h" 
#include "WiFi.h" 
#include "AudioFileSourceBuffer.h"

// Enter your WiFi setup here: 
const char *SSID = "ssid"; 
const char *PASSWORD = "password"; 

// Randomly picked URL 
const char* URL[]={"http://16813.live.streamtheworld.com:80/WRVOHD3_SC","http://jazz.streamr.ru/jazz-64.mp3",
"http://stream.live.vc.bbcmedia.co.uk/bbc_radio_fourfm","http://strm112.1.fm/blues_mobile_mp3",
"http://icestreaming.rai.it/4.mp3","http://studioone-stream.iowapublicradio.org/StudioOne.mp3","http://media-ice.musicradio.com/Heart80s"};
int buttonPin = 33; //CH+
String url_tmp;
int ch;
int val = 0;
AudioGenerator *mp3; 
AudioFileSourceHTTPStream *file; 
AudioOutputI2S *out; 
AudioFileSourceBuffer *buff;

void setup() { 
  pinMode(buttonPin, INPUT_PULLUP);
   Serial.begin(115200); 
   delay(1000); 
   WiFi.disconnect(); 
   WiFi.mode(WIFI_STA); 
   WiFi.begin(SSID, PASSWORD); 
   // Try to connect to Wifi forever 
   while (WiFi.status() != WL_CONNECTED) { 
       Serial.println("...Connecting to WiFi"); 
       delay(1000); 
   } 
   Serial.println("Connected"); 
    out = new AudioOutputI2S(0, 1); // (0,1) is using the internal DAC
    url_tmp=URL[0];
  play();

   
} 
void loop() 
{
static int lastms = 0;
val = digitalRead(buttonPin);
if (!val == 1) {
  Serial.print("Ch: ");
  delay(200);
  ch++;
  set_ch();
  Serial.println(ch);
  //StopPlaying();
  play();
}

  if (mp3->isRunning()) {
    if (millis()-lastms > 1000) {
      lastms = millis();
      Serial.printf("Running for %d ms...\n", lastms);
      Serial.flush();
     }
    if (!mp3->loop()) mp3->stop();
  } else {
    Serial.printf("MP3 done\n");
    delay(1000);
    play();
 }
}

void play()
{ 
audioLogger = &Serial;
file = new AudioFileSourceICYStream(url_tmp.c_str()); 
file->SetReconnect(5, 0); 
out = new AudioOutputI2S(0, 1); // (0,1) is using the internal DAC 
mp3 = new AudioGeneratorMP3();
mp3->begin(file, out); 
}

void StopPlaying()
{
  if (mp3) {
    mp3->stop();
    delete mp3;
    mp3 = NULL;
  }
  if (buff) {
    buff->close();
    delete buff;
    buff = NULL;
  }
  if (file) {
    file->close();
    delete file;
    file = NULL;
  }
}

void set_ch()
{
  if (ch==0) url_tmp=URL[0];
  if (ch==1) url_tmp=URL[1];
  if (ch==2) url_tmp=URL[2];
  if (ch==3) url_tmp=URL[3];
    if (ch==4) url_tmp=URL[4];
    if (ch==5) url_tmp=URL[5];
    if (ch==6) url_tmp=URL[6];
  if (ch>6) ch=0,  url_tmp=URL[0];
}
```

sh-user, 2021-11-28:
I think `StopPlaying()` should be done before `play()`

massimo2021, 2021-11-29:
Thanks

I did it , but I still get the same error and the board reboot  :-(

```
15:35:45.160 -> Running for 47351 ms...
15:35:45.790 -> Ch: 5
15:35:46.525 -> +0 0x3ffb1ec8
15:35:46.525 -> MP3 done
15:35:47.995 -> +0 0x3ffb1ec8
15:35:47.995 -> MP3 done
15:35:49.010 -> abort() was called at PC 0x4013ddf7 on core 1
15:35:49.010 -> 
15:35:49.010 -> Backtrace: 0x4008c59c:0x3ffb1e00 0x4008c7cd:0x3ffb1e20 0x4013ddf7:0x3ffb1e40 0x4013de3e:0x3ffb1e60 0x4013dcfd:0x3ffb1e80 0x4013da80:0x3ffb1ea0 0x4013da45:0x3ffb1ec0 0x400d8e83:0x3ffb1ee0 0x400d18f5:0x3ffb1f10 0x400d1df1:0x3ffb1f50 0x400d12a6:0x3ffb1f70 0x400d145b:0x3ffb1f90 0x400dcb19:0x3ffb1fb0 0x40088ce5:0x3ffb1fd0
15:35:49.010 -> 
15:35:49.010 -> Rebooting...
15:35:49.010 -> ets Jun  8 2016 00:22:57
15:35:49.010 -> 
15:35:49.010 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
15:35:49.045 -> configsip: 0, SPIWP:0xee
15:35:49.045 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
15:35:49.045 -> mode:DIO, clock div:1
15:35:49.045 -> load:0x3fff0018,len:4
15:35:49.045 -> load:0x3fff001c,len:1216
15:35:49.045 -> ho 0 tail 12 room 4
15:35:49.045 -> load:0x40078000,len:9720
15:35:49.045 -> ho 0 tail 12 room 4
15:35:49.045 -> load:0x40080400,len:6352
15:35:49.045 -> entry 0x400806b8
15:35:50.480 -> ...Connecting to WiFi
```