earlephilhower/ESP8266Audio #449: Bad naming choice in AudioGeneratorMOD ligne 59

vcchm, 2021-11-27:
The "DIVIDER" parameter in AudioGeneratorMOD.h 
    enum {DIVIDER = 10};    

conflicts with M5ATOM's "DIVIDER" parameter in FastLED\src/platforms/esp/32/clockless_rmt_esp32.h line 150
#define DIVIDER 2

Suggestion : name it MODDIVIDER instead of DIVIDER in next versions

earlephilhower, 2021-11-27:
:)  One might say that FastLED and ESP826Audio *both* made a poor choice in naming.

The name there came the original upstream sources.  A simple namespacing could also avoid the issue, but a search/replace would be fine, too.  Would you like to make a PR?

vcchm, 2021-11-27:
Indeed - two bad choices !  I tried to do a PR, but not sure I did it the right way...

By the way congrats for the code, really nice. I just sometimes have problems with some mods such as these which do not work..

[Ackerlight1.zip](https://github.com/earlephilhower/ESP8266Audio/files/7612495/Ackerlight1.zip)

PR #451 (2021-11-29): Add MOD test, update DIVIDER name
Fixes #449
commit e254ad95 (2021-11-29): Add MOD test, update DIVIDER name (#451)

Fixes #449
closed 2021-11-29 by earlephilhower