earlephilhower/ESP8266Audio #452: Feature Request: Integration with ESPHome

celer, 2021-11-29:
Hi,

Thank you for writing ESP8266Audio and ESP8266SAM, I was hoping to get these libraries working with ESPHome this weekend but it's a bit beyond me. It seems like the path is to build I2S support for ESPHome and then integrate ESP8266Audio and ESP8266SAM. 

The reason I'd prefer to use ESPHome rather than writing custom code is it's much easier to integrate with home assistant, ESPHome lowers the barrier of entry to using ESP hardware to pretty much being configuration. Some of my friends who can't program have setup their own ESPHome devices, this integration would expand usage of your libraries to non-programmers for all sorts of neat integrations. 

Anyways I tried to go down this path but I kept stumbling across the fact that I2S isn't supported in ESPHome, and to get there requires defining an I2S interface that works with both esp-idf and Arduino and then exposing that via yaml in ESPHome. 

There are multiple feature requests for ESPHome for this:

https://github.com/esphome/feature-requests/issues/599
https://github.com/esphome/feature-requests/issues/1254

Anyways I'd figure I'd let you know that your awesome libraries would be a great fit for integration with ESPHome expanding their usage to a large set of users who could do configuration based automatons. 

It would be super cool to be able to tell ESPHome to create a device that can play a mp3 at a URL, or use the built in text to speech service provided by Nuba Casa and stream the audio to a nearby esphome device. 

My end goal was to have home assistant send me notifications which are spoken via ESP8266SAM from an ESPHome device, but given the lift I'll probably stick with writing some code.

celer, 2021-11-29:
Here is a new issue I missed earlier:

https://github.com/esphome/feature-requests/issues/1528

earlephilhower, 2021-11-29:
While I've heard of ESPHome, I've never used it so take this with a grain of salt.

A quick glance at their model makes it seem like you'd need to make a CustomOutput ( https://esphome.io/components/output/custom.html ) but these outputs only seem to be able to take a binary or floating point value, not strings, as parameters.  So at first glance, I'm not really seeing how you'd be able to integrate some form of textual output at all.

I really don't think I'm the right person to look into this.  Someone who is a user and dev on the ESPHome team really would have a better understanding of how to integrate speech/audio...

celer, 2021-11-30:
Thank you for taking a quick look, I went down the path of creating a custom component, like this but initially stumbled when it came to the package management in platformio and how esphome integrates with it. 

It feels like the platformio IDE had been magically creating stub library packages from your work in ESP8266/Arduino, specifically SD, SDFat and when I told ESPHome to use those packages it had no idea how to do it, even though under the hood it's calling pio lib. 

So my quick and dirty approach was to remove all dependencies to SD/SDFat & reading files from ESP8266Audio which got me to my next problem, which is that I couldn't seem to get the custom component to include the internal headers from ESP8266/Arduino, specifically ESP8266/Arduino/cores/esp8266/core_esp8266_i2s.h. 

Anyways my plan was to create a custom component which listened to MQTT like MrDIY's speaker project which uses your libraries as a starting point. 

Thank you for taking some time to look at this, shortly I'll see if I have any better luck with an ESP32 approach as it looks like ESPHome has both esp-idf and Arduino support in the ESP32 branch and perhaps I can directly access the esp-idf library functions that way.  

Thanks!

Martin-Laclaustra, 2021-12-22:
@celer 
Did you see this attempt?
https://community.home-assistant.io/t/example-esphome-playing-wav-files-with-a-speaker/150302

nagyrobi, 2022-07-01:
https://esphome.io/components/media_player/i2s_audio.html

hmjvaline, 2022-07-29:
Using esphome to make voice speakers is something to be excited about. I have seen many ways to implement rrrtl tone, mp3 http stream, TTS SAM, Google Translation TTS on esp8266 (using rx pins followed by transistors, resistors) and esp32, and use mqtt to control voice content, the reference link is below, although esphome has implemented i2s media player, but only in esp32, obviously it can be used more generally, after all, any development application is always broad.

[esp8266 audio library](https://github.com/earlephilhower/ESP8266Audio)
[esp8266 audio notifier+MQTT](https://gitlab.com/MrDIYca/mrdiy-audio-notifier)
[esp8266 audio notifier youtube demo](https://youtu.be/SPa9SMyPU58) 
[esp8266 audio notifier +google translation TTS](https://github.com/schmurtzm/MrDiy-Audio-Notifier)