earlephilhower/ESP8266Audio #462: ESP8622 D1 mini lite: SPI OLED distorts RX output

Dediggefedde, 2021-12-10:
I have an ESP8622 D1 mini lite (nodeMCU) and successfully set up a Wi-Fi stream radio with this great library.

However, after I connected an OLED screen (hardware 4 Wire SPI), using the u8g2 library, whenever I give the screen a command, there is a buzzing/loud noise on the RX output. Just a `clear()` command creates the buzzing.
Also, all u8g2 commands are without effect after I set up 

    buff = new AudioFileSourceBuffer(file, 8192);
    buff->RegisterStatusCB(StatusCallback, (void *)"buffer");

Is this a known issue or limitation using RX for the audio-signal?
Or could I avoid this with a different library setting? Would switching to I2C help?

Code I use (partially copied from different sources)

    /*********
     * Webradio combines
     * ESP8266 D1 mini 160Hz
     * Wifi library
     * Audio library
     * u8g2 library for OLED display
     * Radio output using Transistor gate on RX pin from 3.3V
     * 
     ********/
    
    #include <Arduino.h>
    #include <ESP8266WiFi.h>
    
    #include "AudioFileSourceICYStream.h"
    #include "AudioFileSourceBuffer.h"
    #include "AudioGeneratorMP3.h"
    #include "AudioOutputI2SNoDAC.h"
    
    #include <U8g2lib.h>
    #ifdef U8X8_HAVE_HW_SPI
    #include <SPI.h>
    #endif
    #ifdef U8X8_HAVE_HW_I2C
    #include <Wire.h>
    #endif
    
    #include <SD.h>
    
    U8G2_SSD1306_128X64_NONAME_F_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/D5, /* data=*/D7, /* cs=*/D8, /* dc=*/D6, /* reset=*/D4);
    
    const char *SSID = "xxx";
    const char *PASSWORD = "xxx";
    const char *URL = "http://jazz.streamr.ru/jazz-64.mp3"; //"https://icecast.ndr.de/ndr/njoy/live/mp3/128/stream.mp3";
    IPAddress staticIP(192, 168, 0, 22);
    IPAddress gateway(192, 168, 0, 1);
    IPAddress subnet(255, 255, 255, 0);
    IPAddress dns(8, 8, 8, 8);
    IPAddress dns2(8, 8, 4, 4);
    
    AudioGeneratorMP3 *mp3;
    AudioFileSourceICYStream *file;
    AudioFileSourceBuffer *buff;
    AudioOutputI2SNoDAC *out;
    
    // Called when a metadata event occurs (i.e. an ID3 tag, an ICY block, etc.
    void MDCallback(void *cbData, const char *type, bool isUnicode, const char *string)
    {
    	const char *ptr = reinterpret_cast<const char *>(cbData);
    	(void)isUnicode; // Punt this ball for now
    	// Note that the type and string may be in PROGMEM, so copy them to RAM for printf
    	char s1[32], s2[64];
    	strncpy_P(s1, type, sizeof(s1));
    	s1[sizeof(s1) - 1] = 0;
    	strncpy_P(s2, string, sizeof(s2));
    	s2[sizeof(s2) - 1] = 0;
    	Serial.printf("METADATA(%s) '%s' = '%s'\n", ptr, s1, s2);
    	Serial.flush();
    }
    
    // Called when there's a warning or error (like a buffer underflow or decode hiccup)
    void StatusCallback(void *cbData, int code, const char *string)
    {
    	const char *ptr = reinterpret_cast<const char *>(cbData);
    	// Note that the string may be in PROGMEM, so copy it to RAM for printf
    	char s1[64];
    	strncpy_P(s1, string, sizeof(s1));
    	s1[sizeof(s1) - 1] = 0;
    	Serial.printf("STATUS(%s) '%d' = '%s'\n", ptr, code, s1);
    	Serial.flush();
    }
    
    void setup()
    {
    	Serial.begin(115200);
    
    	u8g2.begin();
    	u8g2.clear();						// clear the internal memory
    	u8g2.setFont(u8g2_font_ncenB08_tr); // choose a suitable font
    
    	WiFi.disconnect();
    	WiFi.softAPdisconnect(true);
    	WiFi.mode(WIFI_STA);
    	WiFi.config(staticIP, gateway, subnet, dns, dns2);
    	WiFi.begin(SSID, PASSWORD);
    	while (WiFi.status() != WL_CONNECTED)
    	{
    		Serial.println("...Connecting to WiFi");
    		u8g2.clear();
    		u8g2.setCursor(0, 15);
    		u8g2.print("Connecting to WiFi..."); // write something to the internal memory
    		u8g2.updateDisplay();				 // transfer internal memory to the display
    		delay(1000);
    	}
    	Serial.println("Connected");
    	u8g2.clear();
    	u8g2.setCursor(0, 15);
    	u8g2.print("Connected"); // write something to the internal memory
    	u8g2.updateDisplay();	 // transfer internal memory to the display
    
    	audioLogger = &Serial;
    	file = new AudioFileSourceICYStream(URL);
    	file->RegisterMetadataCB(MDCallback, (void *)"ICY");
    	buff = new AudioFileSourceBuffer(file, 8192);
    	buff->RegisterStatusCB(StatusCallback, (void *)"buffer");
    	out = new AudioOutputI2SNoDAC();
    	mp3 = new AudioGeneratorMP3();
    	mp3->RegisterStatusCB(StatusCallback, (void *)"mp3");
    	mp3->begin(buff, out);
    
    	Serial.println("Connecting to RadioJazz.ru");
    
    	u8g2.clear();
    	u8g2.setCursor(0, 15);
    	u8g2.print("Connecting to RadioJazz.ru"); // write something to the internal memory
    	u8g2.updateDisplay();					  // transfer internal memory to the display
    }
    
    void loop()
    {
    	static int lastms = 0;
    	if (mp3->isRunning())
    	{
    		if (millis() - lastms > 1000)
    		{
    			lastms = millis();
    
    			 //u8g2.clear();
    			// u8g2.setFont(u8g2_font_ncenB08_tr);
    			// u8g2.setCursor(0, 15);
    			// u8g2.print("Connected to RadioJazz.ru for ");
    			// u8g2.print(lastms);
    			// u8g2.print(" ms");
    			// u8g2.updateDisplay();	
    
    			Serial.printf("Running for %d ms...\n", lastms);
    			Serial.flush();
    		}
    		if (!mp3->loop())
    			mp3->stop();
    	}
    	else
    	{
    		Serial.printf("MP3 done\n");
    
    		u8g2.clearBuffer();
    		u8g2.setFont(u8g2_font_ncenB08_tr);
    		u8g2.setCursor(0, 15);
    		u8g2.print("Radio Disconneted");
    		u8g2.sendBuffer();
    		delay(1000);
    	}
    }

Dediggefedde, 2021-12-11:
I set up and I2C display now with Clear-Pin 3 and an Amplifier PAM8302A.
The buzzing is smaller now, but still pretty noticable. Like someone taps the speaker line every second...
In case I made a mistake with the wiring, I made a small circuit drawing.

[image]

ddgDDGddg, 2021-12-31:
曾经有过类似经历，但把PAM8302换成TDA2003之后就正常放音，具体原因不知

sh-user, 2022-01-11:
Try to power the PAM8302A from a separate power supply, for experiment add `WiFi.setOutputPower(0); `to `void setup ();` Possible interference from the WiFi transmitter.

Dediggefedde, 2022-01-12:
@sh-user 
Thanks for the hints!

GPIO02 would be D4 on this board. Good to know it's in use by the sound player.
Perhaps GPIO15 (D8) ist the problem, since this is also the hardware CS pin for SPI connections. I also have the sound at I2C and Software I2C, but perhaps there is an interference.
Currently, I use Hardware-I2C and connect the Reset pin of OLED to the ESP8266 one. Still got the pulsed noise, though.

Using a 3V battery instead for the PAM8302A has no effect on the sound.

WiFi.setOutputPower(0) does not work here, since I use the Wifi to stream the radio. Switching it off also cuts the connection to the radio. I tried WiFi.setOutputPower(5), but could not hear any difference.

For clarification I recorded a sound file of the radio playing over separately powered speakers:
https://drive.google.com/file/d/1s4ogudiAMh9zEXsTEPHZj2edOi1DlikB/view?usp=sharing

Here is my current setup (where I reduced cable length and added ~1mF capacitance to the power line) 
[image]

sh-user, 2022-01-13:
Try replacing `AudioFileSourceICYStream` with `AudioFileSourceHTTPStream`, maybe the problem is somewhere in the metadata.
You can also disable `Serial` to check if it breaks `RX`

Dediggefedde, 2022-01-14:
> Try replacing `AudioFileSourceICYStream` with `AudioFileSourceHTTPStream`, maybe the problem is somewhere in the metadata. You can also disable `Serial` to check if it breaks `RX`

Thanks for the suggestions.
Just replacing `AudioFileSourceICYStream` with `AudioFileSourceHTTPStream` without changing parameters seem to reduce the noise slightly, but it's still good to hear.

To disable serial, I removed `Serial.begin()` and all prints as well as the `audioLogger` assignment.
Or do I need to call `Serial.end()` or even disable interrupts on RX (`cbi(UCSRB, RXEN);` etc)?

sh-user, 2022-01-14:
I just removed `Serial ` from everywhere. But it's probably something else. I checked `http://jazz.streamr.ru/jazz-64.mp3`, it works fine (ESP8266 Core 3.0.1). Try excluding `u8g2`

Dediggefedde, 2022-01-14:
I tried using the Adafruit_SSD1306 library, but this led to the same effect.
The pulses also appear on other streams.
If I disable the OLED communication, the audio pulses vanish.

Perhaps those are just too many simultaneous tasks for the 8266, and I should switch to an ESP32... 
I only have atmega/tiny lying around at the moment, though, so I can't try it out. ^^