earlephilhower/ESP8266Audio #463: Playing mp3 files in a function on ESP32

HeikkiHietala, 2021-12-10:
Hi,

I wanted to thank you for building this library. I am hoping to build a talking clock, with MP3 files for the sounds of "oh one"... "oh nine" and then "one" to "fifty nine". I have the files made already and they all fit under 2MB of SPIFFS memory on the ESP32.

What I am struggling with is trying to build a method for running a function that actually plays the sound. The example has only one sound file, and while I can play my own sound, adding more than one doesn't work. 

So, what should the structure be in the loop that would allow for a sound to be played and then wait for the next?

doublejosh, 2022-01-03:
Same problem-- how to manage various audio files and trigger them as part of other code. Seems so basic, but managing the memory and file loading is immediately confusing.

Also, I'm failing to understand how we need to manage `mp3->isRunning())`, `mp3->loop()` and `mp3->stop()` to create a some kind of `play()` function. It may need take handlers like this, but I'm not sure... 
`void play(uint8_t message, AudioGeneratorMP3 *mp3, AudioFileSourceID3 *id3, AudioOutputI2SNoDAC *out) {`

The only thing like this is the example was the speech synthesis source. The audio quality is horrific, but it's looped.
https://github.com/earlephilhower/ESP8266Audio/blob/master/examples/TalkingClockI2S/TalkingClockI2S.ino

doublejosh, 2022-01-03:
This will at least play an mp3 on a loop. Next stop file loading...

```
#include <Arduino.h>
#ifdef ESP32
  #include <WiFi.h>
  #include "SPIFFS.h"
#else
  #include <ESP8266WiFi.h>
#endif
#include "AudioFileSourceSPIFFS.h"
#include "AudioFileSourceID3.h"
#include "AudioGeneratorMP3.h"
#include "AudioGeneratorTalkie.h"
#include "AudioFileSourcePROGMEM.h"
#include "AudioOutputI2SNoDAC.h"

// Set your ESP8266 build to 160MHz
// Use the "Tools->ESP8266/ESP32 Sketch Data Upload" menu to write the MP3 to SPIFFS

AudioGeneratorMP3 *mp3;
AudioFileSourceSPIFFS *file;
AudioOutputI2SNoDAC *out;
AudioFileSourceID3 *id3;
AudioGeneratorTalkie *talkie;

void MDCallback(void *cbData, const char *type, bool isUnicode, const char *string)
{
  (void)cbData;
  Serial.printf("ID3 callback for: %s = '", type);

  if (isUnicode) {
    string += 2;
  }
  
  while (*string) {
    char a = *(string++);
    if (isUnicode) {
      string++;
    }
    Serial.printf("%c", a);
  }
  Serial.printf("'\n");
  Serial.flush();
}

void play(AudioGeneratorMP3 *mp3, AudioFileSourceID3 *id3, AudioOutputI2SNoDAC *out)
{
  Serial.printf("Playing\n");
  file = new AudioFileSourceSPIFFS("/audio-phrases.mp3");
  id3 = new AudioFileSourceID3(file);
  id3->RegisterMetadataCB(MDCallback, (void*)"ID3TAG");
  mp3->begin(file, out);
}

void setup() 
{
  WiFi.mode(WIFI_OFF); 
  Serial.begin(115200);
  delay(1000);
  SPIFFS.begin();
  Serial.printf("Sample MP3 playback begins...\n");
  audioLogger = &Serial;
  out = new AudioOutputI2SNoDAC();
  mp3 = new AudioGeneratorMP3();
  mp3->begin(nullptr, out);
}

void loop()
{
  if (mp3->isRunning()) {
    if (!mp3->loop()) mp3->stop();
  } else {
    play(mp3, id3, out);
  }
  Serial.printf("LOOP\n");
}
```

HeikkiHietala, 2022-01-12:
Thanks for tis :)= sorry for the delay. I have built a case structure that allows me at least to loop through the list o files I have on the ESP32, and I am hoping to get that included in the main structure, but I will surely take a look at your solution.

earlephilhower, 2025-10-21:
This seems like a good use case for [BackgroundAudio](https://github.com/earlephilhower/BackgroundAudio).  Or, staying with ESP8266Audio you currently need to recreate the audio stack after completion of the first sound (`delete` the existing objects and `new` the new ones)...

closed 2025-10-21 by earlephilhower