earlephilhower/ESP8266Audio #464: Compilation of example PlayMP3fromSPIFFS fails

HeikkiHietala, 2021-12-15:
Hi,

without changing in the code, I tried to compile the example, but it fails with the following:

`
In file included from C:\Users\heikki.hietala\Dropbox\Arduino\libraries\ESP8266Audio-master\src\AudioGeneratorMIDI.cpp:65:
C:\Users\heikki.hietala\Dropbox\Arduino\libraries\ESP8266Audio-master\src\libtinysoundfont/tsf.h: In function 'void tsf_channel_midi_control(tsf*, int, int, int)':
C:\Users\heikki.hietala\Dropbox\Arduino\libraries\ESP8266Audio-master\src\libtinysoundfont/tsf.h:2100:1: error: insn does not satisfy its constraints:
 }
 ^
(insn 858 343 344 51 (set (reg:SF 19 f0 [407])
        (mem/u/c:SF (symbol_ref/u:SI ("*.LC248") [flags 0x2]) [0  S4 A32])) "C:\Users\heikki.hietala\Dropbox\Arduino\libraries\ESP8266Audio-master\src\libtinysoundfont/tsf.h":2053 47 {movsf_internal}
     (nil))
during RTL pass: postreload
C:\Users\heikki.hietala\Dropbox\Arduino\libraries\ESP8266Audio-master\src\libtinysoundfont/tsf.h:2100:1: internal compiler error: in extract_constrain_insn, at recog.c:2210
Please submit a full bug report,
with preprocessed source if appropriate.
See <https://gcc.gnu.org/bugs/> for instructions.
"C:\\Users\\heikki.hietala\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\xtensa-esp32-elf-gcc\\gcc8_4_0-esp-2021r2/bin/xtensa-esp32-elf-g++" -DHAVE_CONFIG_H "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DUNITY_INCLUDE_CONFIG_H -DWITH_POSIX -D_GNU_SOURCE "-DIDF_VER=\"v4.4-dev-3569-g6a7d83af19-dirty\"" -DESP_PLATFORM "-IC:\\Users\\heikki.hietala\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.1/tools/sdk/esp32/include/config" "-IC:\\Users\\heikki.hietala\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.1/tools/sdk/esp32/include/newlib/platform_include" "-IC:\\Users\\heikki.hietala\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.1/tools/sdk/esp32/include/freertos/include" "-IC:\\Users\\heikki.hietala\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.1/tools/sdk/esp32/include/freertos/include/esp_additions/freertos" "-IC:\\Users\\heikki.hietala\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.1/tools/sdk/esp32/include/freertos/port/xtensa/include" "-IC:\\Users\\heikki.hietala\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.1/tools/sdk/esp32/include/freertos/include/esp_additions" "-IC:\\Users\\heikki.hietala\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.1/tools/sdk/esp32/include/esp_hw_support/include" "-IC:\\Users\\heikki.hietala\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.1/tools/sdk/esp32/include/esp_hw_support/include/soc" "-IC:\\Users\\heikki.hietala\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.1/tools/sdk/esp32/include/esp_hw_support/include/soc/esp32" "-IC:\\Users\\heikki.hietala\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.1/tools/sdk/esp32/include/esp_hw_support/port/esp32" "-IC:\\Users\\heikki.hietala\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.1/tools/sdk/esp32/include/heap/include" "-IC:\\Users\\heikki.hietala\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.1/tools/sdk/esp32/include/log/include" "-IC:\\Users\\heikki.hietala\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.1/tools/sdk/esp32/include/lwip/include/apps" "-IC:\\Users\\heikki.hietala\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.1/tools/sdk/esp32/include/lwip/include/apps/sntp" "-IC:\\Users\\heikki.hietala\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.1/tools/sdk/esp32/include/lwip/lwip/src/include" "-IC:\\Users\\heikki.hietala\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.1/tools/sdk/esp32/include/lwip/port/esp32/include" "-IC:\\Users\\heikki.hietala\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.1/tools/sdk/esp32/include/lwip/port/esp32/include/arch" "-IC:\\Users\\heikki.hietala\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.1/tools/sdk/esp32/include/soc/include" "-IC:\\Users\\heikki.hietala\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.1/tools/sdk/esp32/include/soc/esp32" "-IC:\\Users\\heikki.hietala\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.1/tools/sdk/esp32/include/soc/esp32/include" "-IC:\\Users\\heikki.hietala\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.1/tools/sdk/esp32/include/hal/esp32/include" "-IC:\\Users\\heikki.hietala\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.1/tools/sdk/esp32/include/hal/include" "-IC:\\Users\\heikki.hietala\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.1/tools/sdk/esp32/include/hal/platform_port/include" "-IC:\\Users\\heikki.hietala\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.1/tools/sdk/esp32/include/esp_rom/include" "-IC:\\Users\\heikki.hietala\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.1/tools/sdk/esp32/include/esp_rom/include/esp32" "-IC:\\Users\\heikki.hietala\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.1/tools/sdk/esp32/include/esp_rom/esp32" "-IC:\\Users\\heikki.hietala\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.1/tools/sdk/esp32/include/esp_common/include" "-IC:\\Users\\heikki.hietala\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.1/tools/sdk/esp32/include/esp_system/include" "-IC:\\Users\\heikki.hietala\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.1/tools/sdk/esp32/include/esp_system/port/soc" "-IC:\\Users\\heikki.hietala\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.1/tools/sdk/esp32/include/esp_system/port/public_compat" "-IC:\\Users\\heikki.hietala\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.1/tools/sdk/esp32/include/esp32/include" "-IC:\\Users\\heikki.hietala\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.1/tools/sdk/esp32/include/xtensa/include" "-IC:\\Users\\heikki.hietala\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.1/tools/sdk/esp32/include/xtensa/esp32/include" "-IC:\\Users\\heikki.hietala\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.1/tools/sdk/esp32/include/driver/include" "-IC:\\Users\\heikki.hietala\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.1/tools/sdk/esp32/include/driver/esp32/include" "-IC:\\Users\\heikki.hietala\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.1/tools/sdk/esp32/include/esp_pm/include" "-IC:\\Users\\heikki.hietala\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.1/tools/sdk/esp32/include/esp_ringbuf/include" "-IC:\\Users\\heikki.hietala\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.1/tools/sdk/esp32/include/efuse/include" "-IC:\\Users\\heikki.hietala\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.1/tools/sdk/esp32/include/efuse/esp32/include" "-IC:\\Users\\heikki.hietala\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.1/tools/sdk/esp32/include/vfs/include" "-IC:\\Users\\heikki.hietala\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.1/tools/sdk/esp32/include/esp_wifi/include" "-IC:\\Users\\heikki.hietala\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.1/tools/sdk/esp32/include/esp_event/include" "-IC:\\Users\\heikki.hietala\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.1/tools/sdk/esp32/include/esp_netif/include" "-IC:\\Users\\heikki.hietala\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.1/tools/sdk/esp32/include/esp_eth/include" "-IC:\\Users\\heikki.hietala\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.1/tools/sdk/esp32/include/tcpip_adapter/include" "-IC:\\Users\\heikki.hietala\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.1/tools/sdk/esp32/include/esp_phy/include" "-IC:\\Users\\heikki.hietala\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.1/tools/sdk/esp32/include/esp_phy/esp32/include" "-IC:\\Users\\heikki.hietala\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.1/tools/sdk/esp32/include/esp_ipc/include" "-IC:\\Users\\heikki.hietala\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.1/tools/sdk/esp32/include/app_trace/include" "-IC:\\Users\\heikki.hietala\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.1/too [... cut ...]
C:\Users\heikki.hietala\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.1/tools/sdk/esp32/include/hal/include/hal/i2s_types.h:96:5: note: declared here
     I2S_COMM_FORMAT_I2S_MSB   __attribute__((deprecated)) = 0x01, /*!< I2S format MSB, (I2S_COMM_FORMAT_I2S |I2S_COMM_FORMAT_I2S_MSB) correspond to `I2S_COMM_FORMAT_STAND_I2S`*/
     ^~~~~~~~~~~~~~~~~~~~~~~

`

I reinstalled the library, tried installing a previous version (1.7 and 1.6), but they all fail. Does anyone have a fix for this?

earlephilhower, 2021-12-15:
This is a compiler bug and a dup of #440.  IIRC one user fixed it by completely removing the ESP32 toolchain and reinstalling, but please check the old bug to be sure.

closed 2021-12-15 by earlephilhower

HeikkiHietala, 2021-12-15:
Thanks for the very quick response. I'll try to use the external LittleFS first, then remove ESP32 from IDE and reinstall.

As I seem to understand, this library can be used to plat multiple files as in a clock saying "fourteen" "thirtyfive" in two separate MP3s, but I may have to return to you on that.