earlephilhower/ESP8266Audio #465: ESP32 : Feedback for "wemos D1 mini 32" and "doit esp32 devkit v4"

schmurtzm, 2021-12-16:
Hi,

I figure that there is different behavior depending the ESP32 board used :

- I have tested on "wemos D1 mini 32" : I have very bad sound with internal DAC. It works only if you set "CPU frequency" to 80MHZ in arduino IDE.
- I have tested "doit esp32 devkit v4" with better result on the sound quality (similar to esp8266 nodac I think) from the internal DAC. I let the default settings in arduino IDE flash options for this board.

For both when you use the internal DAC you can put your speaker between ping 25 and GND.

I wasn't able to use "no DAC" option. It works on my esp8266 but not on my ESP32, I have only crackling sounds. For what I understand that's a shame because the sound quality could be better comparing to the internal DAC. The output pin is Rx too on ESP32 ?

I presume that I could have better results with an external DAC. The flash frequency will have an impact too with external dac ?

Thanks for this awesome project ;)

softhack007, 2021-12-18:
> For both when you use the internal DAC you can put your speaker between ping 25 and GND.

Did you directly connect the speaker? Because usually it's not good to draw more than 10-15mA from a PIN. The cracking could actually be a reset, or some protection circuit that tries to limit output power. I'd always put a resistor between then PIN and speaker, like 220 Ohms depending on the size of the speaker. Do you see better results with a resistor in the line?

FedericoBusero, 2021-12-18:
The default pin for NoDAC on ESP32 is pin 22. Sound quality is really good with the standard oversampling 32. And even better with oversampling 64.

softhack007, 2021-12-19:
Another observation - it seems that the I2S "internal DAC" driver is broken in https://github.com/espressif/arduino-esp32 version 2.0.1 (based on IDF 4.4). On my ESP32 I only get loud noise when compiling with the latest arduino-esp32. It sounds somehow like what is described in this bug report: https://github.com/espressif/esp-idf/issues/7951.

The solution for me was to downgrade the esp32 package to version 1.0.6. Recompiled and uploaded - clear sound from PIN 26+PIN25 👍

FedericoBusero, 2021-12-20:
I2S implementation for ESP32 worked well until version 2.0.0rc2. But in that version I2S for ESP32C3 didn´t work. 

In 2.0.1 i2s doesn´t work on ESP32 but works for ESP32C3

softhack007, 2021-12-21:
see also https://github.com/espressif/arduino-esp32/issues/5938 - and a few other related ones.

schmurtzm, 2021-12-22:
Thank you for all these information.
I hope they will repair internal dac quickly for ESP32 !

@softhack007  you're right : I have to use pin 22 for internal DAC, I don't know why I had sound on the I2S pins too 😅 

I've made some progress on my "doit esp32 devkit v4" :
 - I'm now able to use internal DAC and an external with success.
 - I migrate my code vscode/platformio. Easier to switch between arduino core versions. And now I have a debugger which works.

And I also have a few failures :
 - I'm not able to make my WeMos D1 MINI ESP32 work correctly (still very bad sound) (Tried with espressif32@3.2.1 ... which is probably 1.0.6 in reality).  I'm OK with my other ESP32 for now but if someone has an idea for "Wemos D1 mini ESP32" ...

Now I have a problem which is common to all my boards with ESP8266Audio (including esp8266/esp32) : as [on this issue](https://github.com/earlephilhower/ESP8266Audio/issues/395) #395, it hang each time it finish to play [Google TTS](http://translate.google.com/translate_tts?ie=UTF-8&q=test&tl=fr&client=tw-ob&ttsspeed=1). 

Any idea is welcome :)

thanks 😃

ensingerphilipp, 2022-01-16:
Hi @schmurtzm - how exactly did you manage to get internal DAC on esp32 devkit v4 to work? 

Im having the same problem and im using platformio

FedericoBusero, 2022-01-16:
- Use ESP core version 1.0.6 or 2.0.0rc2
- Use the latest version of ESP8266Audio

Do not confuse internal DAC with NoDac, which is something different.

Internal DAC
- internal DAC is fixed on pin 25
- out = new AudioOutputI2S(0, 1);

NoDac
- the pin can be selected, default on 22
- change the pin eg out->SetPinout(26, 25, 22); sets it on pin 22
- out = AudioOutputI2SNoDAC();

schmurtzm, 2022-01-16:
> 

@ensingerphilipp  Hi, I've synthetised what I learned in this example : https://github.com/earlephilhower/ESP8266Audio/issues/469

I also have converted a project to platformio (it 's an aaudio player controlled by MQTT which is based on esp8266audio) here :  https://github.com/schmurtzm/MrDiy-Audio-Notifier

The plateformio.ini can work for ESP8266 and ESP32 and you can switch between external DAC, internal DAC and no DAC directly from the web UI (previous versions it was set in the code, not in the web UI).

It should help 😉

ensingerphilipp, 2022-01-16:
Thank you @schmurtzm @FedericoBusero 

In the end i was able to easily get to ESPCore Version 1.0.6 by specifying

platform = espressif32@3.3

in the platformio.ini