earlephilhower/ESP8266Audio #468: Using with PlatformIO on the ESP8266

abaskin, 2021-12-22:
I found that I needed to add the following to the lib_deps section of the platformio.ini file in order for the library to compile successfully with version 3.2.0 of the espressif8266 platform. Not a big deal, but maybe noting this in the documentation would make the library easier to use.
```
lib_deps =
  ESP8266HTTPClient
  SD
  SDFS
  ESP8266SdFat
```

schmurtzm, 2021-12-22:
Thanks, I use platformio too but I think that if you just add `lib_ldf_mode = deep` You don't have to look for other libraries dependencies. 
cf. [Dependency Finder Mode](https://docs.platformio.org/en/latest/librarymanager/ldf.html#id5)

abaskin, 2021-12-22:
Sorry about the duplicate. Using lib_ldf_mode = deep resolved the issue with the exception of ESP8266SdFat. I still need to include it in lib_deps to get the library to compile successfully.