earlephilhower/ESP8266Audio #471: RTTTL on ESP32 : play the last note forever,er,er,er,er,er...

schmurtzm, 2021-12-24:
Hi, I try to play with ESP32, now that I've [understood and tested the differences](https://github.com/earlephilhower/ESP8266Audio/issues/469) between no DAC, internal DAC and external DAC, I encounter some problems.
The first one is about RTTTL. This and is proper to ESP32 (I don't have the problem on ESP8266).

I take the example "PlayRTTTLToI2SDAC.ino" included in the library, I flash my ESP32 and it works but the last note is repeated for ever.
The esp32 is not crashed, it is possible to restart the complete play of the tone without problem but at the end it never stop. "rtttl->stop()" has no effect. I have tested on arduino IDE, platformio , with Wemos ESP32, ESP32 dev kit V4 , with older ESP32 framework, with older ESP8266Audio library : always the same result.

I've recorded [the result sound from my ESP32 here](https://dl.dropbox.com/s/i4y3qrln1kw2exw/RTTTL_on_ESP32.mp3).

Any idea what might be causing this problem?

------------------------------------------
PS : I have a second issue [which has been opened 7 months ago by other guys here](https://github.com/earlephilhower/ESP8266Audio/issues/395).  If some people at ease with this library can have a look 😉  ...

Thanks again for this awesome library !

Martin-Laclaustra, 2021-12-25:
This problem was previously reported (by me) and it requires further investigation.
https://github.com/earlephilhower/ESP8266Audio/issues/327
As your report is very valuable and thorough, you may consider re-posting it in the former thread (to keep all information in one place) and to keep posting there future advances.
All help is welcome.
One information that could help in deciding where to look is testing whether the problem happens also with "no DAC" and with "external DAC" if you have the hardware to do so.

schmurtzm, 2021-12-25:
#327 updated !
When I saw [that](https://github.com/maxprofs-llcio/ESP8266Audio/commit/00cf3052c58195bf06f2c4d1bf669f366f727ff3) I was thinking that I had a solution but it has been already commited 😢 
There are some similarities with #395 in the behavior but not sure that it is the same problem (looks more like a crash for the #395).

schmurtzm, 2021-12-26:
To make more tests quicky, you can try the code I am currently working on : 
https://github.com/schmurtzm/MrDiy-Audio-Notifier

This is a project initially from MrDIY, which is a great compilation of all the ESP8266 audio capacity : a player controlled by MQTT able to play mp3, tone, TTS sam voice and  TTS from google translate.
It has been ported to platformio and is compatible with ESP32 (excepted the bugs mentioned here).
This code also allows to switch easily between no DAC (version initally used by mr DIY) , external DAC or internal DAC.
You’ll also find some useful comments for wiring your DAC quickly or to improve the code.

commit fe7a1f79 (2021-12-27): Fix RTTTL stop on ESP32 (#473)

Fixes #327 
Fixes #471
closed 2021-12-27 by earlephilhower