earlephilhower/ESP8266Audio #474: AudioGeneratorMOD - scartches/noise due to integer overflow

softhack007, 2021-12-29:
Hi,

Thanks for creating this audio library, which even allowed me to dig out some "amiga MODs" and play them on a micro-controller :-)

Actually I have noticed some "over-saturation" noise in the MOD generator, especially when several channels are playing at high volume. It seems to me that the following code is having a 15-bit overflow:
https://github.com/earlephilhower/ESP8266Audio/blob/fe7a1f7951236d660a11167a35e05ae495e60b04/src/AudioGeneratorMOD.cpp#L738-L739
https://github.com/earlephilhower/ESP8266Audio/blob/fe7a1f7951236d660a11167a35e05ae495e60b04/src/AudioGeneratorMOD.cpp#L744
https://github.com/earlephilhower/ESP8266Audio/blob/fe7a1f7951236d660a11167a35e05ae495e60b04/src/AudioGeneratorMOD.cpp#L806-L815

Probably the overflow is caused by the "16-bit up-scaling" that was not present in the original stellarplayer. 
Changing `sumL`, `sumR` and `out` to `int32_t` fixes the problem.

Maybe I can make a PR over the weekend.

Cheers, Frank.

earlephilhower, 2021-12-29:
There was a bug in older versions of the MOD lib where I was off-by-one in shifting the audio output from fixed point to integer, causing overflow that way and bad sounding mods, but I haven't noticed any other problems with it, so good ears on you!

> Maybe I can make a PR over the weekend.

That would be great, especially since you not only found the problem but have a fix!

softhack007, 2022-01-01:
_update -_ after spending a few hours of debugging:

1. the mixer code from stellarplayer is a f****** buggy piece of c***. Probably not even the original author could explain why it works 90% of the time. And yes, with some fixes it seems to work also for BITDEPTH = 16 (instead of 15).

2. int16 vs int32 is only a partial the solution. The code below shifts samples into "unsigned" ranges. Then - I'm using AudioOutputI2S with internal DAC - this operation is repeated in the output driver, leading to crazy artefacts. BTW, i think this also explains why you did not experience the same "scratching" that i heared...
https://github.com/earlephilhower/ESP8266Audio/blob/fe7a1f7951236d660a11167a35e05ae495e60b04/src/AudioGeneratorMOD.cpp#L821-L824

3. I am playing MODs from SPIFFS. Guess that something's wrong in the FatBuffer code, as GetSample sometimes computes stupidly large values (-> noise). Not sure what happens exactly. A workaround is to test sumL and sumR against INT16_MIN / INT16_MAX, and not updating sample outputs when clipping would occur (this ignores "bad" samples by repeating the last sample).

4. Initially I was re-using the same instance of "AudioGeneratorMOD" in my scetch, using mod->stop() and mod->begin() to play another MOD file. This lead to "cracking" when the next MOD file starts playing. My workaround is to delete AudioGeneratorMOD, and then creating a new instance for the next file to play. Maybe something you could look into ...

5. Guru Meditation when playing MODs with more than 4 channels - due to "CHANNELS = 4" in the .h file. After changing to "CHANNELS = 8", playback works also for 8CHAN; however I'm not sure if this is the best solution, because it increases the memory footprint. What do you think?

I'll create a pull request tomorrow or monday to fix (or work around) some of the problems - actually i need some time to clear out all my extra debugging code.

softhack007, 2022-01-01:
PS: below some MODs for reproducing problems:
* https://modarchive.org/index.php?request=view_by_moduleid&query=87180 (for reproducing my original problem report)
* https://modarchive.org/index.php?request=view_by_moduleid&query=77902 (FartBuffer problems at the end, even with `mod->SetBufferSize(12 * 1024 )` )
* https://modarchive.org/index.php?request=view_by_moduleid&query=42975 (clipping, clipping, clipping...)

And some stranger things for the guru
* https://modarchive.org/index.php?request=view_by_moduleid&query=35121 (8 Channel)
* https://modarchive.org/index.php?request=view_by_moduleid&query=189813 (6 Channel)
* https://modarchive.org/index.php?request=view_by_moduleid&query=52420 (12 channel)  - did not try with ESPAudio yet

earlephilhower, 2022-01-02:
Hey, good work digging into this.  I would not be so harsh on Stellaplayer...he wrote it for a tiny microcontroller with way less RAM and MIPS than we have, and it works pretty well in many cases.  And I'm sure that I busted some things as I ported to the ESP systems!

Anyway, if you have a Linux machine, or WSL on Windows, it is easy to build and test changes to the MOD player.  Look at the `tests/host` directory.  `make mod` will build a sample player that runs on the host and writes a WAV file out.  It's how I check for memory bugs/etc., but it also makes it much easier to test/debug any changes you're looking at.

softhack007, 2022-01-04:
Thanks for the tests/host tip, that's a lot better 👍

You are right about stellarplayer - the code is a bit "fragile", however it turned out that it works basically. Maybe I just experienced a high WTF/min (https://www.osnews.com/story/19266/wtfsm/) when trying to understand it ;-)

commit c55d240a (2022-01-05): Fix for issue #474, and a few other bugs in MODGenerator (#478)

Fixes #474

This bug was already present in stellaplayer. Obviously "effectNumber" must be checked, because "effectParameterX" just holds the "upper nibble" of effectParameter

- do channel mixing in 32bit, to avoid noise from over/underflow
- write audio output as "signed int" (not unsigned)
- add int16 saturation check ("clipping")
* avoid crash when MOD file has to many CHANNELS
- avoid guru meditation, by logging an error and refusing playback in case MOD.numberOfChannels > CHANNELS