earlephilhower/ESP8266Audio #476: There is noise after playing music

gianglongtbe, 2022-01-03:
I used the example but got noise after finishing can you tell me how to fix it

https://user-images.githubusercontent.com/49691777/147898223-6c2990fc-faa5-4415-998c-7f7a2e60f97f.mp4

?

gianglongtbe, 2022-01-03:
[image]

Why does noise appear after every music playback ends

gianglongtbe, 2022-01-04:
I have looked at the entries but there is no improvement.  Could it be the software causing the noise?

softhack007, 2022-01-07:
Hmmm :thinking: reminds me of this discussion : https://github.com/earlephilhower/ESP8266Audio/pull/271

Maybe a call to output->flush() should be added in the audioGenerator class. As a quick test, you could try this in your scetch:
```C++
  if (!wav->loop()) {
     out->flush();
     wav->stop();
  }
} else {
```