earlephilhower/ESP8266Audio #483: Will not compile in arduino IDE

tkim61, 2022-01-12:
I am just trying to compile the example PlayWAVFromPROGMEM and am encountering error.  No edits have been made, using macOS Big Sur using a node32s board.  I am also using a proprietary amplifier, though it seems to be similar to most other amplifiers I have found on the internet.

`In file included from /Users/TJKim/Documents/Arduino/libraries/ESP8266Audio/src/AudioGeneratorMIDI.cpp:65:
/Users/TJKim/Documents/Arduino/libraries/ESP8266Audio/src/libtinysoundfont/tsf.h: In function 'void tsf_channel_midi_control(tsf*, int, int, int)':
/Users/TJKim/Documents/Arduino/libraries/ESP8266Audio/src/libtinysoundfont/tsf.h:2100:1: error: insn does not satisfy its constraints:
 }
 ^
(insn 858 343 344 51 (set (reg:SF 19 f0 [407])
        (mem/u/c:SF (symbol_ref/u:SI ("*.LC248") [flags 0x2]) [0  S4 A32])) "/Users/TJKim/Documents/Arduino/libraries/ESP8266Audio/src/libtinysoundfont/tsf.h":2053 47 {movsf_internal}
     (nil))
during RTL pass: postreload
/Users/TJKim/Documents/Arduino/libraries/ESP8266Audio/src/libtinysoundfont/tsf.h:2100:1: internal compiler error: in extract_constrain_insn, at recog.c:2210
libbacktrace could not find executable to open
Please submit a full bug report,
with preprocessed source if appropriate.
See <https://gcc.gnu.org/bugs/> for instructions.
Multiple libraries were found for "WiFiClient.h"
 Used: /Users/TJKim/Library/Arduino15/packages/esp32/hardware/esp32/2.0.2/libraries/WiFi
 Not used: /Applications/Arduino.app/Contents/Java/libraries/WiFi
Multiple libraries were found for "SD.h"
 Used: /Users/TJKim/Library/Arduino15/packages/esp32/hardware/esp32/2.0.2/libraries/SD
 Not used: /Applications/Arduino.app/Contents/Java/libraries/SD
exit status 1
Error compiling for board Node32s.`

Any help to get the example code compiling?

softhack007, 2022-01-12:
>internal compiler error: in extract_constrain_insn, at recog.c:2210 libbacktrace could not find executable to open Please submit a full bug report, with preprocessed source if appropriate. See <https://gcc.gnu.org/bugs/> for instructions.

Are you using arduino-esp32 version 2.0.x? I had the same ICE (internal compiler error). Installing an older ESP32 board package (version 1.0.6) solved the problem for me.

earlephilhower, 2022-01-12:
That's a compiler bug in the toolchain you're using, as the error message says.  Nothing we can do here, unfortunately, as it's a GCC issue.
````
internal compiler error: in extract_constrain_insn, at recog.c:2210
libbacktrace could not find executable to open
Please submit a full bug report,
with preprocessed source if appropriate.
See <https://gcc.gnu.org/bugs/> for instructions.
````

As @softhack007 said, you can try an earlier or later version of ESP's toolchain.   Or, if you are really stuck, you can delete tsh.f and the AudioGeneratorMIDI.cpp/h files from your local copy of the library.

closed 2022-01-12 by earlephilhower

tkim61, 2022-01-12:
Switching to 1.0.6 fixed the compile issue.  Code seems to run perfectly fine on the board now, though I am not hearing any output.