earlephilhower/ESP8266Audio #484: External DAC Error

medvosa, 2022-01-13:
Hello!
I've bought this DAC: https://aliexpress.ru/item/32807703240.html?sku_id=64387073698
But I can't get in working
My pinout:
VIN->3v3
GND->GND
LCK->D2
DIN->RX0 (I don't have RX without number, my rx0 is GPIO 3)
BCK-> D15 (
SCK->GND
I have an ESP-32 module with exactly the same pinout as shown here
[image]

Here's my code
```c++
/**
 * working sound for internal dac - D25
 * SD : 
 *      CS - D5
 *      MOSI - D23
 *      CLK - D18
 *      MSIO - D19
 * 
 * DISPLAY :
 *      SCL - D22
 *      SDA - D21
 * 
 * CONTROL :
 *      NEXT - D13
 *      PREV - D12
*/

#include <SPIFFS.h>
#include <HTTPClient.h>
#include <Arduino.h>
#include <AudioFileSourceSD.h>
#include <AudioFileSourceID3.h>
#include <AudioOutputSPDIF.h>
#include <AudioGeneratorMP3.h>
#include <AudioOutputI2SNoDAC.h>
#include <AudioOutputI2S.h>
#include <SD.h>
#include <Adafruit_GFX.h>
#include <Adafruit_SSD1306.h>
#define SCREEN_WIDTH 128 // OLED display width, in pixels
#define SCREEN_HEIGHT 64 // OLED display height, in pixels
Adafruit_SSD1306 display(SCREEN_WIDTH, SCREEN_HEIGHT, &Wire, -1);
#define SPI_SPEED SD_SCK_MHZ(40)
#define SPDIF_OUT_PIN 27

File dir;
AudioFileSourceID3 *id3;
AudioFileSourceSD *source = NULL;
AudioOutputSPDIF *output = NULL;
AudioGeneratorMP3 *decoder = NULL;

AudioOutputI2S *out;

void MDCallback(void *cbData, const char *type, bool isUnicode, const char *string)
{
  (void)cbData;
  Serial.printf("ID3 callback for: %s = '", type);

  if (isUnicode) {
    string += 2;
  }
  
  while (*string) {
    char a = *(string++);
    if (isUnicode) {
      string++;
    }
    Serial.printf("%c", a);
  }
  Serial.printf("'\n");
  Serial.flush();
}

class PlayOrder{
  public:
    String list[100];
    int num=-1;
    void addToList(String v){
      num++;
      list[num]=v;
    }
    String prev(){
      if(num>0)
        num--;
      String ret = list[num];
      return ret;
    }
    void showOrderSerial(){
      Serial.println("");
      Serial.println("Play order");
      for(int i=0; i<=num; i++){
        Serial.print(i+1);
        Serial.print(". ");
        Serial.println(list[i]);
      }
      Serial.println("");
    }
};

class ButtonsState{
  public:
    int next;
    int prev;
};

class Display{
  public:
    void showName(String name){
      display.clearDisplay();
      display.setTextSize(1);
      display.setTextColor(WHITE);
      display.setCursor(10, 20);
      display.println(name);
      display.display(); 
    }
};

ButtonsState buttons;
PlayOrder po;
Display dis;

void checkInput(){
  if(digitalRead(14)==0)
  buttons.next+=1;
  else buttons.next=0;
  if(digitalRead(12)==0)
  buttons.prev += 1;
  else buttons.prev=0;
}

void setup() {
  pinMode(14, INPUT_PULLUP);
  pinMode(12, INPUT_PULLUP);
  Serial.begin(115200);
  Serial.println();
  delay(1000);

  if(!display.begin(SSD1306_SWITCHCAPVCC, 0x3C)) { // Address 0x3D for 128x64
    Serial.println(F("SSD1306 allocation failed"));
  }
  display.clearDisplay();

  display.setTextSize(1);
  display.setTextColor(WHITE);
  display.setCursor(0, 10);
  // Display static text
  display.println("Starting...");
  display.display(); 

  audioLogger = &Serial;  
  source = new AudioFileSourceSD();
  
  id3 = new AudioFileSourceID3(source);
  
  id3->RegisterMetadataCB(MDCallback, (void*)"ID3TAG");
  out = new AudioOutputI2S(/*0,1*/0,1);//uncomment for internal
  decoder = new AudioGeneratorMP3();

  // NOTE: SD.begin(...) should be called AFTER AudioOutputSPDIF() 
  //       to takover the the SPI pins if they share some with I2S
  //       (i.e. D8 on Wemos D1 mini is both I2S BCK and SPI SS)
  #if defined(ESP8266)
    SD.begin(SS, SPI_SPEED);
  #else
    SD.begin( );
  #endif
  dir = SD.open("/"); 
}

void loop() {
  checkInput();
  if ((decoder) && (decoder->isRunning()) && buttons.next!=1 && buttons.prev!=1) {
    if (!decoder->loop()) decoder->stop();
  } else {
    if(buttons.prev==1){
      //switch to next track
      File file = SD.open(po.prev().c_str());
      if (file) {  
        Serial.println("Playing "+String(file.name()));    
        if (String(file.name()).endsWith(".mp3")) {
          source->close();
          if (source->open(file.name())) { 

            dis.showName(file.name()); 
            po.addToList(file.name());
            po.showOrderSerial();

            Serial.printf_P(PSTR("Playing '%s' from SD card...\n"), file.name());
            decoder->begin(id3, out);
          } else {
            Serial.printf_P(PSTR("Error opening '%s'\n"), file.name());
          }
        } 
      } else {
        Serial.println(F("Playback form SD card done\n"));
        delay(1000);
      }
    }
    else{
      //switch to next track
      File file = dir.openNextFile();
      if (file) {  
        Serial.println("Playing "+String(file.name()));    
        if (String(file.name()).endsWith(".mp3")) {
          source->close();
          if (source->open(file.name())) { 

            dis.showName(file.name()); 
            po.addToList(file.name());
            po.showOrderSerial();

            Serial.printf_P(PSTR("Playing '%s' from SD card...\n"), file.name());
            decoder->begin(id3, out);
          } else {
            Serial.printf_P(PSTR("Error opening '%s'\n"), file.name());
          }
        } 
      } else {
        Serial.println(F("Playback form SD card done\n"));
        delay(1000);
      }       
    }
  }
}
```

Could you please suggest me what am I doing wrong?

softhack007, 2022-01-13:
Hi,
Can you give some more details on what's not working? Did you try one of the examples, like "PlayMP3FromSPIFFS"?

.

In case you don't get sound output, I can give you some hints:
* It seems that you have a PCM 5102 based DAC, so in general you can follow instructions in the readme https://github.com/earlephilhower/ESP8266Audio#pcm5102-dac
* On the ESP32, the default pins for I2S signals are: **bclk= 26;  wclk= 25;  dout= 22**;
* please check that you don't have `LED_BUILTIN` or other stuff active on any of the I2S pins. 
* It seems that you need GPIO22 for I2C (display?), so you should use `out->SetPinout(...)` to change I2S pins.
* the 5102 is a DAC that usually needs an amplifier board before you hear anything from a loudspeaker - unless you have bought one with built in amplifier. 
* When connecting the amp board, It might be safer to put a resistor (~470ohm) between the PCM5102 L+R output and the amp board inputs.

I hope this helps you a bit to find the problem.

schmurtzm, 2022-01-15:
Take a look to my issue, it could help for pinout : 
https://github.com/earlephilhower/ESP8266Audio/issues/469

Start with the simplest example.
You could also make a quick test with internal dac. 

And take a look to my repo to have an implementation of ESP8266audio which can be configured on the web interface : 
https://github.com/schmurtzm/MrDiy-Audio-Notifier

medvosa, 2022-01-17:
@schmurtzm @softhack007 THANKS A LOT

schmurtzm, 2022-01-18:
> What do I do with sck contact?
Probably put it to GND.

medvosa, 2022-01-19:
Thanks everybody who helped.

I've commented all display logic temporarily from my source.
DAC still does not work. Once I've got a weak sound from it, but it looked like PWM sound, maybe I am configuring AudioOutputI2S incorrectly?

```C++
#include <SPIFFS.h>
#include <HTTPClient.h>
#include <Arduino.h>
#include <AudioFileSourceSD.h>
#include <AudioFileSourceID3.h>
#include <AudioOutputSPDIF.h>
#include <AudioGeneratorMP3.h>
#include <AudioOutputI2SNoDAC.h>
#include <AudioOutputI2S.h>
#include <SD.h>
#include <Adafruit_GFX.h>
#include <Adafruit_SSD1306.h>
#define SCREEN_WIDTH 128 // OLED display width, in pixels
#define SCREEN_HEIGHT 64 // OLED display height, in pixels
// Adafruit_SSD1306 display(SCREEN_WIDTH, SCREEN_HEIGHT, &Wire, -1);
#define SPI_SPEED SD_SCK_MHZ(40)
#define SPDIF_OUT_PIN 27

File dir;
AudioFileSourceID3 *id3;
AudioFileSourceSD *source = NULL;
AudioOutputSPDIF *output = NULL;
AudioGeneratorMP3 *decoder = NULL;

AudioOutputI2S *out;

void MDCallback(void *cbData, const char *type, bool isUnicode, const char *string)
{
  (void)cbData;
  Serial.printf("ID3 callback for: %s = '", type);

  if (isUnicode) {
    string += 2;
  }
  
  while (*string) {
    char a = *(string++);
    if (isUnicode) {
      string++;
    }
    Serial.printf("%c", a);
  }
  Serial.printf("'\n");
  Serial.flush();
}

class PlayOrder{
  public:
    String list[100];
    int num=-1;
    void addToList(String v){
      num++;
      list[num]=v;
    }
    String prev(){
      if(num>0)
        num--;
      String ret = list[num];
      return ret;
    }
    void showOrderSerial(){
      Serial.println("");
      Serial.println("Play order");
      for(int i=0; i<=num; i++){
        Serial.print(i+1);
        Serial.print(". ");
        Serial.println(list[i]);
      }
      Serial.println("");
    }
};

class ButtonsState{
  public:
    int next;
    int prev;
};

class Display{
  public:
    void showName(String name){
      // display.clearDisplay();
      // display.setTextSize(1);
      // display.setTextColor(WHITE);
      // display.setCursor(10, 20);
      // display.println(name);
      // display.display(); 
    }
};

ButtonsState buttons;
PlayOrder po;
Display dis;

void checkInput(){
  if(digitalRead(14)==0)
  buttons.next+=1;
  else buttons.next=0;
  if(digitalRead(12)==0)
  buttons.prev += 1;
  else buttons.prev=0;
}

void setup() {
  pinMode(14, INPUT_PULLUP);
  pinMode(12, INPUT_PULLUP);
  Serial.begin(115200);
  Serial.println();
  delay(1000);

  // if(!display.begin(SSD1306_SWITCHCAPVCC, 0x3C)) { // Address 0x3D for 128x64
  //   Serial.println(F("SSD1306 allocation failed"));
  // }
  // display.clearDisplay();

  // display.setTextSize(1);
  // display.setTextColor(WHITE);
  // display.setCursor(0, 10);
  // // Display static text
  // display.println("Starting...");
  // display.display(); 

  audioLogger = &Serial;  
  source = new AudioFileSourceSD();
  
  id3 = new AudioFileSourceID3(source);
  
  id3->RegisterMetadataCB(MDCallback, (void*)"ID3TAG");
  out = new AudioOutputI2S();//uncomment for internal
  decoder = new AudioGeneratorMP3();

  // NOTE: SD.begin(...) should be called AFTER AudioOutputSPDIF() 
  //       to takover the the SPI pins if they share some with I2S
  //       (i.e. D8 on Wemos D1 mini is both I2S BCK and SPI SS)
  #if defined(ESP8266)
    SD.begin(SS, SPI_SPEED);
  #else
    SD.begin( );
  #endif
  dir = SD.open("/"); 
}

void loop() {
  checkInput();
  if ((decoder) && (decoder->isRunning()) && buttons.next!=1 && buttons.prev!=1) {
    if (!decoder->loop()) decoder->stop();
  } else {
    if(buttons.prev==1){
      //switch to next track
      File file = SD.open(po.prev().c_str());
      if (file) {  
        Serial.println("Playing "+String(file.name()));    
        if (String(file.name()).endsWith(".mp3")) {
          source->close();
          if (source->open(file.name())) { 

            dis.showName(file.name()); 
            po.addToList(file.name());
            po.showOrderSerial();

            Serial.printf_P(PSTR("Playing '%s' from SD card...\n"), file.name());
            decoder->begin(id3, out);
          } else {
            Serial.printf_P(PSTR("Error opening '%s'\n"), file.name());
          }
        } 
      } else {
        Serial.println(F("Playback form SD card done\n"));
        delay(1000);
      }
    }
    else{
      //switch to next track
      File file = dir.openNextFile();
      if (file) {  
        Serial.println("Playing "+String(file.name()));    
        if (String(file.name()).endsWith(".mp3")) {
          source->close();
          if (source->open(file.name())) { 

            dis.showName(file.name()); 
            po.addToList(file.name());
            po.showOrderSerial();

            Serial.printf_P(PSTR("Playing '%s' from SD card...\n"), file.name());
            decoder->begin(id3, out);
          } else {
            Serial.printf_P(PSTR("Error opening '%s'\n"), file.name());
          }
        } 
      } else {
        Serial.println(F("Playback form SD card done\n"));
        delay(1000);
      }       
    }
  }
}```

medvosa, 2022-02-01:
I made a screenshot of data paassed to dac, looks like something correct. What else can I do wrong?https://media.discordapp.net/attachments/804717131249221672/938161284195688509/unknown.png?width=1200&height=409

medvosa, 2022-02-05:
Did anybody have the same problem?