earlephilhower/ESP8266Audio #486: MIDI from LittleFS sounds distorted (but WAV from Progmem is fine)

henrik6, 2022-01-16:
The short violin file from PlayWAVFromProgmem example plays back quite clear without noise on my D1 Mini / ESP8266, however the "fuer_elise.mid" from  PlayMIDIFromLittleFS example sounds totally distorted, you can still recognize the piano notes but it sounds as if they go through an overdrive guitar pedal first...

I already tried to lower the "gain" from -10 to -15, -20,..:

(AudioGeneratorMIDI.cpp)
 tsf_set_output (g_tsf, TSF_MONO, freq, -20 /* dB gain -10 */ );

but it does no help.

The speaker (8 Ohm, 1W) is directly connected to the board's RX. 
Using a resistance of 1K or 10K only reduces the volume, but not the distortion. 

Yes, I know you should not directly connect a speaker to the ESP, but it reveals it is not generally a problem, as the violin wav sketch plays fine with clean sound.

So any idea what could cause the massive distortion ?

earlephilhower, 2022-01-16:
The MIDI example is written for a DAC, so on directly connecting a speaker to the 8266.  Check the source, you should be able to modify it like PlayWAVfromPROGMEM to use the I2sNoDAC.

closed 2022-01-16 by earlephilhower

henrik6, 2022-01-18:
Sadly, with I2sNoDAC it produces even more distortion:

`out = new AudioOutputI2SNoDAC(); `
`res = midi->begin(mid, out);//dac`