earlephilhower/ESP8266Audio #493: Connect to aac stream with TLS1.3 handshake

marchingband, 2022-02-05:
I am trying to connect to some campus radio stations in Canada, such as this one for CFUV
https://ais-sa1.streamon.fm/7132_64k.aac
I have the StreamMP3FromHTTP.ino example, modified to use the AudioGeneratorAAC class, which works for other .aac streams, but the response from these streamon.fm URL's is just an empty stream. Using wireshark it looks like the connection to streamon.fm uses TLS1.3 handshake. Is it possible to make this happen on an ESP8266?

@earlephilhower do you have any tips on how?

Thanks so so much!

marchingband, 2022-02-05:
might WiFiClientSecure.h be capable?

marchingband, 2022-02-05:
I modified AudioFileSourceHTTPStream.cpp
```
const int httpsPort = 443;
bool AudioFileSourceHTTPStream::open(const char *url)
{
  pos = 0;
  //
  client.setInsecure(); //the magic line, use with caution
  client.connect(url, httpsPort);
  //
  http.begin(client, url);
  http.setReuse(true);
...etc
```

and AudioFileSourceHTTPStream.h
```
#include <Arduino.h>
#ifdef ESP32
  #include <HTTPClient.h>
#else
  #include <ESP8266HTTPClient.h>
  #include <WiFiClientSecure.h>
#endif
#include "AudioFileSource.h"
...
private:
    virtual uint32_t readInternal(void *data, uint32_t len, bool nonBlock);
    // WiFiClient client;
    WiFiClientSecure client;
    HTTPClient http;
```

it seems to connect, but I get `Out of memory error! hAACDecoder==NULL`

marchingband, 2022-02-05:
Does this sound like an SBR thing?

earlephilhower, 2022-02-06:
There is not enough RAM on the 8266 to do TLS (~26KB!) and AAC, sorry.

closed 2022-02-06 by earlephilhower

marchingband, 2022-02-06:
FWIW I am trying the same method on ESP32, with the same mods to your library, and I get some logs saying it is doing the handshake, but fails.
```

...Connecting to WiFi
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:419] _eventCallback(): STA IP: 192.168.0.23, MASK: 255.255.255.0, GW: 192.168.0.1
Connected
[V][ssl_client.cpp:59] start_ssl_client(): Free internal heap before TLS 261288
[V][ssl_client.cpp:65] start_ssl_client(): Starting socket
[E][WiFiGeneric.cpp:739] hostByName(): DNS Failed for http://ais-sa1.streamon.fm/7132_64k.aac
[E][WiFiClientSecure.cpp:133] connect(): start_ssl_client: -1
[V][ssl_client.cpp:265] stop_ssl_socket(): Cleaning SSL connection.
[V][HTTPClient.cpp:245] beginInternal(): url: http://ais-sa1.streamon.fm/7132_64k.aac
[D][HTTPClient.cpp:293] beginInternal(): protocol: http, host: ais-sa1.streamon.fm port: 80 url: /7132_64k.aac
[D][HTTPClient.cpp:579] sendRequest(): request type: 'GET' redirCount: 0

[V][ssl_client.cpp:59] start_ssl_client(): Free internal heap before TLS 260812
[V][ssl_client.cpp:65] start_ssl_client(): Starting socket
[V][ssl_client.cpp:104] start_ssl_client(): Seeding the random number generator
[V][ssl_client.cpp:113] start_ssl_client(): Setting up the SSL/TLS structure...
[I][ssl_client.cpp:127] start_ssl_client(): WARNING: Skipping SSL Verification. INSECURE!
[V][ssl_client.cpp:197] start_ssl_client(): Setting hostname for TLS session...
[V][ssl_client.cpp:212] start_ssl_client(): Performing the SSL/TLS handshake...
[E][ssl_client.cpp:36] _handle_error(): [start_ssl_client():216]: (-29184) SSL - An invalid SSL record was received
[E][WiFiClientSecure.cpp:133] connect(): start_ssl_client: -29184
[V][ssl_client.cpp:265] stop_ssl_socket(): Cleaning SSL connection.
[D][HTTPClient.cpp:1118] connect(): failed connect to ais-sa1.streamon.fm:80
[W][HTTPClient.cpp:1417] returnError(): error(-1): connection refused
[D][HTTPClient.cpp:400] disconnect(): tcp is closed
```

marchingband, 2022-02-06:
any tips? It would be cool to use your library for this project!

marchingband, 2022-02-06:
ESP32-audioI2S seems to work in this particular context out-of-tha-box, so I will just use that for now.
Thanks for getting back so quick! And thanks for the spectacular library you have here <3