earlephilhower/ESP8266Audio #496: LittleFS vs LITTLEFS

gearsincorg, 2022-02-09:
In attempting to use the LittleFS file system with this library on an ESP32, I discovered that the esp32 version of the Little File System uses all upper case letters for the include file name, and the class name.

So I needed to tweak AudioFileSourceLittleFS.h as follows:

```
#ifndef _AUDIOFILESOURCESPIFFS_H
#define _AUDIOFILESOURCESPIFFS_H

#include <Arduino.h>
#include <LITTLEFS.h>

#include "AudioFileSource.h"
#include "AudioFileSourceFS.h"

class AudioFileSourceLittleFS : public AudioFileSourceFS
{
  public:
    AudioFileSourceLittleFS() : AudioFileSourceFS(LITTLEFS) { };
    AudioFileSourceLittleFS(const char *filename) : AudioFileSourceFS(LITTLEFS, filename) {};
    // Others are inherited from base
};

#endif
```

This now works, but seems like a mit of a mashup.
Would the preferred method be to create a new headerfile (AudioFileSourceLITTLEFS.h), or to have an #ifdef to choose between the two class names for the two CPU types?  

Also, I just noticed that the initial #ifndef is wrong.  I must fix that as well. :)

gearsincorg, 2022-02-09:
Note:  I made the change from SPIFFS to LITTLEFS to try to eliminate the buzz that was hapenning at the beginning of any audio file I loaded from SPIFFS memory.  (also see here #136) 

Happy to report that using LITTLEFS completely eliminated the buzz.  I then switched from MP3 to ACC to reduce file sizes.

I decided to go with the following code for AudioFileSourceLittleFS.h so that it did not drop ESP8266 support.:

```
#ifndef _AUDIOFILESOURCELITTLEFS_H
#define _AUDIOFILESOURCELITTLEFS_H

#include <Arduino.h>
#ifdef ESP32
	#include <LITTLEFS.h>
#elif
	#include <LittleFS.h>
#endif

#include "AudioFileSource.h"
#include "AudioFileSourceFS.h"

class AudioFileSourceLittleFS : public AudioFileSourceFS
{
  public:

#ifdef ESP32
    AudioFileSourceLittleFS() : AudioFileSourceFS(LITTLEFS) { };
    AudioFileSourceLittleFS(const char *filename) : AudioFileSourceFS(LITTLEFS, filename) {};
#elif
    AudioFileSourceLittleFS() : AudioFileSourceFS(LittleFS) { };
    AudioFileSourceLittleFS(const char *filename) : AudioFileSourceFS(LittleFS, filename) {};
#endif

};

#endif
```

earlephilhower, 2022-02-09:
You must have a very old ESP32 install, or something went weird on your installation.  Check here: https://github.com/espressif/arduino-esp32/tree/master/libraries/LittleFS/src   The file is `LittleFS.h`, same as the ESP8266.

Good catch on the `#define`, though!

closed 2022-02-09 by earlephilhower

gearsincorg, 2022-02-09:
OK, yes.  7 months out of date apparently.  
With as many projects I have, I'm reluctant to do updates without needing to.  
Time to bite the bullet.  Thanks.

gearsincorg, 2022-02-09:
Wow.   Now I've run into #464.  
Thanks for the advice to update.   :eyeroll: