earlephilhower/ESP8266Audio #497: Any chance this can be ported to STM32?

positron96, 2022-02-11:
I see that support for RP2040 has been added recently, so the library can be ported to another MCU architecture. Any chance the library can be ported to STM32s? A lot of them have DACs and/or I2S.

earlephilhower, 2022-02-12:
That should actually be pretty straightforward.  I don't have any STM32 boards, though, so it will have to be a user contribution.

Basically you would need to just make a new AudioOutput class that takes the raw 16 but samples and send either to an onboard I2S peripheral using DMA or an onboard DAC(again via DMA).

If the board already has an I2S Arduino class, then the work on the RP2040 is probably 80% of what you need.

positron96, 2022-02-12:
Thanks for the answer. It's a pity popular STM32F103 has neither DAC nor I2S.

closed 2022-02-12 by earlephilhower