earlephilhower/ESP8266Audio #498: unable to compile for ESP32 using Arduino IDE

exiledyorkie, 2022-02-12:
Hi

I've used the library for an ESP8266 board which worked fine but now I want to use and ESP32 board. It won't compile for any board, if I remove the ESP8266Audio library, then all is fine. I've tried partitions to increase space and changed the board manager multiple times but nothing works. Any idea's please?

The error messages are below.

\`\`\`
Arduino: 1.8.19 (Windows 10), Board: "ESP32 Dev Module, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, Core 1, Core 1, None"

C:\Users\cpmag\Documents\arduino\libraries\ESP8266Audio\src\AudioFileSourceICYStream.cpp:23: warning: "_GNU_SOURCE" redefined

 #define _GNU_SOURCE

<command-line>: note: this is the location of the previous definition

In file included from C:\Users\cpmag\Documents\arduino\libraries\ESP8266Audio\src\AudioGeneratorMIDI.cpp:65:

C:\Users\cpmag\Documents\arduino\libraries\ESP8266Audio\src\libtinysoundfont/tsf.h: In function 'void tsf_channel_midi_control(tsf*, int, int, int)':

C:\Users\cpmag\Documents\arduino\libraries\ESP8266Audio\src\libtinysoundfont/tsf.h:2100:1: error: insn does not satisfy its constraints:

 }

 ^

(insn 858 343 344 51 (set (reg:SF 19 f0 [407])

        (mem/u/c:SF (symbol_ref/u:SI ("*.LC248") [flags 0x2]) [0  S4 A32])) "C:\Users\cpmag\Documents\arduino\libraries\ESP8266Audio\src\libtinysoundfont/tsf.h":2053 47 {movsf_internal}

     (nil))

during RTL pass: postreload

C:\Users\cpmag\Documents\arduino\libraries\ESP8266Audio\src\libtinysoundfont/tsf.h:2100:1: internal compiler error: in extract_constrain_insn, at recog.c:2210

Please submit a full bug report,

with preprocessed source if appropriate.

See <https://gcc.gnu.org/bugs/> for instructions.

Multiple libraries were found for "SD.h"

 Used: C:\Users\cpmag\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.2\libraries\SD

 Not used: C:\Program Files (x86)\Arduino\libraries\SD

Multiple libraries were found for "WiFi.h"

 Used: C:\Users\cpmag\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.2\libraries\WiFi

 Not used: C:\Program Files (x86)\Arduino\libraries\WiFi

exit status 1

Error compiling for board ESP32 Dev Module.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
\`\`\`

earlephilhower, 2022-02-12:
> C:\Users\cpmag\Documents\arduino\libraries\ESP8266Audio\src\libtinysoundfont/tsf.h:2100:1: internal compiler error: in extract_constrain_insn, at recog.c:2210

The Espressif compiler has a bug, it seems.  I suggest opening an issue there.  This code is pretty simple and others may be hitting the GCC bug, too.

closed 2022-02-12 by earlephilhower

exiledyorkie, 2022-02-12:
Okay, thank you for the advice.

I have posted here;

https://github.com/espressif/arduino-esp32/issues/6274

exiledyorkie, 2022-02-13:
> The Espressif compiler has a bug, it seems. I suggest opening an issue there. This code is pretty simple and others may be hitting the GCC bug, too.

Hi, I posted on the ESP32 github and they state that it is the lib itself. No idea to be honest.

Any thoughts, please?

https://github.com/espressif/arduino-esp32/issues/6274

exiledyorkie, 2022-02-13:
Okay, I'll await and see how it goes. Appreciate the replies.