earlephilhower/ESP8266Audio #508: Compile fails with latest Arduino Core ESP32 2.0.3-rc1

Jason2866, 2022-04-11:
Since Arduino Core is using as much as possible IDF calls some routines and functions have changed. The function / define SPECIAL is not valid anymore.
See PR https://github.com/espressif/arduino-esp32/pull/6527

```
lib/lib_audio/ESP8266Audio/src/spiram-fast.h: In member function 'void ESP8266SPIRAM::begin(int, int)':
lib/lib_audio/ESP8266Audio/src/spiram-fast.h:335:26: error: 'SPECIAL' was not declared in this scope
             pinMode(sck, SPECIAL);
                          ^~~~~~~
lib/lib_audio/ESP8266Audio/src/spiram-fast.h:335:26: note: suggested alternative: 'SERIAL'
             pinMode(sck, SPECIAL);
                          ^~~~~~~
                          SERIAL
```

thire66, 2022-05-12:
Is there any workaround for this problem?

I changed "**Special**" to " 0**xF0**" in the spiram-fast.h file. But now i got another issue in the libtinysoundfont/tsf.h file. 

```
In file included from Arduino\libraries\ESP8266Audio\src\AudioGeneratorMIDI.cpp:65:
Arduino\libraries\ESP8266Audio\src\libtinysoundfont/tsf.h: In function 'void tsf_channel_midi_control(tsf*, int, int, int)':
Arduino\libraries\ESP8266Audio\src\libtinysoundfont/tsf.h:2100:1: error: insn does not satisfy its constraints:
 }
 ^
(insn 858 342 343 51 (set (reg:SF 19 f0 [407])
        (mem/u/c:SF (symbol_ref/u:SI ("*.LC248") [flags 0x2]) [0  S4 A32])) "Arduino\libraries\ESP8266Audio\src\libtinysoundfont/tsf.h":2053 47 {movsf_internal}
     (nil))
during RTL pass: postreload
Arduino\libraries\ESP8266Audio\src\libtinysoundfont/tsf.h:2100:1: internal compiler error: in extract_constrain_insn, at recog.c:2210
Please submit a full bug report,
with preprocessed source if appropriate.
See <https://gcc.gnu.org/bugs/> for instructions.
```

Jason2866, 2022-05-12:
@thire66  For which target (MCU) are you building? Which Arduino core version?

thire66, 2022-05-12:
I am using an ESP32 WROVER Module with the latest Arduino Core ESP32 2.0.3.

Maybe i was to quick with my question. I found the following pull Request: https://github.com/earlephilhower/ESP8266Audio/pull/502/files
Adding the changes, solved my compiler problem!

tobozo, 2022-05-17:
What is this `SPECIAL` macro and what does it have to do with ESP32?

Anyway, I applied #502 too and modified `spiram-fast-h` as follows, just in case this `SPECIAL` macro has a reason to exist:

```C++
// (...)

#ifndef SPECIAL
  #define UNDEF_SPECIAL
  #define SPECIAL 0xf0
#endif

class ESP8266SPIRAM {

  // (...)

};

#ifdef UNDEF_SPECIAL
  #undef SPECIAL
#endif
```

Jason2866, 2022-09-28:
Fixed with esp32 cleanup

closed 2022-09-28 by Jason2866