earlephilhower/ESP8266Audio #511: Serial wav = junk data

spinalcode, 2022-04-16:
I have tried using the wav file dump over serial and saving the data directly to a file, but the output is only noise. Has anyone had any success with this?

earlephilhower, 2022-04-16:
The code is trivial and dumps a WAV header followed by the raw 16b samples sent, and I used it to debug some on-chip issues back when I first wrote the library.

Make sure you are recording the serial port as binary with no translation, and be sure to drop off any chars sent before the binary WAV is sent.  Since the generator doesn't know the total length, it sets the length in the WAV header to a value that most apps recognize as "however long the file turns out to be."

Worst case, import the raw capture as raw 2-channel, 16b samples in Audacity (you may need to try big or little endianness depending on how many chars are in your capture before the 1st sample).

closed 2022-04-16 by earlephilhower

spinalcode, 2022-04-17:
Is this the same when using the web radio example? if is the serial poulted by web chatter?

earlephilhower, 2022-04-17:
It could be, if the code or your app is writing things to the Serial port.  See the AudioLogger class which can silence some of the internal chatter.

 That said, you're not going to be able to write out audio samples to the serial port for any realtime app (i.e. web streaming) since the max baud rate is way too low.