earlephilhower/ESP8266Audio #512: Stream MP3 from HTTPS

PaoloRondot, 2022-04-16:
Hi,

I am able to play music from http with the default address given in the lib example `http://kvbstreams.dyndns.org:8000/wkvi-am`, but I noticed that whenver I try to play from n HTTPS address, such as `https://www.liveradio.es/http://cdn.nrjaudio.fm/audio1/fr/30201/mp3_128.mp3`, it doesn't work and I get the `MP3 source file not open` from the `AudioGeneratorMP3::begin(AudioFileSource *source, AudioOutput *output)`.

My question is, is it actually possible to stream from HTTPS ? Do I have to make some changes in the code ?

Many thanks.
Paolo.

PaoloRondot, 2022-04-16:
I just noticed that this question had already been answered: there's not enough memory on the ESP8266 to stream from HTTPS. Is that still right ?

earlephilhower, 2022-04-16:
There's no fixing the lack of memory for HTTPS streaming on the 8266, sorry.  The buffers and stack space needed for HTTPS means there's way less free memory for MP3 decode than is needed.

closed 2022-04-16 by earlephilhower