earlephilhower/ESP8266Audio #516: ESP32 internal DAC only distorted output

fiechr, 2022-05-03:
I'm currently playing around with the internal DAC(s) of the ESP32.

I've tried two older libraries and examples that actually worked and produced a surprisingly good sound.
But one library (XTronic XT DAC Audio) only compiles with framework 1.8.0 and seems to be not supported anymore and the other is or was more of a "tech sample" from a guy named bitluni. Both produce clear audio output (as clear as is can be), but are either outdated or not really documented at all.

After that I stumbled upon ESP8266Audio (which I disregarded at first, because of its name) and the already existing possibility to use the ESP32's internal DAC by using `AudioOutputI2S(0,1)`.

I'm using the same setup that worked before (10k-1k voltage divider on pin 25, connected to the R input of a PAM8403) and have tried the WAV, AAC and MOD from PROGMEM examples, but I get only a very distorted output. I've tried to select different sample rates, but that didn't change anything. The kind of distortion is always the same, it's completely garbled not just overdriven at some points.

I've tried different ESP32 and a PAM8403 but that produces exactly the same output.

```#include <Arduino.h>

#include "AudioFileSourcePROGMEM.h"
#include "AudioGeneratorWAV.h"
#include "AudioOutputI2SNoDAC.h"

// VIOLA sample taken from https://ccrma.stanford.edu/~jos/pasp/Sound_Examples.html
#include "viola.h"

AudioGeneratorWAV *wav;
AudioFileSourcePROGMEM *file;
AudioOutputI2S *out;

void setup()
{
  Serial.begin(115200);
  delay(1000);
  Serial.printf("WAV start\n");

  audioLogger = &Serial;
  file = new AudioFileSourcePROGMEM( viola, sizeof(viola) );
  out = new AudioOutputI2S(0, 1);
  wav = new AudioGeneratorWAV();
  wav->begin(file, out);
}

void loop()
{
  if (wav->isRunning()) {
    if (!wav->loop()) wav->stop();
  } else {
    Serial.printf("WAV done\n");
    delay(1000);
  }
}```

fiechr, 2022-05-05:
It works, if I use `AudioOutputI2S(0, 2)`.

The question that remains: Is it a bug or a feature?

RomanLut, 2022-05-07:
AudioOutputI2S(0, 2) enables Pulse Density Modulation, AudioOutputI2S(0, 2) enables internal DAC. 
DAC produces full rail-to-rail output 0 ... 3.3V while PDM seems to produce very low sound. 
The problem you are facing is probably overloading of PAM input. Try greater divider or 10uF capacitor beeween output and input.

fiechr, 2022-05-07:
Thank you for your reply!

If there is too much voltage at the input of the PAM, why do the other libraries produce pretty good sound with the exact same hardware setup? I'm already using a 10:1 divider, but have also tried 3.3 V directly with the other demos/sketches and it still produces a acceptable sound. According to the datasheet, the IC should accept up to 5,3 V, when powered with 5 V (which it is) -- and I'm currently not even reaching 0.5 V.

fiechr, 2022-05-07:
The problem was the espressif32 framework version!

Version 4.2.0 produces only the noise described.

Version 3.2.1 and 3.3.2 work fine.

Jason2866, 2022-05-07:
@fiechr It is solved in Arduino Core 2.0.3. Not yet released from Platformio crew.

lidense, 2022-05-15:
> The problem was the espressif32 framework version!
> 
> Version 4.2.0 produces only the noise described.
> 
> Version 3.2.1 and 3.3.2 work fine.

Can you please explain how to revers to 3.2.1 in Plarformio?

fiechr, 2022-05-15:
> Can you please explain how to revers to 3.2.1 in Plarformio?

Just open the platformio.ini and add "@ 3.2.1" in the same line after

    platform = espressif32

jacoherotel, 2022-05-15:
platform = espressif32@3.2.1

lidense, 2022-05-15:
> platform = espressif32@3.2.1

I tried that, I had a compilation error and now Platformio does not load anymore even when I go back to  platform = espressif32

lidense, 2022-05-16:
> > platform = espressif32@3.2.1
> 
> I tried that, I had a compilation error and now Platformio does not load anymore even when I go back to platform = espressif32

I tried to uninstall platformio, to do pio system prune but I get the following errors:
Error: Traceback (most recent call last):
  File "/home/mbantu/.local/lib/python3.8/site-packages/platformio/__main__.py", line 121, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "/home/mbantu/.local/lib/python3.8/site-packages/click/core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "/home/mbantu/.local/lib/python3.8/site-packages/click/core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "/home/mbantu/.local/lib/python3.8/site-packages/platformio/commands/__init__.py", line 44, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "/home/mbantu/.local/lib/python3.8/site-packages/click/core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/mbantu/.local/lib/python3.8/site-packages/click/core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/mbantu/.local/lib/python3.8/site-packages/click/core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/mbantu/.local/lib/python3.8/site-packages/click/core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "/home/mbantu/.local/lib/python3.8/site-packages/platformio/commands/system/command.py", line 135, in system_prune
    reclaimed_core_packages = prune_core_packages(force, dry_run)
  File "/home/mbantu/.local/lib/python3.8/site-packages/platformio/commands/system/prune.py", line 49, in prune_core_packages
    return _prune_packages(force, dry_run, silent, remove_unnecessary_core_packages)
  File "/home/mbantu/.local/lib/python3.8/site-packages/platformio/commands/system/prune.py", line 66, in _prune_packages
    for pkg in handler(dry_run=True)
  File "/home/mbantu/.local/lib/python3.8/site-packages/platformio/package/manager/core.py", line 79, in remove_unnecessary_core_packages
    pkg = pm.get_package(spec)
  File "/home/mbantu/.local/lib/python3.8/site-packages/platformio/package/manager/base.py", line 236, in get_package
    for pkg in self.get_installed():
  File "/home/mbantu/.local/lib/python3.8/site-packages/platformio/package/manager/base.py", line 209, in get_installed
    pkg = PackageItem(pkg_dir)
  File "/home/mbantu/.local/lib/python3.8/site-packages/platformio/package/meta.py", line 405, in __init__
    self.metadata = self.load_meta()
  File "/home/mbantu/.local/lib/python3.8/site-packages/platformio/package/meta.py", line 440, in load_meta
    return PackageMetaData.load(manifest_path)
  File "/home/mbantu/.local/lib/python3.8/site-packages/platformio/package/meta.py", line 391, in load
    data = json.load(fp)
  File "/usr/lib/python3.8/json/__init__.py", line 293, in load
    return loads(fp.read(),
  File "/usr/lib/python3.8/json/__init__.py", line 357, in loads
    return _default_decoder.decode(s)
  File "/usr/lib/python3.8/json/decoder.py", line 337, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/lib/python3.8/json/decoder.py", line 355, in raw_decode
    raise JSONDecodeError("Expecting value", s, err.value) from None
json.decoder.JSONDecodeError: Expecting value: line 1 column 1 (char 

Please advise

jacoherotel, 2022-05-16:
> I tried to uninstall platformio, to do pio system prune but I get the following errors: Error: Traceback (most recent call last): File [...]

That is a different error all together. You need to fix your platformIO first. I used the suggestion and it worked for me. Never had the errors you mention here.

lidense, 2022-05-16:
I finally suceeded in uninstalling platformio by deleting .platformio and reinstalling.
I then suceeded in compilig for platform = espressif32@3.3.2
and miracle: the mp3 file now plays normally!!
If I understand correctly this uses an older toolchain??
Can you please clarify? why is the more recent one not working? Is there a doc explaining more in detail how the espressif32 platform is integrated in platformio.
In any case thanks !!!  for this  precious tip, I had been looking for days for a solution

fiechr, 2022-05-16:
@lidense Yes, it is based on an older version of the framework provided by Espressif. Unfortunately it's not unusual that things break when the framework changes. Sometimes new bugs are introduced or stuff works differently. I guess that's something we have to accept for the low price we pay for this kind hardware.

lidense, 2022-05-16:
OK thanks
Is there a risk that this older toolchain might not be available in the future? Is there a way to store a copy locally? where would I find it and how would the platform.ini have to be modified to use the local copy?
Thanks

TheMechanos, 2022-07-04:
Did you find solution to run it without distorted output on latest framework (5.0.0)?

FedericoBusero, 2022-09-28:
This issue can be closed as the bug is fixed: internal DAC on ESP32 is working in ESP8266Audio version >=1.9.7 and IDF 4.4.1 (arduino core 2.0.3) or IDF 4.4.2 (Arduino core 2.0.4/2.0.5)

If it works in IDF 5.0 should be investigated, but is another issue (the i2s API is totally renewed).

closed 2022-09-28 by earlephilhower