earlephilhower/ESP8266Audio #525: platformio first compile error

murarduino, 2022-05-14:
ENV:
[env:pico32]
platform = espressif32
board = pico32
framework = arduino
lib_deps = earlephilhower/ESP8266Audio@^1.9.6

upload_port = COM4
monitor_port = COM4
monitor_speed = 115200

first compile error
> Executing task in folder esp8266audio: C:\Users\luobin\.platformio\penv\Scripts\platformio.exe run <

Processing pico32 (platform: espressif32; board: pico32; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/pico32.html
PLATFORM: Espressif 32 (4.2.0) > ESP32 Pico Kit
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 3.20002.220503 (2.0.2)
 - tool-esptoolpy 1.30300.0 (3.3.0)
 - toolchain-xtensa-esp32 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 32 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ESP8266Audio> 1.9.6
|   |-- <FFat> 2.0.0
|   |   |-- <FS> 2.0.0
|   |-- <FS> 2.0.0
|   |-- <SPIFFS> 2.0.0
|   |   |-- <FS> 2.0.0
|   |-- <HTTPClient> 2.0.0
|   |   |-- <WiFi> 2.0.0
|   |   |-- <WiFiClientSecure> 2.0.0
|   |   |   |-- <WiFi> 2.0.0
|   |-- <LittleFS> 2.0.0
|   |   |-- <FS> 2.0.0
|   |-- <SD> 2.0.0
|   |   |-- <FS> 2.0.0
|   |   |-- <SPI> 2.0.0
|   |-- <SPI> 2.0.0
|-- <SPIFFS> 2.0.0
|   |-- <FS> 2.0.0
Building in release mode
Compiling .pio\build\pico32\src\main.cpp.o
Generating partitions .pio\build\pico32\partitions.bin
Compiling .pio\build\pico32\lib01e\FS\FS.cpp.o
Compiling .pio\build\pico32\lib01e\FS\vfs_api.cpp.o
Compiling .pio\build\pico32\libae4\FFat\FFat.cpp.o
Compiling .pio\build\pico32\lib95c\SPIFFS\SPIFFS.cpp.o
Compiling .pio\build\pico32\lib08e\WiFi\WiFi.cpp.o
Compiling .pio\build\pico32\lib08e\WiFi\WiFiAP.cpp.o
Compiling .pio\build\pico32\lib08e\WiFi\WiFiClient.cpp.o
Compiling .pio\build\pico32\lib08e\WiFi\WiFiGeneric.cpp.o
Compiling .pio\build\pico32\lib08e\WiFi\WiFiMulti.cpp.o
Archiving .pio\build\pico32\lib01e\libFS.a
Archiving .pio\build\pico32\libae4\libFFat.a
Archiving .pio\build\pico32\lib95c\libSPIFFS.a
Compiling .pio\build\pico32\lib08e\WiFi\WiFiSTA.cpp.o
Compiling .pio\build\pico32\lib08e\WiFi\WiFiScan.cpp.o
Compiling .pio\build\pico32\lib08e\WiFi\WiFiServer.cpp.o
Compiling .pio\build\pico32\lib08e\WiFi\WiFiUdp.cpp.o
Compiling .pio\build\pico32\libb5f\WiFiClientSecure\WiFiClientSecure.cpp.o
Compiling .pio\build\pico32\libb5f\WiFiClientSecure\ssl_client.cpp.o
Compiling .pio\build\pico32\libccc\HTTPClient\HTTPClient.cpp.o
Compiling .pio\build\pico32\lib698\LittleFS\LittleFS.cpp.o
Compiling .pio\build\pico32\lib1a3\SPI\SPI.cpp.o
Compiling .pio\build\pico32\lib4d7\SD\SD.cpp.o
Archiving .pio\build\pico32\lib08e\libWiFi.a
Compiling .pio\build\pico32\lib4d7\SD\sd_diskio.cpp.o
Archiving .pio\build\pico32\libb5f\libWiFiClientSecure.a
Compiling .pio\build\pico32\lib4d7\SD\sd_diskio_crc.c.o
Archiving .pio\build\pico32\lib1a3\libSPI.a
Compiling .pio\build\pico32\libfa1\ESP8266Audio\AudioFileSourceBuffer.cpp.o
Archiving .pio\build\pico32\libccc\libHTTPClient.a
Archiving .pio\build\pico32\lib698\libLittleFS.a
Compiling .pio\build\pico32\libfa1\ESP8266Audio\AudioFileSourceFS.cpp.o
Compiling .pio\build\pico32\libfa1\ESP8266Audio\AudioFileSourceFunction.cpp.o
Compiling .pio\build\pico32\libfa1\ESP8266Audio\AudioFileSourceHTTPStream.cpp.o
Compiling .pio\build\pico32\libfa1\ESP8266Audio\AudioFileSourceICYStream.cpp.o
.pio/libdeps/pico32/ESP8266Audio/src/AudioFileSourceICYStream.cpp:23: warning: "_GNU_SOURCE" redefined
 #define _GNU_SOURCE

<command-line>: note: this is the location of the previous definition
Compiling .pio\build\pico32\libfa1\ESP8266Audio\AudioFileSourceID3.cpp.o
Compiling .pio\build\pico32\libfa1\ESP8266Audio\AudioFileSourcePROGMEM.cpp.o
Archiving .pio\build\pico32\lib4d7\libSD.a
Compiling .pio\build\pico32\libfa1\ESP8266Audio\AudioFileSourceSD.cpp.o
Compiling .pio\build\pico32\libfa1\ESP8266Audio\AudioFileSourceSPIRAMBuffer.cpp.o
Compiling .pio\build\pico32\libfa1\ESP8266Audio\AudioFileSourceSTDIO.cpp.o
Compiling .pio\build\pico32\libfa1\ESP8266Audio\AudioFileStream.cpp.o
Compiling .pio\build\pico32\libfa1\ESP8266Audio\AudioGeneratorAAC.cpp.o
Compiling .pio\build\pico32\libfa1\ESP8266Audio\AudioGeneratorFLAC.cpp.o
Compiling .pio\build\pico32\libfa1\ESP8266Audio\AudioGeneratorMIDI.cpp.o
Compiling .pio\build\pico32\libfa1\ESP8266Audio\AudioGeneratorMOD.cpp.o
Compiling .pio\build\pico32\libfa1\ESP8266Audio\AudioGeneratorMP3.cpp.o
Compiling .pio\build\pico32\libfa1\ESP8266Audio\AudioGeneratorMP3a.cpp.o
Compiling .pio\build\pico32\libfa1\ESP8266Audio\AudioGeneratorOpus.cpp.o
Compiling .pio\build\pico32\libfa1\ESP8266Audio\AudioGeneratorRTTTL.cpp.o
Compiling .pio\build\pico32\libfa1\ESP8266Audio\AudioGeneratorTalkie.cpp.o
Compiling .pio\build\pico32\libfa1\ESP8266Audio\AudioGeneratorWAV.cpp.o
In file included from .pio/libdeps/pico32/ESP8266Audio/src/AudioGeneratorMIDI.cpp:65:
.pio/libdeps/pico32/ESP8266Audio/src/libtinysoundfont/tsf.h: In function 'void tsf_channel_midi_control(tsf*, int, int, int)':
.pio/libdeps/pico32/ESP8266Audio/src/libtinysoundfont/tsf.h:2100:1: error: insn does not satisfy its constraints:
 }
 ^
(insn 858 343 344 51 (set (reg:SF 19 f0 [407])
        (mem/u/c:SF (symbol_ref/u:SI ("*.LC248") [flags 0x2]) [0  S4 A32])) ".pio/libdeps/pico32/ESP8266Audio/src/libtinysoundfont/tsf.h":2053 47 {movsf_internal}
     (nil))
during RTL pass: postreload
.pio/libdeps/pico32/ESP8266Audio/src/libtinysoundfont/tsf.h:2100:1: internal compiler error: in extract_constrain_insn, at recog.c:2210
libbacktrace could not find executable to open
Please submit a full bug report,
with preprocessed source if appropriate.
See <https://gcc.gnu.org/bugs/> for instructions.
Compiling .pio\build\pico32\libfa1\ESP8266Audio\AudioLogger.cpp.o
*** [.pio\build\pico32\libfa1\ESP8266Audio\AudioGeneratorMIDI.cpp.o] Error 1

platformio display Error :
[{
	"resource": "/e:/Documents/PlatformIO/Projects/esp8266audio/.pio/libdeps/pico32/ESP8266Audio/src/libtinysoundfont/tsf.h",
	"owner": "cpp",
	"severity": 8,
	"message": "insn does not satisfy its constraints:",
	"startLineNumber": 2100,
	"startColumn": 1,
	"endLineNumber": 2100,
	"endColumn": 1
}]

thire66, 2022-05-14:
@murarduino 
Got yesterday the same problem.
See https://github.com/earlephilhower/ESP8266Audio/pull/502, this helps a lot.

murarduino, 2022-05-14:
Thanks for the tip, but I still don't quite get it.
As #502 said, it is the version of GCC, but how to check the version of gcc in platformio? How should I go about upgrading it?

earlephilhower, 2022-06-10:
Espressif has no newer version available for you.  The 8.x version has a bug in the code generator, so there's nothing we can do here.  The latest version of the library has the PR you linked to  which disables the MIDI code, so just make sure you're running the latest version (1.9.7) of ESP8266Audio and you're set.

closed 2022-06-10 by earlephilhower