earlephilhower/ESP8266Audio #526: M5Stack/Arduino: mp3 player works with Arduino but distorted noise on Platformio

lidense, 2022-05-14:
Hi I have a sketch to read mp3 files from my SD card.
It works when I compile with the Arduino IDE but although it compiles I can only hear strange distorted noises when I compille with Platformio (with arduino framework, M5 core library and ESP8266AUDIO (VERSION esp32_201 AND 202 compatibility, I did  not use the Platformio version of ESP8266Audio which does not compile but rather VERSION esp32_201 AND_202 compatibility.
Please advise (cpp file attached)
Regards
[plat1.pdf](https://github.com/earlephilhower/ESP8266Audio/files/8693193/plat1.pdf)
s

grimstoner, 2022-05-24:
I am experiencing the distorted sound with a fairly simple implementation in the Arduino IDE:

```
file = new AudioFileSourceSD("/OdeToJoy.mp3");
id3 = new AudioFileSourceID3(file);
out = new AudioOutputI2S(0, 1);
out->SetOutputModeMono(true);
//out->SetGain(0.02);
mp3 = new AudioGeneratorMP3();
mp3->begin(id3 , out);
```

Tried playing with the gain, thinking the audio was clipping. Also tried using output_mode 2 on the AudioOutputI2S, to no avail. How'd you get it working in Arduino IDE? 

Running Arduino IDE 1.18.9, latest version of ESP8266AUDIO, and M5Stack lib 0.4.0

grimstoner, 2022-05-24:
Tried making the project in PlatformIO, with the following configuration file:

```
[env:m5stack-core-esp32]
platform = espressif32@3.3.2
board = m5stack-core-esp32
framework = arduino
lib_deps = 
	earlephilhower/ESP8266Audio@1.9.6
	m5stack/M5Stack@0.4.0
	gianbacchio/ESP8266Spiram@^1.0
```

Tried different combinations of m5stack and ESP8266 audio releases... no joy.

lidense, 2022-05-25:
I got it working with platformio using the following platformio.ini configuration for my M5 Stack Core 2
(M5Unified is a sort of common library for different devices, some slight changes to the code might be necessary.)

[env:m5stack-core2]
platform = espressif32@3.3.2
board = m5stack-core2
framework = arduino
lib_deps = m5stack/M5Unified@^0.0.7

grimstoner, 2022-05-25:
So you don't include earlephilhower/ESP8266Audio as a lib_dep specifically? Is it contained in m5stack/M5Unified?

Might also be that you're using a Core2. I'm using the Core.