earlephilhower/ESP8266Audio #527: TS files

lidense, 2022-05-16:
I'm trying to build an app to read radio streams.
One of the radios I'm interested in streams TS: every 10 seconds it streams a TS file when I save this and read it with player this is what mplayer says:
TS file format detected.
NO VIDEO! AUDIO AAC(pid=256) NO SUBS (yet)!  PROGRAM N. 1
Load subtitles in ./
UDIO: 32000 Hz, 2 ch, floatle, 0.0 kbit/0.00% (ratio: 0->256000)
Selected audio codec: [ffaac] afm: ffmpeg (FFmpeg AAC (MPEG-2/MPEG-4 Audio))
and the file plays.
So I understand this is an aac format but when I used the AudioGeneratorAAC I get a decoding error. I suppose this is because I must first extract the audio from TS?
Please advise

earlephilhower, 2022-05-19:
Yes.  TS needs to be demuxed into the AAC stream before use.  The current code has no understanding of MPEG2 bitstreams, unfortunately.  If you want to add an input filter then a PR would always be welcome. :)

closed 2022-05-19 by earlephilhower

lidense, 2022-05-19:
OK thanks, I will try to see if I can find some code around to help me with the demux

lidense, 2023-08-09:
Hi there, I'm back to my project.
Can you please confirm that the esp8266 audio library will allow me to build an Arduino sketch for a double core esp32  to play a webradio that uses HLS (.ts files). Thanks.

lidense, 2023-08-26:
Can you please explain how to demux a .ts file into aac? Im trying to build a HLS webradio. Thanks