earlephilhower/ESP8266Audio #535: ESP32 Compilation on arduino IDE

builderdev212, 2022-05-27:
This is my first time using the library, so there is a very good chance I have missed something during my installation. However, I wasn't to find any examples of a similar problem, so I am not sure exactly what I did wrong. I was attempting to run the PlayAACFromPROGMEM example program.

Board:
```
Arduino: 1.8.19 (Linux), Board: "ESP32 Dev Module, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, Core 1, Core 1, None"
```

Error:
```
/home/builder/Arduino/libraries/ESP8266Audio/src/AudioFileSourceICYStream.cpp:23: warning: "_GNU_SOURCE" redefined
 #define _GNU_SOURCE
 
<command-line>: note: this is the location of the previous definition
In file included from /home/builder/Arduino/libraries/ESP8266Audio/src/AudioFileSourceSPIRAMBuffer.h:27,
                 from /home/builder/Arduino/libraries/ESP8266Audio/src/AudioFileSourceSPIRAMBuffer.cpp:28:
/home/builder/Arduino/libraries/ESP8266Audio/src/spiram-fast.h: In member function 'void ESP8266SPIRAM::begin(int, int)':
/home/builder/Arduino/libraries/ESP8266Audio/src/spiram-fast.h:335:26: error: 'SPECIAL' was not declared in this scope
             pinMode(sck, SPECIAL);
                          ^~~~~~~
/home/builder/Arduino/libraries/ESP8266Audio/src/spiram-fast.h:335:26: note: suggested alternative: 'SERIAL'
             pinMode(sck, SPECIAL);
                          ^~~~~~~
                          SERIAL
/home/builder/Arduino/libraries/ESP8266Audio/src/AudioOutputI2S.cpp: In member function 'bool AudioOutputI2S::begin(bool)':
/home/builder/Arduino/libraries/ESP8266Audio/src/AudioOutputI2S.cpp:204:41: warning: 'I2S_COMM_FORMAT_I2S' is deprecated [-Wdeprecated-declarations]
         comm_fmt = (i2s_comm_format_t) (I2S_COMM_FORMAT_I2S | I2S_COMM_FORMAT_I2S_LSB);
                                         ^~~~~~~~~~~~~~~~~~~
In file included from /home/builder/.arduino15/packages/esp32/hardware/esp32/2.0.3/tools/sdk/esp32/include/driver/include/driver/i2s.h:16,
                 from /home/builder/Arduino/libraries/ESP8266Audio/src/AudioOutputI2S.cpp:23:
/home/builder/.arduino15/packages/esp32/hardware/esp32/2.0.3/tools/sdk/esp32/include/hal/include/hal/i2s_types.h:87:5: note: declared here
     I2S_COMM_FORMAT_I2S       __attribute__((deprecated)) = 0x01, /*!< I2S communication format I2S, correspond to `I2S_COMM_FORMAT_STAND_I2S`*/
     ^~~~~~~~~~~~~~~~~~~
/home/builder/Arduino/libraries/ESP8266Audio/src/AudioOutputI2S.cpp:204:63: warning: 'I2S_COMM_FORMAT_I2S_LSB' is deprecated [-Wdeprecated-declarations]
         comm_fmt = (i2s_comm_format_t) (I2S_COMM_FORMAT_I2S | I2S_COMM_FORMAT_I2S_LSB);
                                                               ^~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/builder/.arduino15/packages/esp32/hardware/esp32/2.0.3/tools/sdk/esp32/include/driver/include/driver/i2s.h:16,
                 from /home/builder/Arduino/libraries/ESP8266Audio/src/AudioOutputI2S.cpp:23:
/home/builder/.arduino15/packages/esp32/hardware/esp32/2.0.3/tools/sdk/esp32/include/hal/include/hal/i2s_types.h:89:5: note: declared here
     I2S_COMM_FORMAT_I2S_LSB   __attribute__((deprecated)) = 0x02, /*!< I2S format LSB, (I2S_COMM_FORMAT_I2S |I2S_COMM_FORMAT_I2S_LSB) correspond to `I2S_COMM_FORMAT_STAND_MSB`*/
     ^~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/builder/Arduino/libraries/ESP8266Audio/src/AudioGeneratorMIDI.cpp:65:
/home/builder/Arduino/libraries/ESP8266Audio/src/libtinysoundfont/tsf.h: In function 'void tsf_channel_midi_control(tsf*, int, int, int)':
/home/builder/Arduino/libraries/ESP8266Audio/src/libtinysoundfont/tsf.h:2100:1: error: insn does not satisfy its constraints:
 }
 ^
(insn 858 343 344 51 (set (reg:SF 19 f0 [407])
        (mem/u/c:SF (symbol_ref/u:SI ("*.LC248") [flags 0x2]) [0  S4 A32])) "/home/builder/Arduino/libraries/ESP8266Audio/src/libtinysoundfont/tsf.h":2053 47 {movsf_internal}
     (nil))
during RTL pass: postreload
/home/builder/Arduino/libraries/ESP8266Audio/src/libtinysoundfont/tsf.h:2100:1: internal compiler error: in extract_constrain_insn, at recog.c:2210
unrecognized DWARF version in .debug_info at 6
unrecognized DWARF version in .debug_info at 6
Please submit a full bug report,
with preprocessed source if appropriate.
See <https://gcc.gnu.org/bugs/> for instructions.
exit status 1
Error compiling for board ESP32 Dev Module.
```

As I said, there's a chance I have something configured incorrectly, and if so, any help would be greatly appreciated. Thanks in advance!

Iainrach, 2022-05-30:
Had the exact same error with esp32/2.0.3 but just uploaded the latest Zip file and worked fine for me.

As a quick check you can open AudioGeneratorMidi.CPP file and check for this. 

#include "AudioGeneratorMIDI.h"

#if __GNUC__ == 8
// Do not build, GCC8 has a compiler bug
#else // __GNUC__ == 8

It's pretty well covered in previous posts.

earlephilhower, 2022-05-31:
Thanks for the reminder.  I recently redid the CI system to build ESP32-head, ESP8266-head, and RP2040-head, fixing several issues as they came up.  But I forgot to bump the version number.

I've just updated the version to 1.9.7 and within 24 hours the Arduino Library Manager should pick up the update.

Tejas2017610, 2023-10-19:
Again the compilation arising in esp32 boards....