earlephilhower/ESP8266Audio #538: The process of playing pno-cs.mp3 is all noise

murarduino, 2022-06-07:
I use the sketch of PlayMP3FromSPIFFS in the examples to play MP3 music. pno_cs.mp3 also makes the music file for this instance.

Use ESP32-PICO-KIT and Max98357 to build the hardware environment for testing, and VScode+platfromIO to configure the ESP development environment under the Arduino architecture.

`#include <Arduino.h>
#include <WiFi.h>
#include "SPIFFS.h"
#include "AudioFileSourceSPIFFS.h"
#include "AudioFileSourceID3.h"
#include "AudioGeneratorMP3.h"
#include "AudioOutputI2SNoDAC.h"

// To run, set your ESP8266 build to 160MHz, and include a SPIFFS of 512KB or greater.
// Use the "Tools->ESP8266/ESP32 Sketch Data Upload" menu to write the MP3 to SPIFFS
// Then upload the sketch normally.  

// pno_cs from https://ccrma.stanford.edu/~jos/pasp/Sound_Examples.html

AudioGeneratorMP3 *mp3;
AudioFileSourceSPIFFS *file;
AudioOutputI2SNoDAC *out;
AudioFileSourceID3 *id3;

// Called when a metadata event occurs (i.e. an ID3 tag, an ICY block, etc.
void MDCallback(void *cbData, const char *type, bool isUnicode, const char *string)
{
  (void)cbData;
  Serial.printf("ID3 callback for: %s = '", type);

  if (isUnicode) {
    string += 2;
  }
  
  while (*string) {
    char a = *(string++);
    if (isUnicode) {
      string++;
    }
    Serial.printf("%c", a);
  }
  Serial.printf("'\n");
  Serial.flush();
}

void setup()
{
  WiFi.mode(WIFI_OFF); 
  Serial.begin(115200);
  delay(1000);
  SPIFFS.begin();
  Serial.printf("Sample MP3 playback begins...\n");

  audioLogger = &Serial;
  file = new AudioFileSourceSPIFFS("/pno-cs.mp3");
  id3 = new AudioFileSourceID3(file);
  id3->RegisterMetadataCB(MDCallback, (void*)"ID3TAG");
  out = new AudioOutputI2SNoDAC();
  mp3 = new AudioGeneratorMP3();
  mp3->begin(id3, out);
}

void loop()
{
  if (mp3->isRunning()) {
    if (!mp3->loop()) mp3->stop();
  } else {
    Serial.printf("MP3 done\n");
    delay(1000);
  }
}
`

During testing I found some issues but not sure:
1. The frequency of pno_cs.mp3 is 48KHz, and the frequency set by the code is 44.1KHz
2. I use a 1W8R small speaker. Tested with both 3.3V and 5V power supply, (including Gain to +15dB and default 9dB), the speakers only made noise during several tests, and could not hear any melody.

murarduino, 2022-06-07:
monitor print:
`> Executing task in folder ESP32mp3_demo: C:\Users\cecgw\.platformio\penv\Scripts\platformio.exe device monitor <

--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Miniterm on COM3  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1284
load:0x40078000,len:12808
load:0x40080400,len:3032
entry 0x400805e4
Sample MP3 playback begins...
+0 0x3ffb2740
ID3 callback for: Title = 'Piano Sample'
ID3 callback for: Performer = 'Artist Goes Here'
ID3 callback for: Album = 'Album Goes Here'
ID3 callback for: Year = '2010'
ID3 callback for: eof = 'id3'
MP3:ERROR_BUFLEN 0`

earlephilhower, 2022-06-07:
That example is `NoDAC` and does not use an I2S DAC like the Max98357.  You need to modify `AudioOutputI2SNoDAC` to `AudioOutputI2S`.

closed 2022-06-07 by earlephilhower

murarduino, 2022-06-09:
Hi Earle:

    Thanks for the pointers, my MP3 player can at least hear the melody.

#include <Arduino.h&gt;
#include <WiFi.h&gt;
#include "SPIFFS.h"
#include "AudioFileSourceSPIFFS.h"
#include "AudioFileSourceID3.h"
#include "AudioGeneratorMP3.h"
#include "AudioOutputI2SNoDAC.h"

// To run, set your ESP8266 build to 160MHz, and include a SPIFFS of 512KB or greater.
// Use the "Tools-&gt;ESP8266/ESP32 Sketch Data Upload" menu to write the MP3 to SPIFFS
// Then upload the sketch normally.  

// pno_cs from https://ccrma.stanford.edu/~jos/pasp/Sound_Examples.html

AudioGeneratorMP3 *mp3;
AudioFileSourceSPIFFS *file;
AudioOutputI2S *out;
AudioFileSourceID3 *id3;

// Called when a metadata event occurs (i.e. an ID3 tag, an ICY block, etc.
void MDCallback(void *cbData, const char *type, bool isUnicode, const char *string)
{
  (void)cbData;
  Serial.printf("ID3 callback for: %s = '", type);

  if (isUnicode) {
    string += 2;
  }
  
  while (*string) {
    char a = *(string++);
    if (isUnicode) {
      string++;
    }
    Serial.printf("%c", a);
  }
  Serial.printf("'\n");
  Serial.flush();
}

void setup()
{
  WiFi.mode(WIFI_OFF); 
  Serial.begin(115200);
  delay(1000);
  SPIFFS.begin();
  Serial.printf("Sample MP3 playback begins...\n");

  audioLogger = &amp;Serial;
  file = new AudioFileSourceSPIFFS("/pno-cs.mp3");
  id3 = new AudioFileSourceID3(file);
  id3-&gt;RegisterMetadataCB(MDCallback, (void*)"ID3TAG");
  out = new AudioOutputI2S();
  out-&gt;SetPinout(26,25,22);
  mp3 = new AudioGeneratorMP3();
  mp3-&gt;begin(id3, out);

}

void loop()
{
  if (mp3-&gt;isRunning()) {
    if (!mp3-&gt;loop()) mp3-&gt;stop();
  } else {
    Serial.printf("MP3 done\n");
    delay(1000);
  }
}

    However, when MP3 is playing, there are still many broken sounds, which affect the quality of music. I have configured the CPU to work at 240HMz. Do I still need to adjust the Count and len values of dma_buf?