earlephilhower/ESP8266Audio #545: PlayWAVFromPROGMEM example viola.h WAV conversion to header

lwbantoto, 2022-06-16:
Hello, how do you convert a wav file to a header file like viola.h?

https://github.com/earlephilhower/ESP8266Audio/tree/master/examples/PlayWAVFromPROGMEM

Thank you,
Lance

earlephilhower, 2022-06-16:
On Linux where I work, I use: `xxd -i file.wav > file.h`.  Then I edit the file to make the variable `PROGMEM` and you're all set.

Windows should have similar tools.

lwbantoto, 2022-06-16:
Hi @earlephilhower. Thank you. I was able to use xxd on Mac. Which file do you edit PROGMEM to? I can't find it.

earlephilhower, 2022-06-16:
Edit the generated `.h` file and make the variable PROGMEM (see the example for where it would go).

f3rn4ndes, 2022-06-16:
xxd for windows
https://sourceforge.net/projects/xxd-for-windows/
I will try to use it.