earlephilhower/ESP8266Audio #551: ESP32-S2, Internal DAC, not working

TheMechanos, 2022-07-01:
Hey, I try to play a .wav file from SPIFFS on internal DAC on ESP32-S2. 

      #include "AudioGeneratorWAV.h"
      #include "AudioOutputI2SNoDAC.h"
      #include <AudioOutputNull.h>
      #include <AudioOutputI2S.h>
      #include <AudioFileSourceSPIFFS.h>
      
      AudioGeneratorWAV *wav;
      AudioFileSourceSPIFFS *file;
      AudioOutputI2S *out;
      
      void setup() {
        Serial.begin(115200);
      
        if (!SPIFFS.begin(true)) {
            Serial.println("An Error has occurred while mounting SPIFFS");
            return;
        } else {
            Serial.println("SPIFFS Ok!");
        }
      
      
        audioLogger = &Serial;
        file = new AudioFileSourceSPIFFS("/test.wav");
        out = new AudioOutputI2S(0,1);
        wav = new AudioGeneratorWAV();
        wav->begin(file, out);
      }

And I have a error:
`AudioGeneratorWAV::begin: output's begin did not return true`

I think it's a problem with the dac implementation in ESP32-S2. 
After many changes, I was unable to start this. 

Someone knows how to solve it?

GerLech, 2022-07-24:
I had the same problem, and solved it by going back to ESP32 Version 1.0.6

FedericoBusero, 2022-08-24:
Where can I find a working example of DAC use with ESP32-S2 (not using ESP8266Audio)? I tried to add support, but i2s_driver_install returns error.

```
AudioOutputI2S::begin...
...
      i2s_mode_t mode = (i2s_mode_t)(I2S_MODE_MASTER | I2S_MODE_TX);
      if (output_mode == INTERNAL_DAC)
      {
#ifdef CONFIG_IDF_TARGET_ESP32
        mode = (i2s_mode_t)(mode | I2S_MODE_DAC_BUILT_IN);
#elif defined(CONFIG_IDF_TARGET_ESP32S2)
        // nothing to do?
#else
        return false;      
#endif
...
      if (output_mode == INTERNAL_DAC || output_mode == INTERNAL_PDM)
      {
#if CONFIG_IDF_TARGET_ESP32
        i2s_set_pin((i2s_port_t)portNo, NULL);
        i2s_set_dac_mode(I2S_DAC_CHANNEL_BOTH_EN);
#elif defined(CONFIG_IDF_TARGET_ESP32S2)
        i2s_set_pin((i2s_port_t)portNo, NULL);
#else
        return false;
#endif
      }
...
```

FedericoBusero, 2022-08-24:
It seems ESP32-S2 has no DAC mode using I2S

FedericoBusero, 2022-10-23:
To get sound out of ESP32-S2
- I2S-DAC doesn't work because of hardware limitations (I2S cannot go directly to DAC)
- ULP doesn't work because IDF doesn't support ULP-FSM yet
- but the good news: I2SNoDac works! Using the following code, you get audio on pin 17

#include "AudioOutputI2SNoDAC.h"

#define I2S_SPEAKER_SERIAL_CLOCK GPIO_NUM_4 // BCLK
#define I2S_SPEAKER_LEFT_RIGHT_CLOCK GPIO_NUM_5 // WSEL
#define I2S_SPEAKER_SERIAL_DATA GPIO_NUM_17 

AudioOutputI2SNoDAC *out;
...
out = new AudioOutputI2SNoDAC();
out -> SetPinout(I2S_SPEAKER_SERIAL_CLOCK, I2S_SPEAKER_LEFT_RIGHT_CLOCK, I2S_SPEAKER_SERIAL_DATA);