earlephilhower/ESP8266Audio #561: Repeated playback of sound (Mp3 or Wav), unable to fix Unable to install I2S drives error

murarduino, 2022-08-01:
I am using the ESP-PICO-KIT to test the repeated playback of a sound file. Code:
`#include <Arduino.h>
#include <WiFi.h>
#include "SPIFFS.h"

#include "AudioFileSourceSPIFFS.h"
// #include "AudioFileSourceID3.h"
// #include "AudioGeneratorWav.h"
#include "AudioGeneratorMP3.h"
#include "AudioOutputI2S.h"

// To run, set your ESP8266 build to 160MHz, and include a SPIFFS of 512KB or greater.
// Use the "Tools->ESP8266/ESP32 Sketch Data Upload" menu to write the MP3 to SPIFFS
// Then upload the sketch normally.

// pno_cs from https://ccrma.stanford.edu/~jos/pasp/Sound_Examples.html

// AudioGeneratorWAV *wav;
AudioGeneratorMP3 *mp3;
AudioFileSourceSPIFFS *file;
AudioOutputI2S *out;
// AudioFileSourceID3 *id3;

void setup()
{
  WiFi.mode(WIFI_OFF);
  Serial.begin(115200);
  delay(1000);
  SPIFFS.begin();
  Serial.printf("Sample WAV playback begins...\n");

  audioLogger = &Serial;
  file = new AudioFileSourceSPIFFS("/202.mp3");
  out = new AudioOutputI2S();
  out->SetPinout(15, 13, 2);
  out->SetGain(0.5f);
  mp3 = new AudioGeneratorMP3();
  mp3->begin(file, out);
}

void loop()
{
  if (mp3->isRunning())
  {
    if (!mp3->loop())
    {
      mp3->stop();
      delete file;
      delete mp3;
      mp3 = new AudioGeneratorMP3();
    }
  }
  else
  {
    Serial.printf("WAV done\n");
    file = new AudioFileSourceSPIFFS("/202.mp3");
    vTaskDelay(500);
    mp3->begin(file,out);

  }
}`

I add delete pointer object in (!mp3->loop) statement and even cast pointer to null after delete (tried with Wav file too). Sound files can be played repeatedly, but there has been no way to eliminate errors like Unable to install I2S drives.
What should I do to get rid of this error?

DenysChuhlib, 2022-09-26:
Download master version library https://github.com/earlephilhower/ESP8266Audio/archive/refs/heads/master.zip

murarduino, 2022-09-27:
> Download master version library https://github.com/earlephilhower/ESP8266Audio/archive/refs/heads/master.zip

Is there any difference?

codingABI, 2023-08-18:
I got the same **"ERROR: Unable to install I2S drives"** when I try to repeat a audio sample even with the newest version "https://github.com/earlephilhower/ESP8266Audio/archive/refs/heads/master.zip" (from today 18.08.2023) on a ESP32 lolin32. 

`mp3->begin(` calls `installi2s_driver_(` and I think this fails when the driver is already running. So `i2s_driver_uninstall` seems to be needed after `mp3->stop();`. This could be done by the destructor  `delete out;`, but as far I can see the problem seems to be, that `mp3->stop();`  sets `i2sOn=false` and this prevents the destructor for AudioOutputI2S from calling `i2s_driver_uninstall`

@murarduino: Try to add 
`#include "driver/i2s.h"`
and change your loop section to
`{`
`mp3->stop();`
`delete out;`
`i2s_driver_uninstall((i2s_port_t) 0); // Prevents "Unable to install I2S drives"` 
`delete file;`
`delete mp3;`
`mp3 = new AudioGeneratorMP3();`
`out = new AudioOutputI2S();`
`out->SetPinout(15,13, 2);`
`out->SetGain(0.5f);`
`}`