earlephilhower/ESP8266Audio #570: MP3 not playing from SD

PaoloRondot, 2022-09-19:
Dear ESP8266Audio community,

I am trying to play MP3 sounds from an SD card and outputing it to an I2S DAC UDA1334. 
I was able to play from SPIFFS but as I need to store more sounds, I want to switch to SD.

Here is my code, which comes from the example with little modifications:

```
#include <Arduino.h>
#ifdef ARDUINO_ARCH_RP2040
void setup() {}
void loop() {}
#else
#include "AudioFileSourceSD.h"
#include "AudioGeneratorMP3.h"
#include "AudioOutputI2S.h"

// You may need a fast SD card. Set this as high as it will work (40MHz max).
#define SPI_SPEED SD_SCK_MHZ(4)
#define CS_PIN D8

// On ESP32 you can adjust the SPDIF_OUT_PIN (GPIO number). 
// On ESP8266 it is fixed to GPIO3/RX0 and this setting has no effect

File dir;
AudioFileSourceSD *source = NULL;
AudioOutputI2S *output = NULL;
AudioGeneratorMP3 *decoder = NULL;
bool first = true;

void setup() {
  Serial.begin(115200);
  Serial.println();
  delay(1000);

  audioLogger = &Serial;  
  source = new AudioFileSourceSD();
  output = new AudioOutputI2S();
  decoder = new AudioGeneratorMP3();

  // NOTE: SD.begin(...) should be called AFTER AudioOutputSPDIF() 
  //       to takover the the SPI pins if they share some with I2S
  //       (i.e. D8 on Wemos D1 mini is both I2S BCK and SPI SS)
  if (!SD.begin(CS_PIN, SPI_SPEED)) {
    Serial.println("Probleme carte SD");
    return;
  }
  Serial.println("SD initialisee.");
  dir = SD.open("/"); 
}

void loop() {
  if ((decoder) && (decoder->isRunning())) {
    if (!decoder->loop()) decoder->stop();
  } else {
    File file = dir.openNextFile();
    if (file) { 
      if (!first) {
        source->close();
        if (source->open(file.name())) { 
          Serial.printf_P(PSTR("Playing '%s' from SD card...\n"), file.name());
          decoder->begin(source, output);
        } else {
          Serial.printf_P(PSTR("Error opening '%s'\n"), file.name());
        }
      }else first = false;
    } else {
      Serial.println(F("Playback from SD card done\n"));
      delay(1000);
    }       
[... 263 lines cut ...]
3fffff90:  3fffdad0 00000000 3ffee97c 3ffee990  
3fffffa0:  3fffdad0 00000000 3ffee97c 4020ed68  
3fffffb0:  feefeffe feefeffe 3ffe864c 40100dcd  
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3460, room 16 
tail 4
chksum 0xcc
load 0x3fff20b8, len 40, room 4 
tail 4
chksum 0xc9
csum 0xc9
v0006e060
~ld
```

I browsed through the different issue, but didn't find any solution. 
Any idea?

Cheers,
Dorton.

PaoloRondot, 2022-09-21:
It does the same until 20Mhz `#define SPI_SPEED SD_SCK_MHZ(20)`, but above, the SD card just doesn't work: `SD.begin(CS_PIN, SPI_SPEED) = false`

wimmatthijs, 2023-03-25:
@PaoloRondot I'm a bit late here, i'm doing an ESP8266 audio project myself now.

For me it helped to change the ChipSelect Pin of the SD card to D0 because i2S is already using D8

PaoloRondot, 2023-03-25:
That's what I did in the end and it works now. 
Thanks!

closed 2023-03-25 by PaoloRondot

CobaltEcho, 2023-08-20:
I'm trying to use this example, but I keep having an issue compiling with  the SPI_SPEED variable.

The example you shows calls out `if (!SD.begin(CS_PIN, SPI_SPEED)) {` but the Ardiono reference says `SD.begin(cspin)`, and there is no other varable. (Ref: https://www.arduino.cc/reference/en/libraries/sd/begin/)

Also, is that `#endif` really suppose to be there at the end?

Thanks again for posting this, it's getting me a lot closer to doing what I need to do!

PaoloRondot, 2023-08-20:
Hi @CobaltEcho,
The `#endif` should definitely be after the last include, not where it is right now.
As for the `SD.begin`, note that I'm not using the Arduino SD class. Actually, you don't need to add any other imports that the ones I have. This is because SD is already included in `AudioFileSourceSD.h`. (If you are using VS Code and PlatformIO, you can ctrl+left click to navigate in function declarations and source code. I'ld advise you look into the ESP8266Audio files.)
But if you are able to compile without the `SPI_SPEED` and it works, I'ld just go for this.

CobaltEcho, 2023-08-20:
Thanks for the response @PaoloRondot !
I moved my project over to VSCode/PlatformIO because I know it can sometimes do things Arduino cant.  I included the `#endif`, but not sure I have it in the right spot as its throwing errors?
```
void loop() {
  if ((decoder) && (decoder->isRunning())) {
    if (!decoder->loop()) decoder->stop();
  } else {
    File file = dir.openNextFile();
    if (file) { 
      if (!first) {
        source->close();
        if (source->open(file.name())) { 
          Serial.printf_P(PSTR("Playing '%s' from SD card...\n"), file.name());
          decoder->begin(source, output);
        } else {
          Serial.printf_P(PSTR("Error opening '%s'\n"), file.name());
        }
      }else first = false;
    } else {
      Serial.println(F("Playback from SD card done\n"));
      delay(1000);
    }       
  } #endif;
}
```

I see a lot of the codes have the `#define SPI_SPEED SD_SCK_MHZ(4)`, but cant figure out why im getting `error: 'SD_SCK_MHZ' was not declared in this scope`.

PaoloRondot, 2023-08-20:
Well actually I don't think you need this `#endif` and `#ifdef` at all:

```
#include <Arduino.h>
#include "AudioFileSourceSD.h"
#include "AudioGeneratorMP3.h"
#include "AudioOutputI2S.h"

// You may need a fast SD card. Set this as high as it will work (40MHz max).
#define SPI_SPEED SD_SCK_MHZ(4)
#define CS_PIN D8

// On ESP32 you can adjust the SPDIF_OUT_PIN (GPIO number). 
// On ESP8266 it is fixed to GPIO3/RX0 and this setting has no effect

File dir;
AudioFileSourceSD *source = NULL;
AudioOutputI2S *output = NULL;
AudioGeneratorMP3 *decoder = NULL;
bool first = true;

void setup() {
  Serial.begin(115200);
  Serial.println();
  delay(1000);

  audioLogger = &Serial;  
  source = new AudioFileSourceSD();
  output = new AudioOutputI2S();
  decoder = new AudioGeneratorMP3();

  // NOTE: SD.begin(...) should be called AFTER AudioOutputSPDIF() 
  //       to takover the the SPI pins if they share some with I2S
  //       (i.e. D8 on Wemos D1 mini is both I2S BCK and SPI SS)
  if (!SD.begin(CS_PIN, SPI_SPEED)) {
    Serial.println("Probleme carte SD");
    return;
  }
  Serial.println("SD initialisee.");
  dir = SD.open("/"); 
}

void loop() {
  if ((decoder) && (decoder->isRunning())) {
    if (!decoder->loop()) decoder->stop();
  } else {
    File file = dir.openNextFile();
    if (file) { 
      if (!first) {
        source->close();
        if (source->open(file.name())) { 
          Serial.printf_P(PSTR("Playing '%s' from SD card...\n"), file.name());
          decoder->begin(source, output);
        } else {
          Serial.printf_P(PSTR("Error opening '%s'\n"), file.name());
        }
      }else first = false;
    } else {
      Serial.println(F("Playback from SD card done\n"));
      delay(1000);
    }       
  }
}
```
for the `#define SPI_SPEED SD_SCK_MHZ(4)`: when you create your project on platformIO you can choose the board. What I did is I chose NodeMCU 1.0 (ESP-12E Module) and framework: Arduino.
Let me know if this helps