earlephilhower/ESP8266Audio #571: Can't compile AudiooutputSPDIF errors

karmakazi, 2022-09-30:
I had this working last year but after upgrading my Mac I have had trouble compiling. I've fixed most of the issues but this one I can't seem to find any answer to. Here's the error log. Can somebody point me in the right direction? I'm compiling for the M5core2.

/Users/andrewh/Documents/Arduino/libraries/ESP8266Audio/src/AudioOutputI2S.cpp: In member function 'bool AudioOutputI2S::SetPinout()':
/Users/andrewh/Documents/Arduino/libraries/ESP8266Audio/src/AudioOutputI2S.cpp:86:41: error: 'i2s_pin_config_t' has no non-static data member named 'mck_io_num'
         .data_in_num = I2S_PIN_NO_CHANGE};
                                         ^
/Users/andrewh/Documents/Arduino/libraries/ESP8266Audio/src/AudioOutputI2S.cpp: In member function 'bool AudioOutputI2S::begin(bool)':
/Users/andrewh/Documents/Arduino/libraries/ESP8266Audio/src/AudioOutputI2S.cpp:224:28: error: 'I2S_MCLK_MULTIPLE_DEFAULT' was not declared in this scope
           .mclk_multiple = I2S_MCLK_MULTIPLE_DEFAULT, // Unused
                            ^~~~~~~~~~~~~~~~~~~~~~~~~
/Users/andrewh/Documents/Arduino/libraries/ESP8266Audio/src/AudioOutputI2S.cpp:224:28: note: suggested alternative: 'RTC_CLK_CONFIG_DEFAULT'
           .mclk_multiple = I2S_MCLK_MULTIPLE_DEFAULT, // Unused
                            ^~~~~~~~~~~~~~~~~~~~~~~~~
                            RTC_CLK_CONFIG_DEFAULT
/Users/andrewh/Documents/Arduino/libraries/ESP8266Audio/src/AudioOutputI2S.cpp:225:28: error: 'I2S_BITS_PER_CHAN_DEFAULT' was not declared in this scope
           .bits_per_chan = I2S_BITS_PER_CHAN_DEFAULT // Use bits per sample
                            ^~~~~~~~~~~~~~~~~~~~~~~~~
/Users/andrewh/Documents/Arduino/libraries/ESP8266Audio/src/AudioOutputI2S.cpp:225:28: note: suggested alternative: 'I2S_BITS_PER_SAMPLE_8BIT'
           .bits_per_chan = I2S_BITS_PER_CHAN_DEFAULT // Use bits per sample
                            ^~~~~~~~~~~~~~~~~~~~~~~~~
                            I2S_BITS_PER_SAMPLE_8BIT
/Users/andrewh/Documents/Arduino/libraries/ESP8266Audio/src/AudioOutputI2S.cpp:227:7: error: 'i2s_config_t' has no non-static data member named 'mclk_multiple'
       };
       ^
/Users/andrewh/Documents/Arduino/libraries/ESP8266Audio/src/AudioOutputSPDIF.cpp: In constructor 'AudioOutputSPDIF::AudioOutputSPDIF(int, int, int)':
/Users/andrewh/Documents/Arduino/libraries/ESP8266Audio/src/AudioOutputSPDIF.cpp:109:22: error: 'I2S_MCLK_MULTIPLE_DEFAULT' was not declared in this scope
     .mclk_multiple = I2S_MCLK_MULTIPLE_DEFAULT, // Unused
                      ^~~~~~~~~~~~~~~~~~~~~~~~~
/Users/andrewh/Documents/Arduino/libraries/ESP8266Audio/src/AudioOutputSPDIF.cpp:109:22: note: suggested alternative: 'RTC_CLK_CONFIG_DEFAULT'
     .mclk_multiple = I2S_MCLK_MULTIPLE_DEFAULT, // Unused
                      ^~~~~~~~~~~~~~~~~~~~~~~~~
                      RTC_CLK_CONFIG_DEFAULT
/Users/andrewh/Documents/Arduino/libraries/ESP8266Audio/src/AudioOutputSPDIF.cpp:110:22: error: 'I2S_BITS_PER_CHAN_DEFAULT' was not declared in this scope
     .bits_per_chan = I2S_BITS_PER_CHAN_DEFAULT // Use bits per sample
                      ^~~~~~~~~~~~~~~~~~~~~~~~~
/Users/andrewh/Documents/Arduino/libraries/ESP8266Audio/src/AudioOutputSPDIF.cpp:110:22: note: suggested alternative: 'I2S_BITS_PER_SAMPLE_8BIT'
     .bits_per_chan = I2S_BITS_PER_CHAN_DEFAULT // Use bits per sample
                      ^~~~~~~~~~~~~~~~~~~~~~~~~
                      I2S_BITS_PER_SAMPLE_8BIT
/Users/andrewh/Documents/Arduino/libraries/ESP8266Audio/src/AudioOutputSPDIF.cpp:112:3: error: 'i2s_config_t' has no non-static data member named 'mclk_multiple'
   };
   ^
/Users/andrewh/Documents/Arduino/libraries/ESP8266Audio/src/AudioOutputSPDIF.cpp: In member function 'bool AudioOutputSPDIF::SetPinout(int, int, int)':
/Users/andrewh/Documents/Arduino/libraries/ESP8266Audio/src/AudioOutputSPDIF.cpp:163:3: error: 'i2s_pin_config_t' has no non-static data member named 'mck_io_num'
   };
   ^

scarolan, 2023-01-11:
Same issue here with the 'mck_io_num' error.  Running Windows 11 and Arduino IDE 2.0.3

kn1cht, 2023-02-22:
This issue has been already fixed in #565 but the latest version `1.9.7` seems not to include the fix yet.
https://github.com/earlephilhower/ESP8266Audio/compare/1.9.7...12131e9

As a workaround until the new version appears, I pulled the library from GitHub and succeeded to compile my project.

(platformio.ini)
````
lib_deps = 
	# ...
	https://github.com/earlephilhower/ESP8266Audio#master
```