earlephilhower/ESP8266Audio #587: How to play a wav file repeatedly, stop it on an event and start it or another file after

kcriqui, 2022-11-23:
I'm working with a QT Py RP2040 with LittleFS and AudioOutputI2SNoDAC. I modified the examples and am able to play a single file. FWIW, I put an RC low pass filter on the output and it drives a small Class D amplifier nicely. What I'd like to be able to do is start an arbitrary wav file playing and have it repeat indefinitely until an event. i.e. start playing on button press and restart at the beginning until the button is released. I'm still stuck on playing a file more than once. I think my problem is knowing which objects can be reused. Here's what I have now:

```
#include <Arduino.h>

#include "AudioGeneratorWAV.h"
#include "AudioOutputI2SNoDAC.h"
#include <AudioFileSourceLittleFS.h>

#define I2SoutPin 5
#define I2SclkBasePin 1
#define I2Sgain 0.5

AudioGeneratorWAV *wav;
AudioFileSourceLittleFS *file;
AudioOutputI2SNoDAC *out;

#define buttonL 25  // QTPy pin SCL
#define buttonR 20  // QTPy pin TX

void setup() {
  Serial.begin(115200);
  delay(3000);

  pinMode(buttonL, INPUT_PULLUP);
  pinMode(buttonR, INPUT_PULLUP);

  audioLogger = &Serial;

  out  = new AudioOutputI2SNoDAC(I2SoutPin, I2SclkBasePin); // output pin, clock pin+1
  out->SetGain(I2Sgain);
  wav  = new AudioGeneratorWAV();
}

void startWAV(char *filename, AudioGeneratorWAV *wav, AudioOutputI2SNoDAC *out) {
  Serial.print("starting ");
  Serial.println(filename);
  file = new  AudioFileSourceLittleFS(filename);
  int32_t filesize = file->getSize();
  Serial.print("file size = ");
  Serial.println(filesize);
  wav->begin(file, out);
  Serial.println("playing");
}

void loop() {
  Serial.println("loop start");
  Serial.println("waiting for button");
  while (digitalRead(buttonL)) {}            // wait for left button to be pressed
  startWAV("/1.wav", wav, out);

  if (wav->isRunning()) {
    Serial.println("running");
    if (!wav->loop()) {
      Serial.println("end of wav");
      wav->stop();
      //  delete wav;
      delete file;
      //  delete out;
      pinMode(I2SoutPin, INPUT);
    }
  }
}

```
It runs OK the first time but stops at the wav->begin the second time through. At that point I have to reset the board to get the IDE to respond.  Here what I see on the serial monitor

```
loop start
waiting for button
starting /1.wav
file size = 36176
playing
running
end of wav
loop start
waiting for button
starting /1.wav
file size = 36176
```

kcriqui, 2022-12-02:
Closing since this isn't actually an issue. Pointers to a better forum for questions about the library appreciated.

HeikkiHietala, 2022-12-02:
Well, this is exactly what I'd like to achieve too - I have a clock that says the time, but I can't get the bugger to shut up, It keeps repeating the time, and all my efforts in curbing the babble have been unsuccessful.

kcriqui, 2022-12-02:
I ended up switching to CircuitPython :/ The Adafruit audiocore and audiopwmio libraries are pretty easy to work with.

closed 2022-12-02 by kcriqui

HeikkiHietala, 2022-12-02:
Wooow, that looks exactly like what I need to achieve. Thanks for the hint!